RemixVSL/iomemory-vsl #1: oops after flashing new firmware

vladie, 2016-09-17:
hi

i've installed fresh copy of ubuntu 16.04.1
downloaded latest firmware, utilities from sandisk
and your's driver
after flashing new firmware with fio-update-firmware i've started rebooting process and iomemory-vsl driver oops'ed. i've attached screenshot to this issue.

[image]

vladie, 2016-09-17:
after rebooting all worked fine.

snuf, 2016-12-03:
Ah interesting, and thank you for the screenshot and bug!

The spin_locks are in the kscsi_host.c and are called on initialization and tear down of the SCSI driver, and is linked to loading and unloading of the module.

I know it's been a while, but do you recall if you manually loaded the module after flashing ?

Sorry for the delay btw!

vladie, 2016-12-03:
I've compiled the module, loaded it and then flashed firmware. After
flashing is finished, i've initiated host reboot. In few seconds after
this, kernel have oops'ed.

snuf, 2016-12-03:
Ah, that probably explains it. The new firmware isn't active until you've rebooted the host and the card has gone to it's own initialisation phase. The firmware update process should normally say that you must reboot before the new firmware is activated. The newer drivers are not compatible with the older firmware, which is still active on the card till reboot.

Perhaps the readme is a bit vague about this when re-reading it, I'll update it to state the update has to be done before the new driver is installed and used.

On the other hand I think the driver should not load if the correct firmware is not active, I'll have to take a look at how it checks that and see if there is a way to actually tell if the right firmware is running instead of loaded.

plappermaul, 2019-02-24:
Close due to inactivity ...

closed 2019-02-24 by plappermaul