RemixVSL/iomemory-vsl #3: kernel 4.4.0-64-generic :(

fullmoon4465, 2017-03-03:
Any way to get this working with newer kernel?

Building module:
cleaning build area....
'make' DKMS_KERNEL_VERSION=4.4.0-64-generic............(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.4.0-64-generic (x86_64)
Consult /var/lib/dkms/iomemory-vsl/3.2.10/build/make.log for more information.

snuf, 2017-03-08:
Yep can do, will have a look and reply when done.

snuf, 2017-03-08:
@fullmoon4465 could you put your /var/lib/dkms/iomemory-vsl/3.2.10/build/make.log in http://pastebin.com/ and post the link ? just to make sure I'm not seeing something else than you are.

snuf, 2017-03-08:
@fullmoon4465 looking at my results it seems like a test concurrency/order issue in the make process, am investigating.

fullmoon4465, 2017-03-08:
I'm actually using Miratnis Fuel 9.0, then upgraded to 9.2 which is where the new kernel came in to play with the bootstrap image you build. I can probably try it again tomorrow. After you use the bootstrap though (live disk), I don't think it will do what I need it to do without a reboot on the server, meaning I can't do what I wanted pre install of the OS openstack.

I can however compile just fine after fuel has deployed openstack as its not CentOS but Mitaka on Ubuntu 14.04. Then use ceph to add the osd's after this driver's installed. TMI but notes for my self. will get back

snuf, 2017-03-08:
@fullmoon4465 Ah! thanks for the context there! (am familiar with Fuel) I've been playing around with make and the -j flag and noticed something odd on a single core VM, it kind of feels like it's doing a fork bomb when -j is present no matter the amount of CPUs you give it. When removing the -j flag from the Makefile it does compile in one go here. 
I'm out of juice for now, will continue tomorrow. Are you in the bay area per chance ? (need to figure out private messaging hahaha)

fullmoon4465, 2017-03-08:
Nah midwest. But we all work remote anyway until hardware is needed lol

snuf, 2017-03-08:
Haha cool, too bad you're not closer or I would have hopped by! :)

snuf, 2017-03-14:
@fullmoon4465 sorry for the delay, had some things to take care of in the meantime. Problem should be fixed, I'll close the ticket for now, and thanks for filing the bug! If the problem persist please re-open :)

closed 2017-03-14 by snuf