RemixVSL/iomemory-vsl #4: Unable to compile on debian stretch

mhlavac80, 2017-09-22:
/usr/bin/make \
    -j8 \
-C /lib/modules/4.9.0-3-amd64/build \
FIO_DRIVER_NAME=iomemory-vsl \
FIO_SCSI_DEVICE=False \
FUSION_DRIVER_DIR=/tmp/iomemory-vsl/driver_source \
SUBDIRS=/tmp/iomemory-vsl/driver_source \
EXTRA_CFLAGS+="-I/tmp/iomemory-vsl/driver_source/include -DBUILDING_MODULE -DLINUX_IO_SCHED" \
INSTALL_MOD_DIR=extra/fio \
INSTALL_MOD_PATH= \
KFIO_LIB=kfio/x86_64_cc63_libkfio.o_shipped \
modules
make[2]: Entering directory '/usr/src/linux-headers-4.9.0-3-amd64'
  CC [M]  /tmp/iomemory-vsl/driver_source/main.o
  CC [M]  /tmp/iomemory-vsl/driver_source/license.o
  CC [M]  /tmp/iomemory-vsl/driver_source/pci.o
  CC [M]  /tmp/iomemory-vsl/driver_source/sysrq.o
  CC [M]  /tmp/iomemory-vsl/driver_source/driver_init.o
  CC [M]  /tmp/iomemory-vsl/driver_source/kfio.o
  CC [M]  /tmp/iomemory-vsl/driver_source/state.o
  CC [M]  /tmp/iomemory-vsl/driver_source/errno.o
  CC [M]  /tmp/iomemory-vsl/driver_source/kcache.o
  CC [M]  /tmp/iomemory-vsl/driver_source/kfile.o
In file included from /usr/src/linux-headers-4.9.0-3-common/include/linux/module.h:18:0,
                 from /tmp/iomemory-vsl/driver_source/driver_init.c:28:
/tmp/iomemory-vsl/driver_source/driver_init.c:50:52: error: initializer element is not constant
 KFIO_MODULE_PARAM_ARRAY(numa_node_override, charp, num_numa_node_override, S_IRUGO | S_IWUSR);
                                                    ^
/usr/src/linux-headers-4.9.0-3-common/include/linux/moduleparam.h:445:39: note: in definition of macro ‘module_param_array_named’
  = { .max = ARRAY_SIZE(array), .num = nump,                      \
                                       ^~~~
/tmp/iomemory-vsl/driver_source/include/fio/port/common-linux/kfio.h:141:54: note: in expansion of macro ‘module_param_array’
 #define KFIO_MODULE_PARAM_ARRAY(name,type,nump,perm) module_param_array(name,type,nump,perm)
                                                      ^~~~~~~~~~~~~~~~~~
/tmp/iomemory-vsl/driver_source/driver_init.c:50:1: note: in expansion of macro ‘KFIO_MODULE_PARAM_ARRAY’
 KFIO_MODULE_PARAM_ARRAY(numa_node_override, charp, num_numa_node_override, S_IRUGO | S_IWUSR);
 ^~~~~~~~~~~~~~~~~~~~~~~
/tmp/iomemory-vsl/driver_source/driver_init.c:50:52: note: (near initialization for ‘__param_arr_numa_node_override.num’)
 KFIO_MODULE_PARAM_ARRAY(numa_node_override, charp, num_numa_node_override, S_IRUGO | S_IWUSR);
                                                    ^
/usr/src/linux-headers-4.9.0-3-common/include/linux/moduleparam.h:445:39: note: in definition of macro ‘module_param_array_named’
  = { .max = ARRAY_SIZE(array), .num = nump,                      \
                                       ^~~~
/tmp/iomemory-vsl/driver_source/include/fio/port/common-linux/kfio.h:141:54: note: in expansion of macro ‘module_param_array’
 #define KFIO_MODULE_PARAM_ARRAY(name,type,nump,perm) module_param_array(name,type,nump,perm)
                                                      ^~~~~~~~~~~~~~~~~~
/tmp/iomemory-vsl/driver_source/driver_init.c:50:1: note: in expansion of macro ‘KFIO_MODULE_PARAM_ARRAY’
 KFIO_MODULE_PARAM_ARRAY(numa_node_override, charp, num_numa_node_override, S_IRUGO | S_IWUSR);
 ^~~~~~~~~~~~~~~~~~~~~~~
/usr/src/linux-headers-4.9.0-3-common/scripts/Makefile.build:298: recipe for target '/tmp/iomemory-vsl/driver_source/driver_init.o' failed
make[5]: *** [/tmp/iomemory-vsl/driver_source/driver_init.o] Error 1
make[5]: *** Waiting for unfinished jobs....
/usr/src/linux-headers-4.9.0-3-common/Makefile:1507: recipe for target '_module_/tmp/iomemory-vsl/driver_source' failed
make[4]: *** [_module_/tmp/iomemory-vsl/driver_source] Error 2
Makefile:150: recipe for target 'sub-make' failed
make[3]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.9.0-3-amd64'
Makefile:82: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/tmp/iomemory-vsl/driver_source'
ERROR:
debian/rules:94: recipe for target 'build-arch-stamp' failed
make: *** [build-arch-stamp] Error 1
dpkg-buildpackage: error: debian/rules build gave error exit status 2

snuf, 2017-09-23:
Hi,

Thanks for reporting! fixed this one issue, but there are a few more after this, so am fixing those atm.

Cheers,

Funs

snuf, 2017-09-23:
Hi Again,

I've added tests and made changes so it compiles, the version that compiles on stretch is in https://github.com/snuf/iomemory-vsl/tree/git_issue_4 changes can be viewed here: a8ced6f4fcf5e86936ad11b48aa656a301703431

Mind you I don't have access to a card atm, and have not been able to test the effects of the changes and won't be till at least the 4th of Oct.

Cheers,

Funs

mhlavac80, 2017-09-23:
Thanks for the quick fix.  It compiles and loads into the kernel nicely:

[  907.018232] <6>fioinf VSL configuration hash: 81e088889e3a8391880d69dcc82721b9f9f17093
[  907.018353] <6>fioinf 
[  907.018355] <6>fioinf Copyright (c) 2006-2014 Fusion-io, Inc. (acquired by SanDisk Corp. 2014)
[  907.018356] <6>fioinf Copyright (c) 2014-2015 SanDisk Corp. and/or all its affiliates. All rights reserved.
[  907.018357] <6>fioinf For Terms and Conditions see the License file included
[  907.018359] <6>fioinf with this driver package.
[  907.018360] <6>fioinf 
[  907.018362] <6>fioinf ioDrive driver 3.2.10.1509 pinnacles@4d66ccc0e0f0 loading...
[  907.019484] <6>fioinf ioDrive 0000:01:00.0: mapping controller on BAR 5
[  907.019597] <6>fioinf ioDrive 0000:01:00.0: MSI enabled
[  907.019600] <6>fioinf ioDrive 0000:01:00.0: using MSI interrupts
[  907.049753] <6>fioinf ioDrive 0000:01:00.0.0: Starting master controller
[  907.822457] <6>fioinf ioDrive 0000:01:00.0.0: Adapter serial number is 124123
[  908.595161] <6>fioinf ioDrive 0000:01:00.0.0: Board serial number is 124123
[  908.595162] <6>fioinf ioDrive 0000:01:00.0.0: Adapter serial number is 124123
[  908.597409] <6>fioinf ioDrive 0000:01:00.0.0: Default capacity        320.000 GBytes
[  908.597410] <6>fioinf ioDrive 0000:01:00.0.0: Default sector size     512 bytes
[  908.597410] <6>fioinf ioDrive 0000:01:00.0.0: Rated endurance         4.00 PBytes
[  908.597411] <6>fioinf ioDrive 0000:01:00.0.0: 85C temp range hardware found
[  908.599046] <6>fioinf ioDrive 0000:01:00.0.0: Firmware version 7.1.17 116786 (0x700411 0x1c832)
[  908.599047] <6>fioinf ioDrive 0000:01:00.0.0: Platform version 10 
[  908.599047] <6>fioinf ioDrive 0000:01:00.0.0: Firmware VCS version 116786 [0x1c832]
[  908.599061] <6>fioinf ioDrive 0000:01:00.0.0: Firmware VCS uid 0xaeb15671994a45642f91efbb214fa428e4245f8a
[  908.601284] <6>fioinf ioDrive 0000:01:00.0.0: Powercut flush: Enabled
[  908.697266] <6>fioinf ioDrive 0000:01:00.0.0: PCIe power monitor enabled (master). Limit set to 24.750 watts.
[  908.697267] <6>fioinf ioDrive 0000:01:00.0.0: Thermal monitoring: Enabled
[  908.697268] <6>fioinf ioDrive 0000:01:00.0.0: Hardware temperature alarm set for 85C.
[  908.703085] <6>fioinf ioDrive 0000:01:00.0: Found device fct0 (Fusion-io ioDrive 320GB 0000:01:00.0) on pipeline 0
[  908.706628] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Upgrade: Detected 11bit ECC media via probe of EB 0.
[  908.709983] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Upgrade: Found old LEB map revision 6 in EB 0.
[  908.710002] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: entering raw mode for LEB map recovery. This may take several minutes.
[  908.710325] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: reading old LEB map page 1/512...
[  908.865584] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: reading old LEB map page 2/512...
[  909.027898] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: resetting after LEB map recovery...
[  909.069178] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Powercut flush: Enabled
[  909.154816] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: old LEB map recovered successfully.
[  909.154817] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Upgrade: attach disabled.
[  909.155608] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Upgrade: Scanning for old NV data...
[  922.100944] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Upgrade: Average erase count is 2840
[  922.100970] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Upgrade: Retrieving old NV data from block 3268
[  922.102801] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: probed fct0
[  922.102803] <6>fioinf Fusion-io ioDrive 320GB 0000:01:00.0: Attaching explicitly disabled
[  922.102804] <3>fioerr Fusion-io ioDrive 320GB 0000:01:00.0: auto attach failed with error EINVAL: Invalid argument

What's interesting, is that without the kernel module loaded, I can run fio-status and other commands and get a response back.  As soon as I load the kernel module, I get:
root@whisper:/tmp/usr/bin# ./fio-status 
error: failed to mmap memory
No devices were found in this system. Please check if the iomemory-vsl driver is loaded.
root@whisper:/tmp/usr/bin# lsmod |grep io
iomemory_vsl         1245184  0

Also, no block device has shown up in /dev/fio*

snuf, 2017-10-02:
Hey there,

Finally got the card in, fixed the issue and it seems to work now, could you do a pull and try again ?

[ 1367.574605] <6>fioinf ioDrive 0000:00:09.0.0: Starting master controller
[ 1367.642017] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:08.0: sector_size=512
[ 1367.642020] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:08.0: setting channel range data to [2 .. 2047]
[ 1367.733748] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:08.0: Found metadata in EBs 1434-1413, loading...
[ 1368.023387] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:08.0: setting recovered append point 1413+196116480
[ 1368.028144] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:08.0: Creating device of size 320000000000 bytes with 625000000 sectors of 512 bytes (584127840 mapped).
[ 1368.034540] fioinf Fusion-io ioDrive Duo 640GB 0000:00:08.0: Creating block device fioa: major: 253 minor: 0 sector size: 512...
[ 1368.035638] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:08.0: Attach succeeded.
[ 1368.548521] <6>fioinf ioDrive 0000:00:09.0.0: Adapter serial number is 440178
[ 1369.526004] <6>fioinf ioDrive 0000:00:09.0.0: Board serial number is 443247
[ 1369.526009] <6>fioinf ioDrive 0000:00:09.0.0: Default capacity        320.000 GBytes
[ 1369.526010] <6>fioinf ioDrive 0000:00:09.0.0: Default sector size     512 bytes
[ 1369.526011] <6>fioinf ioDrive 0000:00:09.0.0: Rated endurance         5.00 PBytes
[ 1369.526013] <6>fioinf ioDrive 0000:00:09.0.0: 85C temp range hardware found
[ 1369.531427] <6>fioinf ioDrive 0000:00:09.0.0: Firmware version 7.1.13 109322 (0x70040d 0x1ab0a)
[ 1369.531429] <6>fioinf ioDrive 0000:00:09.0.0: Platform version 10 
[ 1369.531430] <6>fioinf ioDrive 0000:00:09.0.0: Firmware VCS version 109322 [0x1ab0a]
[ 1369.531442] <6>fioinf ioDrive 0000:00:09.0.0: Firmware VCS uid 0x02a0b00fd435f92c3cebbf431f69bd563a77cfe3
[ 1369.534297] <6>fioinf ioDrive 0000:00:09.0.0: Powercut flush: Enabled
[ 1369.640582] <6>fioinf ioDrive 0000:00:09.0.0: PCIe power monitor enabled (slave). Limit set to 24.750 watts.
[ 1369.640584] <6>fioinf ioDrive 0000:00:09.0.0: Thermal monitoring: Enabled
[ 1369.640587] <6>fioinf ioDrive 0000:00:09.0.0: Hardware temperature alarm set for 85C.
[ 1369.648975] <6>fioinf ioDrive 0000:00:09.0: Found device fct1 (Fusion-io ioDrive Duo 640GB 0000:00:09.0) on pipeline 0
[ 1370.298210] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:09.0: probed fct1
[ 1370.423920] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:09.0: sector_size=512
[ 1370.423924] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:09.0: setting channel range data to [2 .. 2047]
[ 1370.498329] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:09.0: Found metadata in EBs 1434-1408, loading...
[ 1370.795505] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:09.0: setting recovered append point 1408+198180864
[ 1370.800285] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:09.0: Creating device of size 320000000000 bytes with 625000000 sectors of 512 bytes (581683056 mapped).
[ 1370.800383] fioinf Fusion-io ioDrive Duo 640GB 0000:00:09.0: Creating block device fiob: major: 253 minor: 16 sector size: 512...
[ 1370.801122] <6>fioinf Fusion-io ioDrive Duo 640GB 0000:00:09.0: Attach succeeded.
root@stretch:/home/test/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.10# ls /dev/fio*
/dev/fioa  /dev/fiob

Cheers,

Funs

snuf, 2017-10-02:
After some more extensive testing it seems I'm getting page allocation failures on a dirty device with the 3.9.0 kernel, initially I thought it was due 16 CPUs but it seems to happen with 4 cpus too now. So am investigating that.

snuf, 2017-10-02:
Hmm problem seems to be related to passthrough-PCI to the stretch VM, investigating.

snuf, 2017-10-06:
Problem not related to passthrough, but due to other issue. fixed now. Moved to the 3.2.15 codebase too and adapted to work with 4.9.0, is in't is own branch and also in master, tested with passthrough on stretch and ubuntu 17.04 (4.10.0).

closed 2017-10-06 by snuf

mhlavac80, 2017-10-08:
Everything works like a dream, thanks!