RemixVSL/iomemory-vsl #5: trouble compiling against 4.13.9

rmonk, 2017-10-31:
I'm on Fedora 26, running 4.13.9-200.fc26.x86_64

When trying to build, I was getting errors related to bi_error not existing.  After some research, it seems that it's been renamed to bi_status?  I replaced those in the kblock.c.  Now, however, I get:

/root/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:1779:5: error: implicit declaration of function 'blk_queue_bounce'; did you mean 'blk_queue_io_opt'? [-Werror=implicit-function-declaration]
     blk_queue_bounce(queue, &bio);
     ^~~~~~~~~~~~~~~~

I tried looking for this, but it seems it's been removed?  I do see it in the map for the kernel, but there's no prototype in the includes.

I'm sure I'm doing something wrong, but I can't seem to figure it out, any help would be appreciated.

snuf, 2017-10-31:
Hi Richard,

Thanks for your bug, don't think you're doing anything wrong!

Indeed, my auto-compile also noticed that in 4.13.7 that bi_error had "evaporated" and was moved to bi_status, however bi_status is a different type than bi_error was, so just renaming it might yield some interesting results (probably not looking at elixir, but might later down the line). Had unfortunately not come around to committing the change and have auto-test run.

With regard to blk_queue_bounce I'm curious if you have CONFIG_BOUNCE in your kernel config and would love to know if this is a stock kernel for Fedora ? 
*grep BOUNCE /boot/config-*
[I love elixir](http://elixir.free-electrons.com/linux/v4.13.9/source/block/blk.h#L358)

Hope to get to the first point by the end of the day if time permits, the second one I'm curious about but will see if there is a way to get around it.

Cheers,

Funs

rmonk, 2017-10-31:
As a data point, that method seemed to work fine.  I transferred ~ 170G of data off the iodrive with 0 issues.

Fett2, 2017-12-27:
rmonk, I am running into a similar issue as you. When you say 

> I added in the function definitions from elixir and got a little farther:

Can you tell me what functions you added, and to where?

snuf, 2017-12-28:
finally checked it in....

closed 2017-12-28 by snuf