RemixVSL/iomemory-vsl #6: Trouble compiling on 4.13.13-2

Fett2, 2017-12-27:
I'm trying to compile this in Proxmox 5.1, which is kernel 4.13.13-2-pve. I'm getting the following:

make[1]: Entering directory '/usr/src/linux-headers-4.13.13-2-pve'                                                                                                                                    CC [M]  /root/iomemorysource/root/usr/src/iomemory-vsl-3.2.15/kblock.o                                                                                                                            /root/iomemorysource/root/usr/src/iomemory-vsl-3.2.15/kblock.c: In function __kfio_bio_complete:                                                                                                  /root/iomemorysource/root/usr/src/iomemory-vsl-3.2.15/kblock.c:1024:8: error: struct bio has no member named bi_error; did you mean bi_iter?                                                       bio->bi_error = error;                                                                                                                                                                                 

^~                                                                                                                                                                                          /root/iomemorysource/root/usr/src/iomemory-vsl-3.2.15/kblock.c: In function kfio_make_request:                                                                                                    /root/iomemorysource/root/usr/src/iomemory-vsl-3.2.15/kblock.c:1778:5: error: implicit declaration of function blk_queue_bounce [-Werror=implicit-function-declaration]                                blk_queue_bounce(queue, &bio);   
                                                                                                                                                                   ^~~~~~~~~~~~~~~~      
                                                                                                                                                                         /root/iomemorysource/root/usr/src/iomemory-vsl-3.2.15/kblock.c: In function complete_list_entries:                                                                                                /root/iomemorysource/root/usr/src/iomemory-vsl-3.2.15/kblock.c:1945:20: error: struct bio has no member named bi_error; did you mean bi_iter?                                                          error = bio->bi_error;                                                                                                                                                                                         

^~                                                                                                                                                                              /root/iomemorysource/root/usr/src/iomemory-vsl-3.2.15/kblock.c: In function kfio_req_completor:                                                                                                   /root/iomemorysource/root/usr/src/iomemory-vsl-3.2.15/kblock.c:2229:8: error: struct bio has no member named bi_error; did you mean bi_iter?                                                       bio->bi_error = error;                                                                                                                                                                                 ^~

Fett2, 2017-12-27:
I noticed in the issue posted by rmonk he mentions changing the bi_error in kblock.c to bi_status. I've done that, and now the only error I get is:

make[1]: Entering directory '/usr/src/linux-headers-4.13.13-2-pve'
  CC [M]  /root/iomemory-vsl-master/root/usr/src/iomemory-vsl-3.2.15/kblock.o
/root/iomemory-vsl-master/root/usr/src/iomemory-vsl-3.2.15/kblock.c: In function ‘kfio_make_request’:
/root/iomemory-vsl-master/root/usr/src/iomemory-vsl-3.2.15/kblock.c:1778:5: error: implicit declaration of function ‘blk_queue_bounce’ [-Werror=implicit-function-declaration]
     blk_queue_bounce(queue, &bio);
     ^~~~~~~~~~~~~~~~

snuf, 2017-12-28:
Hi Fett2,

blk_queue_bounce is not in the kernel header files anymore, if you checkout master you should be able to compile and use. Has been simple fio tested against 4.13.13 and 4.14.9.

Cheers,

Fett2, 2017-12-28:
I was able to compile, thank you!

snuf, 2017-12-28:
welcome!

closed 2017-12-28 by snuf