RemixVSL/iomemory-vsl #8: References to missing `3.2.10` files during `dh_install`

missmah, 2018-02-16:
A small sampling of the files `dh_install` says are missing:

```
dh_install: iomemory-vsl-source missing files: usr/src/iomemory-vsl-3.2.10/katomic.c
dh_install: iomemory-vsl-source missing files: usr/src/iomemory-vsl-3.2.10/kbitops.c
dh_install: iomemory-vsl-source missing files: usr/src/iomemory-vsl-3.2.10/kblock.c
dh_install: iomemory-vsl-source missing files: usr/src/iomemory-vsl-3.2.10/kcache.c
dh_install: iomemory-vsl-source missing files: usr/src/iomemory-vsl-3.2.10/kcondvar.c
```

It then makes these complaints about `3.2.15` files:
```
dh_install: usr/src/iomemory-vsl-3.2.15/Kbuild exists in root but is not installed to anywhere
dh_install: usr/src/iomemory-vsl-3.2.15/cdev.c exists in root but is not installed to anywhere
dh_install: usr/src/iomemory-vsl-3.2.15/license.c exists in root but is not installed to anywhere
dh_install: usr/src/iomemory-vsl-3.2.15/kblock.c exists in root but is not installed to anywhere
dh_install: usr/src/iomemory-vsl-3.2.15/sysrq.c exists in root but is not installed to anywhere
dh_install: usr/src/iomemory-vsl-3.2.15/state.c exists in root but is not installed to anywhere
```

`grep` has been unable to help me figure out why it's still looking for the wrong files so far, though I'll keep investigating.

missmah, 2018-02-16:
`dpkg-buildpackage: source version 3.2.10.1509-1.0` is printed out early on, so despite `grep` not finding it, something somewhere must be still telling it to build `3.2.10`

missmah, 2018-02-16:
Looks like `changelog` is where it gets that from - not what I would have expected. I'll try changing it there and see if it builds properly.

snuf, 2018-02-16:
iomemory-vsl-source.install is the cullprit most likely, that's where it gets from which source files it needs to package.

missmah, 2018-02-16:
Got it, I had done a replace in that file for `3.2.15` and decided that it had not helped, but it was actually just that it did help, and there were a few files that were added since `3.2.10` and a few others which were removed, both of which needed to be manually accounted for. Now that I've updated that file and the `changelog`, it appears that I'm successful in building!

Unfortunately, now I'm having an issue with signing:
```
Finished running lintian.
Now signing changes and any dsc files...
 signfile iomemory-vsl_3.2.15.1700-1.0.dsc Support <support@fusionio.com>
gpg: keyring `/home/layla/.gnupg/secring.gpg' created
gpg: keyring `/home/layla/.gnupg/pubring.gpg' created
gpg: skipped "Support <support@fusionio.com>": secret key not available
gpg: /tmp/debsign.TVpTiy2k/iomemory-vsl_3.2.15.1700-1.0.dsc: clearsign failed: secret key not available
debsign: gpg error occurred!  Aborting....
debuild: fatal error at line 1295:
running debsign failed
```
I'll proceed to ask google about this one... if I get it all working, I'll submit another PR for the remainder of the `3.2.10` -> `3.2.15` fixes. I'll also likely submit a branch for my kernel version if it works, in-case anyone comes across and is afraid of trying a different version than their actual kernel version.

missmah, 2018-02-16:
It looks like the answers may be:
1. Use `debuild -us -uc` to make an unsigned package (not ideal)
2. Generate a key and use `debuild -kB57F5641` (replace `B57F5641` with your `keyid`)
3. Some other solution I haven't found yet, which there must be, since the instructions bundled with the original drivers don't specify anything about signing...

missmah, 2018-02-16:
Success! @snuf, thanks for the help!

missmah, 2018-02-16:
As mentioned, here's the PR for fixing this branch up to build `3.2.15` cleanly out of the box:
https://github.com/snuf/iomemory-vsl/pull/9

Also, the new branch for my kernel version is here, though I don't know how to make a PR for a new branch creation on your repo:
https://github.com/missmah/iomemory-vsl/tree/4.13.0-32

snuf, 2018-02-16:
Hi @missmah 

Finally had some free cycles to look a bit deeper! first of all thank you for contributing, it is highly appreciated! :)

Master seems to be already correct for building the deb packages, and was tested on 4.13 and 4.14 (also a bunch of other through some automated testing). I think I need to start cleaning the development forks out, and just have the test suite report in the README, that would clear things up nicely and prevent duplicate work :(

The container that does the module compile and deb package testing does the following:
RUN git clone https://github.com/snuf/iomemory-vsl
WORKDIR iomemory-vsl
RUN echo $iomemory_branch
RUN git checkout $iomemory_branch
RUN cd root && find usr/ -type f > ../debian/iomemory-vsl-source.install
RUN cd root && find usr/share/doc -type f > ../debian/iomemory-vsl.install
# kernel version, headers and branch will be the same
RUN deb_kernel_version=$iomemory_branch \
  KERNEL_SRC=/usr/src/linux-headers-$iomemory_branch \
  dpkg-buildpackage \
  -rfakeroot -b --no-check-builddeps -us -uc

and then uploads it to an S3 bucker. At the moment I'm in the middle of trying to make the entire build process "releasable", so the CI/CD will also be usable for other module builds.

Sorry you had to go through this, and I didn't look in depth sooner, because it could have saved you the work.

Cheers,

Funs

plappermaul, 2019-02-24:
close because auf inactivity

closed 2019-02-24 by plappermaul