RemixVSL/iomemory-vsl #11: Fails to build on 4.16.6-1-ARCH

benwis, 2018-05-05:
*/While I realize you do your testing on Ubuntu and with a lower kernel version, I was hoping you might be able to help me fix it so I can run it my IODrive II module on Arch Linux.

I've attached the make.log from the build. It's primary complaint appears to be this:make[2]: *** No rule to make target '/var/lib/dkms/iomemoryvsl/3.2.15/build/kfio/x86_64_cc73_libkfio.o', needed by '/var/lib/dkms/iomemory-vsl/3.2.15/build/iomemory-vsl.o'.  Stop.

Any suggestions?
[make.log](https://github.com/snuf/iomemory-vsl/files/1976361/make.log)

snuf, 2018-05-05:
Hey @benwis 

The problem here is a GCC "problem", or more a linking problem in conjunction with the GCC version you're running. It's not really a problem. I've already compiled on 4.16 before, but not with your GCC version. So the fix is copying the the lib that is in the named dir to your version that's named missing.

The lib actually makes the calls to the lower level sub systems. What I adapt is the interface to the kernel, as far as I can....

In your case, and mine, it means adding the x86_64_cc73_libkfio.o that is missing. Which basically means copying the last lib that is in the named directory to that specific name and the compile should work. I'll make sure to add this ver to the bunch.

Please let me know if this helps and works ?

Cheers,

Funs

snuf, 2018-05-05:
Line 47 of the Makefile shows the CCMAJOR and CCMINOR being used to determine the TARGET name, that is then used for the filename of the library to link against:

 47 TARGET ?= $(FIOARCH)_cc$(CCMAJOR)$(CCMINOR)
 48 
 49 KFIO_LIB ?= kfio/$(TARGET)_libkfio.o_shipped

benwis, 2018-05-05:
Thanks for the quick response. I've got a different error now after adding the cc73 object file to the kfio directory:
WARNING: could not find /var/lib/dkms/iomemory-vsl/3.2.15/build/kfio/.x86_64_cc73_libkfio.o.cmd for /var/lib/dkms/iomemory-vsl/3.2.15/build/kfio/x86_64_cc73_libkfio.o
FATAL: modpost: GPL-incompatible module iomemory-vsl.ko uses GPL-only symbol '__rcu_read_unlock'
make[2]: *** [scripts/Makefile.modpost:92: __modpost] Error 1
make[1]: *** [Makefile:1566: modules] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.16.6-1-ARCH/build'
make: *** [Makefile:83: modules] Error 2

Attached log: 
[make.log](https://github.com/snuf/iomemory-vsl/files/1976800/make.log)

benwis, 2018-05-05:
I was able to build and install on 4.16.6. However, to do so I had to change all of the License strings in these places to GPL. Not sure which is the primary culprit. It appears that Sandisk used some GPL connections to rcu_read_unlock.

/home/user/Downloads/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.15/Kbuild:
    9  
   10  
   11: $(src)/license.c: $(src)/Kbuild
   12: 	printf '#include "linux/module.h"\nMODULE_LICENSE("Proprietary");\n' >$@
   13  
   14  
   ..
   16  
   17  $(FIO_DRIVER_NAME)-y := main.o
   18: $(FIO_DRIVER_NAME)-y += license.o
   19  $(FIO_DRIVER_NAME)-y += pci.o
   20  $(FIO_DRIVER_NAME)-y += sysrq.o

/home/user/Downloads/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.15/kblock.c:
  641      /*
  642       * The following is used because access to udev events are restricted
  643:      * to GPL licensed symbols.  Since we have a propriatary licence, we can't
  644       * use the better interface.  Instead we must poll for the device creation.
  645       */

/home/user/Downloads/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.15/kfio_config.sh:
  504      local test_dir="${CONFIGDIR}/${kfioc_flag}"
  505      local result=0
  506:     local license="
  507: MODULE_LICENSE(\"Proprietary\");
  508  "
  509  
  ...
  514  
  515      if [ 1 -eq "$FUSION_DEBUG" ]; then
  516:         license="
  517: MODULE_LICENSE(\"GPL\");
  518  "
  519      fi
  520:     echo "$license" >> "$test_dir/kfioc_test.c"
  521  
  522      if [ 1 -eq "$VERBOSE" ]; then

/home/user/Downloads/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.15/license.c:
    1  #include "linux/module.h"
    2: MODULE_LICENSE("GPL");
    3

closed 2018-05-12 by benwis