RemixVSL/iomemory-vsl #12: gcc 7.3

holger24, 2018-05-25:
Hello,

first, many thanks for providing this driver!

With gcc 7.3 this does not compile:

make[2]: *** No rule to make target '/usr/src/fio/iomemory-vsl-3.2.15/kfio/x86_64_cc73_libkfio.o', needed by '/usr/src/fio/iomemory-vsl-3.2.15/iomemory-vsl.o'.  Stop.

How or where can one get this x86_64_cc73_libkfio.o?

Regards,
Holger

benwis, 2018-05-27:
Hi Holger24,
See this issue: https://github.com/snuf/iomemory-vsl/issues/11

Short version is you can go into iomemory-vsl/root/user/srciomemory-vsl-3.12.15/kfio and copy  x86_64_cc72_libkfio.o_shipped to whatever version you need. I needed 81 so I did  x86_64_cc81_libkfio.o_shipped

snuf, 2018-05-28:
Hi @holger24 

@benwis also ran into this, I've had a fix staged for a while, but didn't commit it. It's in master now. The fix copies the last version to the version of GCC that is used.

@benwis, to your earlier #11 issue, I've added the "gpl" flag which swaps out the license *cough*. Also thanks for helping @holger24!

Cheers,

Funs

closed 2018-05-28 by snuf

benwis, 2018-06-01:
Thanks snuf!