RemixVSL/iomemory-vsl #13: Fails to build on centos 7.5/3.10.0-862.3.2.el7.x86_64

alexey2k, 2018-05-30:
Trying to compile driver for the latest centos 7.5 got following:

/root/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c: In function 'kfio_elevator_change':
/root/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:2075:5: warning: passing argument 1 of 'elevator_exit' from incompatible pointer type [enabled by default]
     elevator_exit(q->elevator);
     ^
In file included from include/linux/blkdev.h:262:0,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/port-internal.h:62,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:58:
include/linux/elevator.h:257:13: note: expected 'struct request_queue *' but argument is of type 'struct elevator_queue *'
 extern void elevator_exit(struct request_queue *, struct elevator_queue *);
             ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:2075:5: error: too few arguments to function 'elevator_exit'
     elevator_exit(q->elevator);
     ^
In file included from include/linux/blkdev.h:262:0,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/port-internal.h:62,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:58:
include/linux/elevator.h:257:13: note: declared here
 extern void elevator_exit(struct request_queue *, struct elevator_queue *);
             ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/ktime.c:156:13: warning: 'fusion_timer_callback' defined but not used [-Wunused-function]
 static void fusion_timer_callback(struct timer_list *t) {

Any suggestion?

Thank you in advance.

alexey2k, 2018-05-31:
Ok found fix for the issue:
https://forums.servethehome.com/index.php?threads/centos-7-fusionio-users-do-not-upgrade-to-kernel-3-10-0-862-2-3-el7-yet.19760/#post-192219

closed 2018-05-31 by alexey2k

snuf, 2018-06-01:
Hi Alexey2k,

Interestingly enough the fix is in the code, but it's tied to a kernel version and not a nice check:
2071 #if !defined(__VMKLNX__)
2072 # if LINUX_VERSION_CODE >= KERNEL_VERSION(4,11,0)
2073     elevator_exit(q, q->elevator);
2074 # else
2075     elevator_exit(q->elevator);
2076 # endif

I'll update to make it a nice check so it doesn't use the kernel version but a check to see what elevator_exit looks like.

Thanks for opening an issue!

Cheers,

Funs