RemixVSL/iomemory-vsl #16: Silent io drops

plappermaul, 2018-07-23:
Hello,

iomemory-vsl silently drops io operations (I observed reads at least). Situation is as follows:

- Linux can issue io request of size up to /sys/block/<device>/queue/max_sectors_kb size.
- This limit has raised to 1MB in /include/linux/blkdev.h with Parameter BLK_DEF_MAX_SECTORS since version 4.x 
- iomemory-vsl has some (yet unknown) limit of 257 blocks it can process per request via scatter gather lists.
- if a single requests exceeds 257*4K kfio_sgl_map_bio() will return -EINVAL
- The caller kfio_map_to_fbio() will drop the bio afterwards.

An easy reproducer of the error should be to increase max_sectors_kb of fio device to 1024KB. Afterwards read some MB with dd from same device location twice into two files. The content will be different.

`# echo 1024  > /sys/block/fioa/queue/max_sectors_kb
# ./run_check.sh
read 2 times 256K from /dev/fioa1, checksums are
16171   256
16171   256
read 2 times 2M from /dev/fioa1, checksums are
17189  2048
28665  2048
# echo 256  > /sys/block/fioa/queue/max_sectors_kb
# ./run_check.sh
read 2 times 256K from /dev/fioa1, checksums are
16171   256
16171   256
read 2 times 2M from /dev/fioa1, checksums are
54935  2048
54935  2048
`
Remark: During my debugging session I noticed that iomemory-vsl sometimes issues twice the maximum request size. Not yet any idea where this comes from.

plappermaul, 2018-07-24:
There seem to be several issues.

1. Kernel default for BIO_MAX_PAGES is > 256 since [kernel 4.14](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/include/linux/bio.h?h=v4.18-rc6&id=225311a46411c37e20e73d99f4382f141e12f6f9) if you compile with CONFIG_THP_SWAP. 

2. Doing normal read IO on the device will kick in kernel [readahead](https://elixir.bootlin.com/linux/latest/source/mm/readahead.c). It may raise the requested bytes by up to 2M.

3. DirectIO analysis yet to come ...

4. 257 block limit may be caused by the pre compiled object file

plappermaul, 2018-07-25:
Reminder for myself:

DirectIO sizes must be fixed in fs/block_dev.c

__blkdev_direct_IO()
```
struct backing_dev_info *bdi = bdev->bd_bdi;
int max_pages;
max_pages = min_t(unsigned long, bdi->io_pages , BIO_MAX_PAGES);
...
nr_pages = iov_iter_npages(iter, max_pages);
```

blkdev_direct_IO()
```
struct file *file = iocb->ki_filp;
struct inode *inode = bdev_file_inode(file);
struct block_device *bdev = I_BDEV(inode);
struct backing_dev_info *bdi = bdev->bd_bdi;
int nr_pages;
int max_pages;

max_pages = min_t(unsigned long, bdi->io_pages, BIO_MAX_PAGES );
nr_pages = iov_iter_npages(iter, max_pages + 1);
if (!nr_pages)
  return 0;
if (is_sync_kiocb(iocb) && nr_pages <= max_pages)
  return __blkdev_direct_IO_simple(iocb, iter, nr_pages);
return __blkdev_direct_IO(iocb, iter, min(nr_pages, max_pages));
```

bplein, 2018-07-25:
The most recent 3.2.15 VSL for Debian compiles fine on a default Ubuntu 16.04 installation (I found this issue here due to my testing today with Ubuntu). 

I'm not sure this repo is necessary anymore with Ubuntu 16.04. Can you reproduce with original Fusion-io/SanDisk drivers?

plappermaul, 2018-07-26:
My issues occur because I'm currently running CentOS 7.5 with LTS Kernel 4.14.xx. I gave different driver versions a try but in 4.1x a lot of things changed and only this repo takes care for newer kernels. 

If you want to check if your setup has errors too just insert the following lines into the code

```
            /* The list will grow: check for overflow here. */
            if (unlikely(lsg->num_entries >= lsg->max_entries))
            {
/* give a hint about dropped I/Os */
                infprint("------------------------------------------------------\n");
                infprint("list overflow (%u >= %u) in kfio_sgl_map_bio\n",lsg->num_entries,lsg->max_entries);
                dump_stack();

                rval = -EINVAL;
                break;
            }
```

plappermaul, 2018-07-26:
Readahead in kernel only takes lazy care about iosizes. Fix like this:

```
static unsigned long get_init_ra_size(unsigned long size, unsigned long max)
{
  unsigned long newsize = roundup_pow_of_two(size);

  if (newsize <= max / 32)
    newsize = newsize * 4;
  else if (newsize <= max / 4)
    newsize = newsize * 2;
  else
    newsize = max / 2; /* Fix. Might be doubled later on ... */
  return newsize;
}
```

plappermaul, 2018-07-26:
Status for now

readahead -> ok, fixed (see above)
directio -> ok, fixed (see above)

dmesg is clean after initial tests.

plappermaul, 2018-07-26:
Hi I have a pull ready for the dump_stack() logging feature.

Maybe someone can allow me to send pull requests.

Thanks in advance

snuf, 2018-07-26:
@plappermaul 

Sorry for not responding earlier, have been a bit busy the past months. 

Great work!!

I think you should be able to just do a pull request, there is no reason you should not be able to, how are you trying to?

Cheers,

Funs

plappermaul, 2018-07-26:
Just created a branch and wanted to upload it.

# git push origin enable_debug
Username for 'https://github.com': ...
Password for '...':
remote: Permission to snuf/iomemory-vsl.git denied to plappermaul.
fatal: unable to access 'https://github.com/snuf/iomemory-vsl/': The requested URL returned error: 403

Maybe I must be invited as coworker=

plappermaul, 2018-07-26:
[Patch](https://marc.info/?l=linux-block&m=153263375303406&w=2) for readahead sent to linux-block

plappermaul, 2018-07-27:
257 segment limit is really inside the object file. At least one other caller has been identified who wants to allocate more than 257 pages. Don't know yet if it is better to patch diect IO in the kernel or to hack the lob.

```
0000000000090560 <somewhere_in_lob>:

...

905c1:  8b 83 c8 02 00 00       mov    0x2c8(%rbx),%eax   // memory = sgl.memory;

905c7:  ba 01 01 00 00          mov    $0x101,%edx        // max_nvec = 257;

905cc:  48 89 df                mov    %rbx,%rdi

905cf:  48 05 ff 0f 00 00       add    $0xfff,%rax        // memory = memory + (PAGE_SIZE - 1);

905d5:  48 c1 e8 0c             shr    $0xc,%rax          // nvec = memory >> 12;

905d9:  48 83 c0 02             add    $0x2,%rax          // nvec = nvec + 2;

905dd:  48 3d 01 01 00 00       cmp    $0x101,%rax        // if (nvec > 257)

905e3:  48 0f 42 c2             cmovb  %rdx,%rax          //   nvec = max_nvec;

905e7:  89 83 50 03 00 00       mov    %eax,0x350(%rbx)   // sgl.nvec = nvec;

...

90616:  48 89 df                mov    %rbx,%rdi          // call_out_sgl = sgl;

90619:  45 8d 7c 24 01          lea    0x1(%r12),%r15d

9061e:  48 89 51 08             mov    %rdx,0x8(%rcx)

90622:  48 89 88 f0 05 00 00    mov    %rcx,0x5f0(%rax)

90629:  4c 89 a8 f8 05 00 00    mov    %r13,0x5f8(%rax)

90630:  83 83 50 0f 00 00 01    addl   $0x1,0xf50(%rbx)

90637:  48 89 93 40 0f 00 00    mov    %rdx,0xf40(%rbx)

9063e:  e8 00 00 00 00          callq  sgl_handler_1

...

 

0000000000041fc0 <sgl_handler_1>:

41fc0:  55                      push   %rbp

41fc1:  be 01 00 00 00          mov    $0x1,%esi

41fc6:  48 89 e5                mov    %rsp,%rbp

41fc9:  41 54                   push   %r12

41fcb:  49 89 fc                mov    %rdi,%r12          // sgl = call_in_sgl;

41fce:  53                      push   %rbx

41fcf:  48 8b 87 68 02 00 00    mov    0x268(%rdi),%rax

41fd6:  48 c7 c7 00 00 00 00    mov    $0x0,%rdi

41fd9: R_X86_64_32S     .bss+0xacc0

41fdd:  8b 90 ac 03 00 00       mov    0x3ac(%rax),%edx

41fe3:  e8 00 00 00 00          callq  kfio_cache_alloc_nod

41fe8:  48 85 c0                test   %rax,%rax

41feb:  48 89 c3                mov    %rax,%rbx

41fee:  74 48                   je     42038 <sgl_handler_1+0x78>

41ff0:  31 f6                   xor    %esi,%esi

41ff2:  ba 28 01 00 00          mov    $0x128,%edx

41ff7:  48 89 c7                mov    %rax,%rdi

41ffa:  e8 00 00 00 00          callq  kfio_memset-0x4

41fff:  49 8b 84 24 68 02 00    mov    0x268(%r12),%rax

42006:  00

42007:  41 8b 8c 24 50 03 00    mov    0x350(%r12),%ecx    // call_out_nvec = sgl.nvec;

4200e:  00

4200f:  48 8d 53 10             lea    0x10(%rbx),%rdx

42013:  8b b0 ac 03 00 00       mov    0x3ac(%rax),%esi

42019:  48 8b b8 38 01 00 00    mov    0x138(%rax),%rdi

42020:  e8 00 00 00 00          callq  kfio_sgl_alloc_nvec

42025:  85 c0                   test   %eax,%eax

42027:  48 89 de                mov    %rbx,%rsi

4202a:  75 14                   jne    42040 <sgl_handler_1+0x80>

4202c:  5b                      pop    %rbx

4202d:  41 5c                   pop    %r12

4202f:  48 89 f0                mov    %rsi,%rax

42032:  5d                      pop    %rbp

42033:  c3                      retq

42034:  0f 1f 40 00             nopl   0x0(%rax)

42038:  31 f6                   xor    %esi,%esi

4203a:  eb f0                   jmp    4202c <sgl_handler_1+0x6c>

4203c:  0f 1f 40 00             nopl   0x0(%rax)

42040:  48 c7 c7 00 00 00 00    mov    $0x0,%rdi

42043: R_X86_64_32S     .bss+0xacc0

42047:  e8 00 00 00 00          callq  kfio_cache_free-0x4

4204c:  31 f6                   xor    %esi,%esi

4204e:  eb dc                   jmp    4202c <sgl_handler_1+0x6c>

 

00000000000b8120 <somewhere_else_in_lob>:

...

b821a:  49 8b 04 24             mov    (%r12),%rax

b821e:  b9 40 04 00 00          mov    $0x440,%ecx         // call_out_nvec = 1088

b8223:  48 8b 90 68 02 00 00    mov    0x268(%rax),%rdx

b822a:  8b b2 ac 03 00 00       mov    0x3ac(%rdx),%esi

b8230:  48 8b ba 38 01 00 00    mov    0x138(%rdx),%rdi

b8237:  48 89 da                mov    %rbx,%rdx

b823a:  e8 00 00 00 00          callq  kfio_sgl_alloc_nvec

b823f:  85 c0                   test   %eax,%eax

b8241:  41 89 c7                mov    %eax,%r15d

```

bplein, 2018-07-27:
@plappermaul would you be willing to share your test script? I'd like to do some testing on my configuration as well.

plappermaul, 2018-07-27:
Bingo: Patch for readahead has been [accepted](https://marc.info/?l=linux-block&m=153270411821114&w=2).

bplein, 2018-07-27:
congrats!

I know Jens Axboe (not well), I met him and conversed with him quite a bit when I worked at Fusion-io with him. 

He knows these cards well :). And I don't know if anyone knows the Linux block system better than he does.

plappermaul, 2018-07-27:
I'm running CEPH und use IODrives as WAL and RocksDB. Just raw devices. This setup was totally messed up. More [here](https://tracker.ceph.com/issues/24968). I could not believe that my data got totally corrupted with standard setup. 

To make a long story short. After finding the culprit I just put the dump_stack() in and identified the callers.

bplein, 2018-07-27:
Great debugging work.  

Sorry for your [data] loss, however.

By the way, it's been a few years since I was running ioDrives on Linux (I have been using them under VMware ESXi).  Question for you:

If I look at the scheduler settings (back in the day we recommended changing them to 'noop'), they are "none" which is new to me. 
`# cat /sys/block/fiob/queue/scheduler
none`

Is this normal and correct?

plappermaul, 2018-07-27:
Regarding max_sectors_kb. This does not help for directIO (hardcoded to BIO_MAX_PAGES). 

Regarding scheduler: I would say none is good. Reordering requests for fast devices just adds additional CPU (in my opinion).

snuf, 2018-07-27:
@plappermaul I've added you as a collaborator, normally people fork and then push to their fork, and create a pull request from there.

bplein, 2018-07-27:
Thanks @plappermaul. Makes total sense with regards to directIO. I'll stick with ZFS for the time being and that will prevent directIO.  

I've just never seen "none" as an option. I guess that is "new" with respect to when I worked at Fusion-io (2009-2012) and I've not been using Linux with these much since then. We were strong proponents of noop.

plappermaul, 2018-07-28:
direct IO [patches](https://marc.info/?l=linux-block&m=153280543710680&w=2) sent.

plappermaul, 2018-07-29:
So we go ... 

First pull request is [online](https://github.com/snuf/iomemory-vsl/pull/17).

plappermaul, 2018-07-31:
Hi there,

I created the new **next_generation** branch. For the first time in 4 weeks I was able to unload the driver without a crash. IOs work as expected. If you have some spare time you could do me a favor and do some basic tests.

plappermaul, 2018-08-01:
Close the issue after driver rewrite.

closed 2018-08-01 by plappermaul