RemixVSL/iomemory-vsl #38: Update to upstream 3.2.16?

fake-name, 2019-02-22:
So, surprisingly enough, WD has apparently released a update for the drivers on Sept 2018, version 3.2.16.1731. 

It doesn't show up for all distros, but if you pick Ubuntu 16.04, you can view it, or I'll attach the source tarbar here.

[iomemory-vsl_3.2.16.1731-1.0.tar.gz](https://github.com/snuf/iomemory-vsl/files/2892853/iomemory-vsl_3.2.16.1731-1.0.tar.gz)

Diffing the two source trees (3.2.15 from this repo and the distributed 3.2.16), the changes are pretty significant, a significant number of the source files have non-trivial changes. 

I dunno if it'd be worth trying to rebase on the newer upstream.

plappermaul, 2019-02-22:
The iomemory-vsl from this repo is far from being the standard 3.2.x version. That means

- A lot of features of 4.3.x have been backported
- Source files have been rearranged to give only a minimal difference to 4.3.x

In my opinion the 3.2.16 release has no major differences to 3.2.15. Thus we should still be a lot better. If you have time please do a standard 3.2.15 to 3.2.16 diff. If something major has changed we can include it.

Thanks in advance.

charlesnadeau, 2019-02-22:
I tried to build iomemory-vsl_3.2.16.1731-1.0.tar.gz (above) under Ubuntu 18.04.02 (kernel 4.15.0-45-generic) and I am getting these errors:

`make[2]: *** No rule to make target '/var/lib/dkms/iomemory-vsl/3.2.16/build/kfio/x86_64_cc73_libkfio.o', needed by '/var/lib/dkms/iomemory-vsl/3.2.16/build/iomemory-vsl.o'.  Stop.
make[2]: *** Waiting for unfinished jobs....
  CC [M]  /var/lib/dkms/iomemory-vsl/3.2.16/build/kcsr.o
Makefile:1551: recipe for target '_module_/var/lib/dkms/iomemory-vsl/3.2.16/build' failed
make[1]: *** [_module_/var/lib/dkms/iomemory-vsl/3.2.16/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-45-generic'
Makefile:82: recipe for target 'modules' failed
make: *** [modules] Error 2`

I am not a C programmer so i don't really know how to fix it.

I am trying to build this version of the drivers because IOMemory-VSL next_generation compiles but cause my server to go into kernel panic about 10 seconds after the Linux prompt appears. Here is what is displayed about the FusionIO card while booting:

Feb 21 15:58:23 hpdl380g72 kernel: [   29.880380] <6>fioinf VSL configuration hash: 50cc3bdba9fe52b90d1821e59d81452e4a6eac09
Feb 21 15:58:23 hpdl380g72 kernel: [   29.888055] <6>fioinf 
Feb 21 15:58:23 hpdl380g72 kernel: [   29.895373] <6>fioinf Copyright (c) 2006-2014 Fusion-io, Inc. (acquired by SanDisk Corp. 2014)
Feb 21 15:58:23 hpdl380g72 kernel: [   29.902887] <6>fioinf Copyright (c) 2014-2016 SanDisk Corp. and/or all its affiliates. All rights reserved.
Feb 21 15:58:23 hpdl380g72 kernel: [   29.910509] <6>fioinf For Terms and Conditions see the License file included
Feb 21 15:58:23 hpdl380g72 kernel: [   29.918168] <6>fioinf with this driver package.
Feb 21 15:58:23 hpdl380g72 kernel: [   29.925833] <6>fioinf 
Feb 21 15:58:23 hpdl380g72 kernel: [   29.933411] <6>fioinf ioDrive driver 3.2.15.1700 pinnacles@3dd0050df54c loading...
Feb 21 15:58:23 hpdl380g72 kernel: [   29.947670] <6>fioinf ioDrive 0000:0e:00.0: mapping controller on BAR 5
Feb 21 15:58:23 hpdl380g72 kernel: [   29.947755] <6>fioinf ioDrive 0000:0e:00.0: MSI enabled
Feb 21 15:58:23 hpdl380g72 kernel: [   29.947758] <6>fioinf ioDrive 0000:0e:00.0: using MSI interrupts
Feb 21 15:58:23 hpdl380g72 kernel: [   29.977694] <6>fioinf ioDrive 0000:0e:00.0.0: Starting master controller
Feb 21 15:58:23 hpdl380g72 kernel: [   30.854956] <6>fioinf ioDrive 0000:0e:00.0.0: Adapter serial number is 128225
Feb 21 15:58:23 hpdl380g72 kernel: [   31.734017] <6>fioinf ioDrive 0000:0e:00.0.0: Board serial number is 112675
Feb 21 15:58:23 hpdl380g72 kernel: [   31.743228] <6>fioinf ioDrive 0000:0e:00.0.0: Default capacity        320.000 GBytes
Feb 21 15:58:23 hpdl380g72 kernel: [   31.752427] <6>fioinf ioDrive 0000:0e:00.0.0: Default sector size     512 bytes
Feb 21 15:58:23 hpdl380g72 kernel: [   31.761566] <6>fioinf ioDrive 0000:0e:00.0.0: Rated endurance         4.00 PBytes
Feb 21 15:58:23 hpdl380g72 kernel: [   31.770674] <6>fioinf ioDrive 0000:0e:00.0.0: 85C temp range hardware found
Feb 21 15:58:23 hpdl380g72 kernel: [   31.784295] <6>fioinf ioDrive 0000:0e:00.0.0: Firmware version 7.1.17 116786 (0x700411 0x1c832)
Feb 21 15:58:23 hpdl380g72 kernel: [   31.793356] <6>fioinf ioDrive 0000:0e:00.0.0: Platform version 10 
Feb 21 15:58:23 hpdl380g72 kernel: [   31.802266] <6>fioinf ioDrive 0000:0e:00.0.0: Firmware VCS version 116786 [0x1c832]
Feb 21 15:58:23 hpdl380g72 kernel: [   31.811227] <6>fioinf ioDrive 0000:0e:00.0.0: Firmware VCS uid 0xaeb15671994a45642f91efbb214fa428e4245f8a
Feb 21 15:58:23 hpdl380g72 kernel: [   31.823251] <6>fioinf ioDrive 0000:0e:00.0.0: Powercut flush: Enabled
Feb 21 15:58:23 hpdl380g72 kernel: [   31.937074] <6>fioinf ioDrive 0000:0e:00.0.0: PCIe power monitor enabled (master). Limit set to 24.750 watts.
Feb 21 15:58:23 hpdl380g72 kernel: [   31.946078] <6>fioinf ioDrive 0000:0e:00.0.0: Thermal monitoring: Enabled
Feb 21 15:58:23 hpdl380g72 kernel: [   31.954875] <6>fioinf ioDrive 0000:0e:00.0.0: Hardware temperature alarm set for 85C.
Feb 21 15:58:23 hpdl380g72 kernel: [   31.975281] <6>fioinf ioDrive 0000:0e:00.0: Found device fct0 (HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:0e:00.0) on pipeline 0
Feb 21 15:58:23 hpdl380g72 kernel: [   32.001637] <6>fioinf HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:0e:00.0: Upgrade: Detected 11bit ECC media via probe of EB 0.
Feb 21 15:58:23 hpdl380g72 kernel: [   32.025300] <6>fioinf HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:0e:00.0: Upgrade: Found old LEB map revision 6 in EB 0.
Feb 21 15:58:23 hpdl380g72 kernel: [   32.043358] <6>fioinf HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:0e:00.0: entering raw mode for LEB map recovery. This may take several minutes.
Feb 21 15:58:23 hpdl380g72 kernel: [   32.062204] <6>fioinf HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:0e:00.0: reading old LEB map page 1/512...
Feb 21 15:58:23 hpdl380g72 kernel: [   32.283686] <6>fioinf HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:0e:00.0: reading old LEB map page 2/512...
Feb 21 15:58:23 hpdl380g72 kernel: [   32.505177] <6>fioinf HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:0e:00.0: reading old LEB map page 3/512...
Feb 21 15:58:23 hpdl380g72 kernel: [   32.724546] <6>fioinf HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:0e:00.0: reading old LEB map page 4/512...
[... cut ...]
Feb 21 15:58:23 hpdl380g72 kernel: [   97.124996] <6>fioinf HP 640GB MLC PCIe ioDrive Duo for ProLiant Servers 0000:0f:00.0: Attaching explicitly disabled

My card firmware is 3.2.14 (I upgraded it from 2.2.3 using a Centos 6 LiveCD) so I thought IOMemory-VSL next_generation would work but it doesn't.
Based on what is displayed at boot (above), is there something wrong with my card? Anything else I should do to "prepare" my card? What are my drivers options uder Ubuntu 18.04.2.
Thank you very much for your help.

Charles Nadeau

fake-name, 2019-02-23:
> The iomemory-vsl from this repo is far from being the standard 3.2.x version. That means
>
> * A lot of features of 4.3.x have been backported
> * Source files have been rearranged to give only a minimal [...]

I didn't realize 4.x was different. 

Anyways, I wasn't sure if you folks were aware they shipped a new version, and figured I'd give you a poke and let you know. 

> I tried to build iomemory-vsl_3.2.16.1731-1.0.tar.gz (above) under Ubuntu 18.04.02 (kernel 4.15.0-45-generic) and I am getting these errors:
>
> ```
> make[2]: *** No rule to make target [...]

Uh, `3.2.16` doesn't support 18.04?

closed 2019-02-23 by fake-name

fake-name, 2019-02-23:
Ok, one thing possibly of note: If you have the `3.2.16` client utilities installed, they fail to recognize that the driver is loaded:

```

durr@xenbox:/media/Storage/IODrive/3.2.16$ sudo fio-status

Found 2 ioMemory devices in this system
Driver version: Driver not loaded

Adapter: Single Controller Adapter
        Fusion-io ioDrive2 1.205TB, Product Number:F00-001-1T20-CS-0001, SN:1213D1652, FIO SN:1213D1652
        External Power: NOT connected
        PCIe Power limit threshold: Disabled
        Connected ioMemory modules:
          03:00.0:      Product Number:F00-001-1T20-CS-0001, SN:1213D1652

03:00.0 ioDrive2 Adapter Controller, Product Number:F00-001-1T20-CS-0001, SN:1213D1652
        PCI:03:00.0
        Firmware v7.1.17, rev 116786 Public
        Internal temperature: 43.31 degC, max 44.30 degC

Adapter: Single Controller Adapter
        Fusion-io ioDrive2 1.205TB, Product Number:F00-001-1T20-CS-0001, SN:1213D1984, FIO SN:1213D1984
        External Power: NOT connected
        PCIe Power limit threshold: Disabled
        Connected ioMemory modules:
          04:00.0:      Product Number:F00-001-1T20-CS-0001, SN:1213D1984

04:00.0 ioDrive2 Adapter Controller, Product Number:F00-001-1T20-CS-0001, SN:1213D1984
        PCI:04:00.0
        Firmware v7.1.17, rev 116786 Public
        Internal temperature: 43.31 degC, max 43.80 degC

durr@xenbox:/media/Storage/IODrive/3.2.16$ sudo dpkg -r fio-common fio-firmware-fusion fio-preinstall fio-sysvinit fio-util
dpkg: warning: ignoring request to remove fio-firmware-fusion which isn't installed
(Reading database ... 153751 files and directories currently installed.)
Removing fio-preinstall (3.2.16.1731-1.0) ...
Removing fio-sysvinit (3.2.16.1731-1.0) ...
Removing fio-util (3.2.16.1731-1.0) ...
Removing fio-common (3.2.16.1731-1.0) ...
dpkg: warning: while removing fio-common, directory '/usr/lib/fio' not empty so not removed
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
durr@xenbox:/media/Storage/IODrive/3.2.16$ cd ../3.2.15
durr@xenbox:/media/Storage/IODrive/3.2.15$ sudo dpkg -i *.deb
Selecting previously unselected package fio-common.
(Reading database ... 153697 files and directories currently installed.)
Preparing to unpack fio-common_3.2.15.1699-1.0_amd64.deb ...
Unpacking fio-common (3.2.15.1699-1.0) ...
Selecting previously unselected package fio-preinstall.
Preparing to unpack fio-preinstall_3.2.15.1699-1.0_amd64.deb ...
Unpacking fio-preinstall (3.2.15.1699-1.0) ...
Selecting previously unselected package fio-sysvinit.
Preparing to unpack fio-sysvinit_3.2.15.1699-1.0_all.deb ...
Unpacking fio-sysvinit (3.2.15.1699-1.0) ...
Selecting previously unselected package fio-util.
Preparing to unpack fio-util_3.2.15.1699-1.0_amd64.deb ...
Unpacking fio-util (3.2.15.1699-1.0) ...
Setting up fio-common (3.2.15.1699-1.0) ...
Setting up fio-preinstall (3.2.15.1699-1.0) ...
Setting up fio-util (3.2.15.1699-1.0) ...
Setting up fio-sysvinit (3.2.15.1699-1.0) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Processing triggers for systemd (237-3ubuntu10.13) ...
Processing triggers for ureadahead (0.100.0-20) ...
durr@xenbox:/media/Storage/IODrive/3.2.15$ sudo fio-status

Found 2 ioMemory devices in this system
Driver version: 3.2.15 build 1700

Adapter: Single Controller Adapter
        Fusion-io ioDrive2 1.205TB, Product Number:F00-001-1T20-CS-0001, SN:1213D1652, FIO SN:1213D1652
        External Power: NOT connected
        PCIe Power limit threshold: 24.75W
        Connected ioMemory modules:
          fct0: Product Number:F00-001-1T20-CS-0001, SN:1213D1652

fct0    Attached
        ioDrive2 Adapter Controller, Product Number:F00-001-1T20-CS-0001, SN:1213D1652
        Located in slot 0 Center of ioDrive2 Adapter Controller SN:1213D1652
        PCI:03:00.0, Slot Number:4
        Firmware v7.1.17, rev 116786 Public
        1205.00 GBytes device size
        Internal temperature: 43.80 degC, max 44.30 degC
        Reserve space status: Healthy; Reserves: 100.00%, warn at 10.00%
        Contained VSUs:
          fioa: ID:0, UUID:5b545260-5e5d-4ebd-a12c-6477c29d232c

fioa    State: Online, Type: block device
        ID:0, UUID:5b545260-5e5d-4ebd-a12c-6477c29d232c
        1205.00 GBytes device size

Adapter: Single Controller Adapter
        Fusion-io ioDrive2 1.205TB, Product Number:F00-001-1T20-CS-0001, SN:1213D1984, FIO SN:1213D1984
        External Power: NOT connected
        PCIe Power limit threshold: 24.75W
        Connected ioMemory modules:
          fct1: Product Number:F00-001-1T20-CS-0001, SN:1213D1984

fct1    Attached
        ioDrive2 Adapter Controller, Product Number:F00-001-1T20-CS-0001, SN:1213D1984
        Located in slot 0 Center of ioDrive2 Adapter Controller SN:1213D1984
        PCI:04:00.0, Slot Number:2
        Firmware v7.1.17, rev 116786 Public
        1205.00 GBytes device size
        Internal temperature: 43.31 degC, max 43.80 degC
        Reserve space status: Healthy; Reserves: 100.00%, warn at 10.00%
        Contained VSUs:
          fiob: ID:0, UUID:5cc51c17-ad5e-4501-ba0a-77664c7ccdc6

fiob    State: Online, Type: block device
        ID:0, UUID:5cc51c17-ad5e-4501-ba0a-77664c7ccdc6
        1205.00 GBytes device size

```

charlesnadeau, 2019-02-23:
I think I found the problem: [I need to reformat the drive after upgrading the firmware](https://forums.servethehome.com/index.php?threads/hp-iodrive-1-on-linux.23564/#post-219385) from 2.2.3 and 3.2.14. I will try to re-format it using 3.2.14 utils then upgrade the drivers and reboot. I'll do it Monday and will report back here.
Thanks!

Charles

plappermaul, 2019-02-23:
Thanks for the feedback.

I guess reformatting will fix the crash. Nevertheless you could check, why 3.2.16 tools do not recognize the 3.2.15 driver.

fake-name, 2019-02-24:
> I guess reformatting will fix the crash. 

I think @charlesnadeau  should have opened a different issue, I'm not sure why they're posting unrelated issues here. 

> Nevertheless you could check, why 3.2.16 tools do not recognize the 3.2.15 driver.

I'm happy to help, if you can tell me what I should look at. I basically had the 3.2.16 tools installed, they didn't work, and I rolled back and forth a few times to confirm it was a version issue.

plappermaul, 2019-02-24:
Will open a new issue for that

charlesnadeau, 2019-02-26:
After a couple of days of troubleshooting, I managed to find the problem: it was a bad module (hpwdt) triggering a kernel panic upon the discovery of the FusionIO card at boot time. Once I blacklisted this module, i managed to build and load the next_generation version of the driver. Thanks everyone for your help.

Charles