RemixVSL/iomemory-vsl #39: Testing for 3.2.16 version bump

plappermaul, 2019-02-24:
HI,

I just aligned the code with the 3.2.16 branch. Compilation works fine but I have no time to test myself if someone is interested he/she should give it a go:

git clone https://github.com/snuf/iomemory-vsl
cd iomemory-vsl/	cd iomemory-vsl/
git checkout next_generation	
rpmbuild -ba fio-driver.spec

Please report your experience back in this thread.

Thanks in advance.

missmah, 2019-06-28:
Not working properly on `Ubuntu 18.04 LTS`:

After making changes to make `3.2.16` build (the build attempts to build `3.2.15` by default), signing and installing the `.deb` packages, and setting up `DKMS`, I get the following:

```
Found 2 ioMemory devices in this system
Driver version: Driver not loaded

43:00.0
        PCI:43:00.0
        Firmware v7.1.17, rev 116786 Public
        Internal temperature: 63.00 degC, max 63.49 degC

44:00.0
        PCI:44:00.0
        Firmware v7.1.17, rev 116786 Public
        Internal temperature: 59.55 degC, max 60.54 degC
```

Prior to adding `3.2.16`, the output was this:
```
Found 2 ioMemory devices in this system with 1 ioDrive Duo
Driver version: Driver not loaded

Adapter: Dual Adapter
        HP 1.28TB MLC PCIe IO Accelerator, Product Number:641027-B21, SN:415165
        PCIe Power limit threshold: Disabled
        Connected ioMemory modules:
          43:00.0:      Product Number:641027-B21, SN:413412
          44:00.0:      Product Number:641027-B21, SN:413421

43:00.0 ioDIMM 640, SN:413412
        Located in slot 1 Lower of ioDrive Duo HL SN:415165
        PCI:43:00.0
        Firmware v7.1.17, rev 116786 Public
        Internal temperature: 62.51 degC, max 62.51 degC

44:00.0 ioDIMM 640, SN:413421
        Located in slot 0 Upper of ioDrive Duo HL SN:415165
        PCI:44:00.0
        Firmware v7.1.17, rev 116786 Public
        Internal temperature: 59.06 degC, max 59.55 degC
```

So, something has gone wrong with the new version to still have `driver not loaded`, but also somehow cause other information that previously was displayed to now fail to display. Though, I suppose another option for why there is less information is that maybe `fio-status` updated?

Either way, the `3.2.16` driver isn't being properly recognized/utilized.

missmah, 2019-06-28:
Interesting data point, after reboot, each call to `fio-status` returns slightly different results:

```
Found 2 ioMemory devices in this system with 1 ioDrive Duo
Driver version: Driver not loaded

Adapter: Dual Adapter
        HP 1.28TB MLC PCIe IO Accelerator, Product Number:641027-B21, SN:415165
        PCIe Power limit threshold: Disabled
        Connected ioMemory modules:
          43:00.0:      Product Number:641027-B21

43:00.0
        PCI:43:00.0
        Firmware v7.1.17, rev 116786 Public
        Internal temperature: 63.00 degC, max 63.49 degC

44:00.0
        PCI:44:00.0
        Firmware v7.1.17, rev 116786 Public
        Internal temperature: 59.55 degC, max 60.54 degC
```

missmah, 2019-06-28:
After `sudo modprobe iomemory-vsl`, it goes back to:
```
Found 2 ioMemory devices in this system
Driver version: Driver not loaded

43:00.0
        PCI:43:00.0
        Firmware v7.1.17, rev 116786 Public
        Internal temperature: 63.49 degC, max 63.98 degC

44:00.0
        PCI:44:00.0
        Firmware v7.1.17, rev 116786 Public
        Internal temperature: 60.04 degC, max 60.54 degC
```

missmah, 2019-06-28:
Just a bit more context, this is with kernel `4.15.0-54-generic`, and there were no build errors.

missmah, 2019-06-28:
Huh... despite `fio-status` saying the driver is not loaded, the `fio` drives are present in `/dev/fio*`, and I am able to access the filesystem after mounting it... Go figure :-)

So, it seems like the bug is `vis-a-vis` the new driver and `fio-status` incorrectly reporting.

missmah, 2019-06-28:
[image]

snuf, 2019-07-16:
Which version of the tools are you using? I'd like to check to see if I can reproduce your situation.

Fio-status works without the driver loaded. Except for some driver checking (scanning for /dev/fct???) it gets the rest of the info from the PCI bus, in /sys/bus/pci/devices/??????????/config, and resources. The intermittent result failure is something related to it reading from the PCI bus info is my guess.

snuf, 2019-10-12:
Closing can't reproduce with the 3.2.16 tools:
    default: Preparing to unpack fio-util_3.2.16.1731-2_amd64.deb ...
    default: Unpacking fio-util (3.2.16.1731-2) ...
    default: Setting up fio-util (3.2.16.1731-2) ...
    default: Processing triggers for man-db (2.8.5-2) ...
    default: + sudo fio-status
    default: 
    default: Found 2 ioMemory devices in this system with 1 ioDrive Duo
    default: Driver version: 3.2.16 build 1731
    default: 
    default: Adapter: Dual Adapter
    default: 	Fusion-io ioDrive Duo 640GB, Product Number:FS3-204-320-CS, SN:440178, FIO SN:440178
    default: 	External Power: NOT connected
    default: 	PCIe Power limit threshold: 24.75W
    default: 	Connected ioMemory modules:
    default: 	  fct0:	SN:443248
    default: 	  fct1:	SN:443247
    default: 
    default: fct0	Attached
    default: 	ioDIMM3 320G MLC, SN:443248
    default: 	Located in slot 0 Upper of ioDrive Duo HL SN:440178
    default: 	Last Power Monitor Incident: 13 sec
    default: 	PCI:00:04.0, Slot Number:3
    default: 	Firmware v7.1.13, rev 109322 Public
    default: 	320.00 GBytes device size
    default: 	Internal temperature: 67.92 degC, max 68.41 degC
    default: 	Reserve space status: Healthy; Reserves: 100.00%, warn at 10.00%
    default: 	Contained VSUs:
    default: 	  fioa:	ID:0, UUID:43836c20-2782-4279-91a3-25ac72c1a270
    default: 
    default: fioa	State: Online, Type: block device
    default: 	ID:0, UUID:43836c20-2782-4279-91a3-25ac72c1a270
    default: 	320.00 GBytes device size
    default: 
    default: fct1	Attached
    default: 	ioDIMM3 320G MLC, SN:443247
    default: 	Located in slot 1 Lower of ioDrive Duo HL SN:440178
    default: 	PCI:00:05.0, Slot Number:4
    default: 	Firmware v7.1.13, rev 109322 Public
    default: 	320.00 GBytes device size
    default: 	Internal temperature: 61.52 degC, max 61.52 degC
    default: 	Reserve space status: Healthy; Reserves: 100.00%, warn at 10.00%
    default: 	Contained VSUs:
    default: 	  fiob:	ID:0, UUID:e6cf3046-bf97-49c2-810e-81e14620b1d8
    default: 
    default: fiob	State: Online, Type: block device
    default: 	ID:0, UUID:e6cf3046-bf97-49c2-810e-81e14620b1d8
    default: 	320.00 GBytes device size

closed 2019-10-12 by snuf