RemixVSL/iomemory-vsl #40: Branch guidance in the readme is unclear

fake-name, 2019-02-25:
From `README.md`:

> Only the master branch is tested, nothing else. It is currently running in a CEPH cluster with official LTS kernel 4.14 and works quite well. Testing and running with the kernel module has been done on Ubuntu and Redhat. *** the untested branch has only been compiled and loaded not device tested ***

> At the moment there is a lot of code redesign ongoing. You can find the bleeding edge development in the next_generation branch. Despite its name it is more mature than the master. Especially it fixes several known bugs:

Only `master` is tested, but `next_generation` is more mature? And there'a mention of a `untested` branch that does not seem to exist at all. Additionally, there are a *bunch* of miscellaneous branches that appear to be ancient and unused.

Basically, the readme needs to be clarified, and it'd probably also be nice to have the ancient branches removed so they don't clutter the things up.

plappermaul, 2019-02-26:
Good point. Should be better now.

closed 2019-02-26 by plappermaul

fake-name, 2019-02-26:
Much better! 

The "Other Notes" section is still a bit unclear, but other then that I think you've nailed it.