RemixVSL/iomemory-vsl #41: Build fail on debian stretch using kernel 4.19

FatherChon, 2019-04-17:
Trying to build for a IOdrive duo on a fresh install of debian stretch running 4.19 from backports (4.19.28-2~bpo9+1). gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1) 

Tried via dpkg-buildpackage and make module and still run in to this below. Output is from make module. 

```
  CC [M]  /root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16/module_param.o
/root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_create_disk’:
/root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16/kblock.c:660:5: error: implicit declaration of function ‘queue_flag_set’ [-Werror=implicit-function-declaration]
     queue_flag_set(QUEUE_FLAG_WC, rq);
     ^~~~~~~~~~~~~~
In file included from /root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16/port-internal.h:61:0,
                 from /root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16/kfile.c:33:
/root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16/include/fio/port/common-linux/commontypes.h:46:32: error: size of array ‘__C_ASSERT__’ is negative
 #define C_ASSERT(x) extern int __C_ASSERT__ [(x)?1:-1]
                                ^
/root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16/kfile.c:49:1: note: in expansion of macro ‘C_ASSERT’
 C_ASSERT(sizeof(fusion_file_operations_t) >= sizeof(struct file_operations));
 ^~~~~~~~
/usr/src/linux-headers-4.19.0-0.bpo.4-common/scripts/Makefile.build:308: recipe for target '/root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16/kfile.o' failed
make[4]: *** [/root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16/kfile.o] Error 1
make[4]: *** Waiting for unfinished jobs....
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.19.0-0.bpo.4-common/scripts/Makefile.build:308: recipe for target '/root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16/kblock.o' failed
make[4]: *** [/root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16/kblock.o] Error 1
/usr/src/linux-headers-4.19.0-0.bpo.4-common/Makefile:1535: recipe for target '_module_/root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16' failed
make[3]: *** [_module_/root/buildfio/iomemory-vsl-next_generation/root/usr/src/iomemory-vsl-3.2.16] Error 2
Makefile:146: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-0.bpo.4-amd64'
Makefile:101: recipe for target 'modules' failed
make: *** [modules] Error 2
```

Any help would be appreciated, thanks!

plappermaul, 2019-05-03:
Give the next_generation branch a try. I added some modifications for 5.0 kernels.

jadams, 2019-05-07:
Same issue with RHEL 8 and kernel 4.18.0-80.el8.x86_64 using the next_generation branch

```
In file included from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/port-internal.h:61,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfile.c:33:
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/common-linux/commontypes.h:46:32: error: size of array '__C_ASSERT__' is negative
 #define C_ASSERT(x) extern int __C_ASSERT__ [(x)?1:-1]
                                ^~~~~~~~~~~~
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfile.c:49:1: note: in expansion of macro 'C_ASSERT'
 C_ASSERT(sizeof(fusion_file_operations_t) >= sizeof(struct file_operations));
 ^~~~~~~~
```

antoscha, 2019-05-13:
Building on Linux 4.19 with GCC 8.3.0 causes the same problem.

snuf, 2019-07-17:
A test was missing, and one of the tests broke on stack bloating it seems, I've fixed those two. Test built on 4.19.59 with ubuntu 18.04, and Debian Stretch work now. If you have the time please try again with the next_generation branch

closed 2019-08-11 by snuf