RemixVSL/iomemory-vsl #42: Failing to compile on Fedora 29 (5.0.13-200)

Fishwaldo, 2019-05-10:
Hi,
Have this error trying to build next_generation on fedora 29:

```
5.0.13-200.fc29.x86_64 #1 SMP Mon May 6 00:49:54 UTC 2019
```

```  CC [M]  /usr/src/iomemory-vsl-3.2.16/module_param.o
/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_fetch_next_bio’:
/usr/src/iomemory-vsl-3.2.16/kblock.c:449:24: error: incompatible type for argument 1 of ‘spin_lock_irq’
         spin_lock_irq(q->queue_lock);
                       ~^~~~~~~~~~~~
In file included from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/slab.h:15,
                 from /usr/src/iomemory-vsl-3.2.16/include/fio/port/linux/kfio_config.h:157,
                 from /usr/src/iomemory-vsl-3.2.16/include/fio/port/kfio_config.h:40,
                 from /usr/src/iomemory-vsl-3.2.16/port-internal.h:62,
                 from /usr/src/iomemory-vsl-3.2.16/kblock.c:58:
./include/linux/spinlock.h:352:55: note: expected ‘spinlock_t *’ {aka ‘struct spinlock *’} but argument is of type ‘spinlock_t’ {aka ‘struct spinlock’}
 static __always_inline void spin_lock_irq(spinlock_t *lock)
                                           ~~~~~~~~~~~~^~~~
/usr/src/iomemory-vsl-3.2.16/kblock.c:456:26: error: incompatible type for argument 1 of ‘spin_unlock_irq’
         spin_unlock_irq(q->queue_lock);
                         ~^~~~~~~~~~~~
In file included from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/slab.h:15,
                 from /usr/src/iomemory-vsl-3.2.16/include/fio/port/linux/kfio_config.h:157,
                 from /usr/src/iomemory-vsl-3.2.16/include/fio/port/kfio_config.h:40,
                 from /usr/src/iomemory-vsl-3.2.16/port-internal.h:62,
                 from /usr/src/iomemory-vsl-3.2.16/kblock.c:58:
./include/linux/spinlock.h:377:57: note: expected ‘spinlock_t *’ {aka ‘struct spinlock *’} but argument is of type ‘spinlock_t’ {aka ‘struct spinlock’}
 static __always_inline void spin_unlock_irq(spinlock_t *lock)
                                             ~~~~~~~~~~~~^~~~
/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_create_disk’:
/usr/src/iomemory-vsl-3.2.16/kblock.c:618:4: error: #error "Do not know how to disable request queue clustering for this kernel."
 #  error "Do not know how to disable request queue clustering for this kernel."
    ^~~~~
/usr/src/iomemory-vsl-3.2.16/kblock.c:640:5: warning: "KFIOC_HAS_BLK_QUEUE_FLAG_OPS" is not defined, evaluates to 0 [-Wundef]
 #if KFIOC_HAS_BLK_QUEUE_FLAG_OPS
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/iomemory-vsl-3.2.16/kblock.c:643:9: error: implicit declaration of function ‘queue_flag_set_unlocked’; did you mean ‘queued_write_unlock’? [-Werror=implicit-function-declaration]
         queue_flag_set_unlocked(QUEUE_FLAG_DISCARD, rq);
         ^~~~~~~~~~~~~~~~~~~~~~~
         queued_write_unlock
/usr/src/iomemory-vsl-3.2.16/kblock.c:668:6: warning: "KFIOC_HAS_BLK_QUEUE_FLAG_OPS" is not defined, evaluates to 0 [-Wundef]
 # if KFIOC_HAS_BLK_QUEUE_FLAG_OPS
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/iomemory-vsl-3.2.16/kblock.c:671:5: error: implicit declaration of function ‘queue_flag_set’; did you mean ‘blk_queue_flag_set’? [-Werror=implicit-function-declaration]
     queue_flag_set(QUEUE_FLAG_WC, rq);
     ^~~~~~~~~~~~~~
     blk_queue_flag_set
/usr/src/iomemory-vsl-3.2.16/kblock.c:682:6: warning: "KFIOC_HAS_BLK_QUEUE_FLAG_OPS" is not defined, evaluates to 0 [-Wundef]
 # if KFIOC_HAS_BLK_QUEUE_FLAG_OPS
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/iomemory-vsl-3.2.16/kblock.c:690:6: warning: "KFIOC_HAS_BLK_QUEUE_FLAG_OPS" is not defined, evaluates to 0 [-Wundef]
 # if KFIOC_HAS_BLK_QUEUE_FLAG_OPS
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/iomemory-vsl-3.2.16/kblock.c:693:5: error: implicit declaration of function ‘queue_flag_clear_unlocked’; did you mean ‘queued_read_unlock’? [-Werror=implicit-function-declaration]
     queue_flag_clear_unlocked(QUEUE_FLAG_ADD_RANDOM, rq);
     ^~~~~~~~~~~~~~~~~~~~~~~~~
     queued_read_unlock
/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_destroy_disk’:
/usr/src/iomemory-vsl-3.2.16/kblock.c:824:9: error: implicit declaration of function ‘blk_stop_queue’; did you mean ‘blk_sync_queue’? [-Werror=implicit-function-declaration]
         blk_stop_queue(disk->rq);
         ^~~~~~~~~~~~~~
         blk_sync_queue
/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_disk_stat_write_update’:
/usr/src/iomemory-vsl-3.2.16/kblock.c:913:46: error: macro "part_stat_inc" passed 3 arguments, but takes just 2
[... 56 lines cut ...]
            ^~~~~~~~~~
            make_request_fn
/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_mark_lock_pending’:
/usr/src/iomemory-vsl-3.2.16/kblock.c:1766:12: error: ‘struct request_queue’ has no member named ‘request_fn’; did you mean ‘make_request_fn’?
     if (q->request_fn)
            ^~~~~~~~~~
            make_request_fn
/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_unmark_lock_pending’:
/usr/src/iomemory-vsl-3.2.16/kblock.c:1784:12: error: ‘struct request_queue’ has no member named ‘request_fn’; did you mean ‘make_request_fn’?
     if (q->request_fn)
            ^~~~~~~~~~
            make_request_fn
/usr/src/iomemory-vsl-3.2.16/kblock.c: At top level:
/usr/src/iomemory-vsl-3.2.16/kblock.c:175:13: warning: ‘kfio_blk_complete_request’ declared ‘static’ but never defined [-Wunused-function]
 static void kfio_blk_complete_request(struct request *req, int error);
             ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/iomemory-vsl-3.2.16/kblock.c:176:20: warning: ‘kfio_request_to_bio’ declared ‘static’ but never defined [-Wunused-function]
 static kfio_bio_t *kfio_request_to_bio(kfio_disk_t *disk, struct request *req,
                    ^~~~~~~~~~~~~~~~~~~
/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_get_gd_in_flight’:
/usr/src/iomemory-vsl-3.2.16/kblock.c:1020:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
cc1: some warnings being treated as errors
```

dougalII, 2019-07-17:
Same for Ubuntu 19.10 on 5.0.0-20-generic
Tested both the 3.2.15 and 3.2.16 drivers

drcrimzon, 2019-08-03:
Same for Debian buster running 5.0.18-1-pve

snuf, 2019-08-11:
Had a look a couple of days ago and identified where the changes are, referring to the in-kernel changes. It's a matter of making the tests, and finding the right solution for the changes. It seems there are 5 parts of the source impacted. Will upload an annotated branch that I'm working on on Monday.

sakurachan00, 2019-08-26:
seems to happen on any 5.x kernel even in ubuntu 18.04

tripLr, 2019-09-03:
fedora 29, 
uname-r
5.1.21-200.fc29.x86_64

build errors ( similar to above )
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c: In function 'kfio_fetch_next_bio':
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:449:24: error: incompatible type for argument 1 of 'spin_lock_irq'
         spin_lock_irq(q->queue_lock);
                       ~^~~~~~~~~~~~
In file included from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/slab.h:15,
                 from /home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/include/fio/port/linux/kfio_config.h:156,
                 from /home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/include/fio/port/kfio_config.h:40,
                 from /home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/port-internal.h:62,
                 from /home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:58:
./include/linux/spinlock.h:352:55: note: expected 'spinlock_t *' {aka 'struct spinlock *'} but argument is of type 'spinlock_t' {aka 'struct spinlock'}
 static __always_inline void spin_lock_irq(spinlock_t *lock)
                                           ~~~~~~~~~~~~^~~~
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:456:26: error: incompatible type for argument 1 of 'spin_unlock_irq'
         spin_unlock_irq(q->queue_lock);
                         ~^~~~~~~~~~~~
In file included from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/slab.h:15,
                 from /home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/include/fio/port/linux/kfio_config.h:156,
                 from /home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/include/fio/port/kfio_config.h:40,
                 from /home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/port-internal.h:62,
                 from /home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:58:
./include/linux/spinlock.h:377:57: note: expected 'spinlock_t *' {aka 'struct spinlock *'} but argument is of type 'spinlock_t' {aka 'struct spinlock'}
 static __always_inline void spin_unlock_irq(spinlock_t *lock)
                                             ~~~~~~~~~~~~^~~~
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c: In function 'kfio_create_disk':
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:615:4: error: #error "Do not know how to disable request queue clustering for this kernel."
 #  error "Do not know how to disable request queue clustering for this kernel."
    ^~~~~
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c: In function 'kfio_destroy_disk':
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:807:9: error: implicit declaration of function 'blk_stop_queue'; did you mean 'blk_sync_queue'? [-Werror=implicit-function-declaration]
         blk_stop_queue(disk->rq);
         ^~~~~~~~~~~~~~
         blk_sync_queue
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c: In function 'kfio_disk_stat_write_update':
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:902:46: error: macro "part_stat_inc" passed 3 arguments, but takes just 2
         part_stat_inc(cpu, &gd->part0, ios[1]);
                                              ^
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:902:9: error: 'part_stat_inc' undeclared (first use in this function); did you mean 'part_stat_show'?
         part_stat_inc(cpu, &gd->part0, ios[1]);
         ^~~~~~~~~~~~~
         part_stat_show
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:902:9: note: each undeclared identifier is reported only once for each function it appears in
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:903:66: error: macro "part_stat_add" passed 4 arguments, but takes just 3
         part_stat_add(cpu, &gd->part0, sectors[1], totalsize >> 9);
                                                                  ^
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:903:9: error: 'part_stat_add' undeclared (first use in this function); did you mean 'part_stat_show'?
         part_stat_add(cpu, &gd->part0, sectors[1], totalsize >> 9);
         ^~~~~~~~~~~~~
         part_stat_show
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:905:67: error: macro "part_stat_add" passed 4 arguments, but takes just 3
         part_stat_add(cpu, &gd->part0, nsecs[1],   duration * 1000);
                                                                   ^
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c:891:25: warning: unused variable 'gd' [-Wunused-variable]
         struct gendisk *gd = fgd->gd;
                         ^~
/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source/kblock.c: In function 'kfio_disk_stat_read_update':
[... cut ...]
make[1]: Leaving directory '/usr/src/kernels/5.1.21-200.fc29.x86_64'
make: *** [Makefile:101: modules] Error 2
make: Leaving directory '/home/ssd_home2/triplr/rpmbuild/BUILD/iomemory-vsl-3.2.15.1700/driver_source'
+ set +x
ERROR:

error: Bad exit status from /var/tmp/rpm-tmp.AMbUE1 (%build)

RPM build errors:
    Macro expanded in comment on line 42: %{name}-%{fio_tar_version}.tar.gz

    Macro expanded in comment on line 42: %{name}-%{fio_tar_version}.tar.gz

    Bad exit status from /var/tmp/rpm-tmp.AMbUE1 (%build)
[21:14:38:Mon Sep 02:~/git/iomemory-vsl]

tripLr, 2019-09-08:
by the way it was 3.2 TB card i bought, and after seeing the lack of driver support by the manufacture, i decided to send it back

snuf, 2019-09-10:
Finally had some time to test here. Compiled and tested against 5.1.28, prior 5 kernels should work. For now I've placed it in the 5.1.28 branch. Tested writing with fio. There might be a bug with regard to iostat I noticed though.

imbezol, 2019-09-18:
I've tried compiling the next_generation branch and the 5.1.28 branch against 5.0.0-25-generic kernel on Ubuntu 19.04. It seems to get the same error with both branches. Here's the 5.1.28 output:

`[09:37:25]<imbezol@castle:/tmp/io/iomemory-vsl> debuild 
This package has a Debian revision number but there does not seem to be
an appropriate original tar file or .orig directory in the parent directory;
(expected one of iomemory-vsl_3.2.15.1800.orig.tar.gz, iomemory-vsl_3.2.15.1800.orig.tar.bz2,
iomemory-vsl_3.2.15.1800.orig.tar.lzma,  iomemory-vsl_3.2.15.1800.orig.tar.xz or iomemory-vsl.orig)
continue anyway? (y/n) y
 dpkg-buildpackage -us -uc -ui
dpkg-buildpackage: info: source package iomemory-vsl
dpkg-buildpackage: info: source version 3.2.15.1800-1.0
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Support <support@fusionio.com>
 dpkg-source --before-build .
dpkg-buildpackage: info: host architecture amd64
 fakeroot debian/rules clean
dh_testdir
dh_testroot
rm -f build-arch-stamp build-indep-stamp configure-stamp
# Add here commands to clean up after the build process.
#/usr/bin/make clean
rm -rf \
	"/tmp/io/iomemory-vsldriver_source" \
	"/tmp/io/iomemory-vsl/debian/fio" \
	/tmp/io/iomemory-vsl/debian/iomemory-vsl-5.0.0-25-generic \
	/tmp/io/iomemory-vsl/debian/iomemory-vsl-config-5.0.0-25-generic \
	/tmp/io/iomemory-vsl/debian/iomemory-vsl-5.0.0-25-generic.* \
	/tmp/io/iomemory-vsl/debian/iomemory-vsl-config-5.0.0-25-generic.* 
sed -r -i -e 's/^Package: iomemory-vsl(-config)?(-[.0-9]+.*)*$/Package: iomemory-vsl\1/;' debian/control
dh_clean
	rm -f debian/debhelper-build-stamp
	rm -rf debian/.debhelper/
	rm -f -- debian/iomemory-vsl.substvars debian/iomemory-vsl-config.substvars debian/iomemory-vsl-source.substvars debian/files
	rm -fr -- debian/iomemory-vsl/ debian/tmp/ debian/iomemory-vsl-config/ debian/iomemory-vsl-source/
	find .  \( \( \
		\( -path .\*/.git -o -path .\*/.svn -o -path .\*/.bzr -o -path .\*/.hg -o -path .\*/CVS -o -path .\*/.pc -o -path .\*/_darcs \) -prune -o -type f -a \
	        \( -name '#*#' -o -name '.*~' -o -name '*~' -o -name DEADJOE \
		 -o -name '*.orig' -o -name '*.rej' -o -name '*.bak' \
		 -o -name '.*.orig' -o -name .*.rej -o -name '.SUMS' \
		 -o -name TAGS -o \( -path '*/.deps/*' -a -name '*.P' \) \
		\) -exec rm -f {} + \) -o \
		\( -type d -a -name autom4te.cache -prune -exec rm -rf {} + \) \)
 dpkg-source -b .
dpkg-source: warning: no source format specified in debian/source/format, see dpkg-source(1)
dpkg-source: warning: source directory 'iomemory-vsl' is not <sourcepackage>-<upstreamversion> 'iomemory-vsl-3.2.15.1800'
dpkg-source: info: using source format '1.0'
dpkg-source: info: building iomemory-vsl in iomemory-vsl_3.2.15.1800-1.0.tar.gz
dpkg-source: info: building iomemory-vsl in iomemory-vsl_3.2.15.1800-1.0.dsc
 debian/rules build
sed -r -i -e 's/^Package: iomemory-vsl(-config)?(-[.0-9]+.*)*$/Package: iomemory-vsl\1-5.0.0-25-generic/;' debian/control
set -e; \
for file in /tmp/io/iomemory-vsl/debian/iomemory-vsl.*; do \
	new_file=$(echo $file | sed -r -e 's/iomemory-vsl(-config)?\./iomemory-vsl\1-5.0.0-25-generic./;'); \
	cp "$file" "$new_file"; \
done
dh_testdir
# Add here commands to configure the package.
touch configure-stamp
# Add here commands to compile the arch part of the package.
rsync -a --delete ./root/usr/src/iomemory-vsl-3.2.15/ driver_source/
if ! /usr/bin/make \
	-C driver_source \
	KERNELVER=5.0.0-25-generic \
	FIO_DRIVER_NAME=iomemory-vsl \
	FIO_SCSI_DEVICE=False \
	 \
	modules; then \
	printf "ERROR:\n"; \
	exit 1; \
fi
[... 412 lines cut ...]
            make_request_fn
/tmp/io/iomemory-vsl/driver_source/kblock.c: At top level:
/tmp/io/iomemory-vsl/driver_source/kblock.c:175:13: warning: ‘kfio_blk_complete_request’ declared ‘static’ but never defined [-Wunused-function]
 static void kfio_blk_complete_request(struct request *req);
             ^~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/io/iomemory-vsl/driver_source/kblock.c:176:20: warning: ‘kfio_request_to_bio’ declared ‘static’ but never defined [-Wunused-function]
 static kfio_bio_t *kfio_request_to_bio(kfio_disk_t *disk, struct request *req,
                    ^~~~~~~~~~~~~~~~~~~
/tmp/io/iomemory-vsl/driver_source/kblock.c: In function ‘kfio_get_gd_in_flight’:
/tmp/io/iomemory-vsl/driver_source/kblock.c:1009:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:284: /tmp/io/iomemory-vsl/driver_source/kblock.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1606: _module_/tmp/io/iomemory-vsl/driver_source] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.0.0-25-generic'
make[1]: *** [Makefile:101: modules] Error 2
make[1]: Leaving directory '/tmp/io/iomemory-vsl/driver_source'
ERROR:
make: *** [debian/rules:96: build-arch-stamp] Error 1
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui failed
`

snuf, 2019-09-18:
@Fishwaldo that is really odd, in your message it says it's building the iomemory-vsl-3.2.15.1700 source, which is not the next_gen, nor the 5.1.28 branch, those are iomemory-vsl-3.2.16. I'll do a compile of the 5.0.21 kernel, and run that against the 5.1.28 branch to make sure. Could you send me your git branch -v just in case?

snuf, 2019-09-18:
It throws some known warnings against 5.0.21, but I end up with an iomemory-vsl-config-5.0.21_3.2.16.1732_amd64.deb file created when I build the package.

closed 2019-09-26 by snuf