RemixVSL/iomemory-vsl #43: MacOS driver for MacOS 10.13.6+ (HighSierra)

vrdriver, 2019-08-11:
Hello,

I'd like to thank you for your work on the linux driver, but I was curious if you are able to assist with getting a version for Mac working?

I've got a fusionIO IODrive2.

The latest version officially created is for Yosemite however this refuses to load on future versions as it seems to be OS release locked.
https://support.fusionio.com/downloads

If seems at least one other person has had this trouble too.
https://forums.macrumors.com/threads/fusion-io-ioscale-2-and-mac-pro-5-1-pcie-ssd.2015576/
https://www.insanelymac.com/forum/topic/319589-fusionio-iodrive2-on-yosemite-fakepciid-with-custom-injector-kext-needed/

While I can install the "iomemory-vsl-3.2.15.1699.pkg" package and get access to all the utilities (/opt/local/fusion-io/bin/), it doesn't help as the (librvsl.pkg) drivers refuse to install.

Essentially this is what MacOS is spitting out:
pci1aed,2001:

  Type:	Other Mass Storage Controller
  Driver Installed:	No
  MSI:	No
  Bus:	PCI
  Slot:	Slot-4
  Vendor ID:	0x1aed
  Device ID:	0x2001
  Subsystem Vendor ID:	0x1014
  Subsystem ID:	0x0432
  Revision ID:	0x0004
  Link Width:	x4
  Link Speed:	5.0 GT/s

Not sure if you are able to assist, but I thought I'd ask.
Cheers.

snuf, 2019-08-11:
Hey Steve,

 I've not done anything with a hackintosh for a while, but am familiar with the kext, and plist madness. Had a look on Mojave with regard to the driver and think I know where the parts kind of reside that are needed/have changed, but need to brush up on this to see if I can get something to work.

Cheers,

vrdriver, 2019-08-12:
Sure thing. Well I'd be happy to test it for you if you'd like (if you get a chance to look at it that is). It's such a shame that they (Sandisk/WD) haven't actually released the source code for the Mac builds. If it's as simple as changing a version number, I'd be keen to try/host it, but as I'm not really a hardware/device driver developer, I'm probably not the right person to maintain it as yet.  
Either way, if you could point me in the right direction, I'd be happy to give it a go in Xcode and the like.

vrdriver, 2019-08-12:
On a side note, today I called their support line, and got passed around to about 4 people, but eventually I spoke to someone who was not able to confirm (or deny) any future support for MacOS.
This was the quote from the email in the end "would like to inform you that release of VSL 4.3.6 is planned for October but we are not sure which OS version it will be supporting because we have limited information available for now."
Given that there has been limited support for MacOS in recent years, I'm not holding my breath, but I thought you should know about this up-coming release. 
I did ask the question if they were going to release the source code for MacOS, but they couldn't confirm or deny if they would, so I assume it'll be like the previous releases, with just a compiled package file.

vrdriver, 2019-09-17:
Hi Huns, 
I don't mean to push this, but I'm wondering if you had a chance to look at this?
Respectfully, 
Steve

snuf, 2019-09-26:
Hey Steve,

Sorry been a while, I had a brief look, but it seems to be quite a bit more involved. I think the easiest way around it would be to write a dyld shim that contains the missing symbols and preload that (DYLD_INSERT_LIBRARIES). I've done this in the past with linux libraries (LD_PRELOAD), but not with OsX yet. I'm setting up a Dev VM to try this out on.

Cheers,

vrdriver, 2019-09-27:
Hi Funs, I look forward to seeing how you go on this.

snuf, 2019-10-10:
Been a while, have had a look a couple of times, and have a MacOS VM running. It seems like he problem originates from the _kfio_free and _kfio_malloc_* parts that actually call the kernel functions that don't exist anymore. Have also looked at the XNU kernel sources and indeed this has been removed a while ago. The best option is to create a library that contains new _kfio_free and _kfio_malloc_* functions, that in turn use vm_malloc, or something (still looking into that), instead of the missing calls.

vrdriver, 2020-01-02:
I'm guessing this is harder than originally thought?

snuf, 2020-02-05:
Not per-se, the problem is that it takes time to figure out, I've not been able to spend the time :\ Sorry about that.

vrdriver, 2020-02-15:
Hey, totally understandable. Life get in the way of many of our projects. I know the feeling well.

LAGRUNAUER, 2020-02-23:
I will be willing to contribute some funding for developing a beta driver for the Fusion iO drive which can work in MacOs Sierra, Mojave and Catalina, which I will gladly test on our workstations. Let me know if this is feasible. Thanks so much in advance!

snuf, 2020-02-29:
@LAGRUNAUER Thanks and appreciate the offer. I've hit a couple of roadblocks and the learning curve is a bit steeper than I had hoped. I also went down the wrong path a couple of times, which lead to some frustration. The time I can spend on this is the real problem atm, not necessarily money. I'm trying to extend IOMapper in some way with just two functions, which re-implement, or mimic, some behavior that is expected from IOMapper. The expected behavior is actually performed by other functions that seem to be called already, so in reality the functions could be "noops" they just have to be there.

LAGRUNAUER, 2020-03-30:
Thanks so much for the update. and please stay safe!!!

snuf, 2020-03-31:
@LAGRUNAUER thanks you too! I've done some looking into it over the weekend, and I think the only way to "solve" the issue is by patching the kernel module. I know where the bits I need to patch reside, but need to figure out how to insert the patch code, and then point from the not working bits to the patch code.

LAGRUNAUER, 2020-04-01:
Thanks so much for all your help and assistance!!!
Take care and pleas stay safe!
Luis A. Grunauer, Jr., REA

seflhjn, 2024-02-10:
So... how is it going now?

vrdriver, 2024-02-11:
Just for your records, I have retired my MacPro2009+ I don't know how many others would still be using these cards and that set up. Cheers.

bplein, 2024-02-12:
While I have my 2009/2010 Mac Pro, it's offline and I've only used it for Linux over the past several years. And the only reason to run MacOS on it would be if I were to hack it to run modern versions.... this is a dead end.

snuf, 2025-10-08:
Closing this, as it's just too far out and will never happen :\

closed 2025-10-08 by snuf