RemixVSL/iomemory-vsl #44: Failing to compile on Debian 10.0

RandomInsano, 2019-09-07:
Not urgent over here, just reporting.

```
Linux MyMachine 4.19.0-5-amd64 #1 SMP Debian 4.19.37-5+deb10u2 (2019-08-08) x86_64 GNU/Linux
```

Buidling the kernel module threw this error:

```
  CC [M]  /var/lib/dkms/iomemory-vsl/3.2.15/build/cdev.o
In file included from /var/lib/dkms/iomemory-vsl/3.2.15/build/port-internal.h:61,
                 from /var/lib/dkms/iomemory-vsl/3.2.15/build/kfile.c:33:
/var/lib/dkms/iomemory-vsl/3.2.15/build/include/fio/port/common-linux/commontypes.h:46:32: error: size of array ‘__C_ASSERT__’ is negative
 #define C_ASSERT(x) extern int __C_ASSERT__ [(x)?1:-1]
                                ^~~~~~~~~~~~
/var/lib/dkms/iomemory-vsl/3.2.15/build/kfile.c:49:1: note: in expansion of macro ‘C_ASSERT’
 C_ASSERT(sizeof(fusion_file_operations_t) >= sizeof(struct file_operations));
 ^~~~~~~~
make[4]: *** [/usr/src/linux-headers-4.19.0-5-common/scripts/Makefile.build:308: /var/lib/dkms/iomemory-vsl/3.2.15/build/kfile.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-4.19.0-5-common/Makefile:1539: _module_/var/lib/dkms/iomemory-vsl/3.2.15/build] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-5-amd64'
make: *** [Makefile:101: modules] Error 2
```

nepeat, 2019-09-08:
Seems like the same thing happened in #41 but I have no clue what the resolution was. Debian Buster, same kernel version 4.19.0-5.

snuf, 2019-09-10:
@nepeat it was stack bloating and was fixed on the next_generation. The latest for now is the 5.1.28 branch, which I've just tested, please see #42 

@RandomInsano which branch are you building from, next_generation should not have this problem anymore. The latest is the 5.1.28 branch which will merge into next_generation at some point.

ab932, 2019-10-18:
Still having this issue on Debian 10 using next_generation branch...

snuf, 2019-10-20:
@ab932 thanks for

closed 2019-10-23 by snuf