RemixVSL/iomemory-vsl #45: Ubuntu 19.04 + 5.1.28 Branch - Empty Binary Package

imbezol, 2019-10-04:
I end up with nothing in the .deb files after a compile.

```
[07:53:35]<imbezol@castle:~/tmp> git clone https://github.com/snuf/iomemory-vsl
Cloning into 'iomemory-vsl'...
remote: Enumerating objects: 84, done.
remote: Counting objects: 100% (84/84), done.
remote: Compressing objects: 100% (48/48), done.
remote: Total 1208 (delta 45), reused 58 (delta 25), pack-reused 1124
Receiving objects: 100% (1208/1208), 10.42 MiB | 2.82 MiB/s, done.
Resolving deltas: 100% (629/629), done.
[07:53:45]<imbezol@castle:~/tmp> cd iomemory-vsl/
[07:53:55]<imbezol@castle:~/tmp/iomemory-vsl> git checkout 5.1.28
Branch '5.1.28' set up to track remote branch '5.1.28' from 'origin'.
Switched to a new branch '5.1.28'
[07:53:59]<imbezol@castle:~/tmp/iomemory-vsl> ls -al
total 108
drwxrwxr-x 5 imbezol imbezol  4096 Oct  4 07:53 .
drwxrwxr-x 3 imbezol imbezol  4096 Oct  4 07:53 ..
drwxrwxr-x 2 imbezol imbezol  4096 Oct  4 07:53 debian
-rw-rw-r-- 1 imbezol imbezol 46324 Oct  4 07:53 fio-driver.spec
drwxrwxr-x 8 imbezol imbezol  4096 Oct  4 07:53 .git
-rw-rw-r-- 1 imbezol imbezol  6910 Oct  4 07:53 README
-rw-rw-r-- 1 imbezol imbezol  4135 Oct  4 07:53 README.md
drwxrwxr-x 4 imbezol imbezol  4096 Oct  4 07:53 root
[07:54:15]<imbezol@castle:~/tmp/iomemory-vsl> debuild --no-sign
 dpkg-buildpackage -us -uc -ui
dpkg-buildpackage: info: source package iomemory-vsl
dpkg-buildpackage: info: source version 3.2.16.1732
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Support <support@fusionio.com>
 dpkg-source --before-build .
dpkg-buildpackage: info: host architecture amd64
 fakeroot debian/rules clean
dh_testdir
dh_testroot
rm -f build-arch-stamp build-indep-stamp configure-stamp
# Add here commands to clean up after the build process.
#/usr/bin/make clean
rm -rf \
	"/home/imbezol/tmp/iomemory-vsldriver_source" \
	"/home/imbezol/tmp/iomemory-vsl/debian/fio" \
	/home/imbezol/tmp/iomemory-vsl/debian/iomemory-vsl-5.0.0-25-generic \
	/home/imbezol/tmp/iomemory-vsl/debian/iomemory-vsl-config-5.0.0-25-generic \
	/home/imbezol/tmp/iomemory-vsl/debian/iomemory-vsl-5.0.0-25-generic.* \
	/home/imbezol/tmp/iomemory-vsl/debian/iomemory-vsl-config-5.0.0-25-generic.* 
sed -r -i -e 's/^Package: iomemory-vsl(-config)?(-[.0-9]+.*)*$/Package: iomemory-vsl\1/;' debian/control
dh_clean
	rm -f debian/debhelper-build-stamp
	rm -rf debian/.debhelper/
	rm -f -- debian/iomemory-vsl.substvars debian/iomemory-vsl-config.substvars debian/iomemory-vsl-source.substvars debian/files
	rm -fr -- debian/iomemory-vsl/ debian/tmp/ debian/iomemory-vsl-config/ debian/iomemory-vsl-source/
	find .  \( \( \
		\( -path .\*/.git -o -path .\*/.svn -o -path .\*/.bzr -o -path .\*/.hg -o -path .\*/CVS -o -path .\*/.pc -o -path .\*/_darcs \) -prune -o -type f -a \
	        \( -name '#*#' -o -name '.*~' -o -name '*~' -o -name DEADJOE \
		 -o -name '*.orig' -o -name '*.rej' -o -name '*.bak' \
		 -o -name '.*.orig' -o -name .*.rej -o -name '.SUMS' \
		 -o -name TAGS -o \( -path '*/.deps/*' -a -name '*.P' \) \
		\) -exec rm -f {} + \) -o \
		\( -type d -a -name autom4te.cache -prune -exec rm -rf {} + \) \)
 dpkg-source -b .
dpkg-source: warning: no source format specified in debian/source/format, see dpkg-source(1)
dpkg-source: warning: source directory 'iomemory-vsl' is not <sourcepackage>-<upstreamversion> 'iomemory-vsl-3.2.16.1732'
dpkg-source: info: using source format '1.0'
dpkg-source: info: building iomemory-vsl in iomemory-vsl_3.2.16.1732.tar.gz
dpkg-source: info: building iomemory-vsl in iomemory-vsl_3.2.16.1732.dsc
 debian/rules build
sed -r -i -e 's/^Package: iomemory-vsl(-config)?(-[.0-9]+.*)*$/Package: iomemory-vsl\1-5.0.0-25-generic/;' debian/control
set -e; \
for file in /home/imbezol/tmp/iomemory-vsl/debian/iomemory-vsl.*; do \
[... 1430 lines cut ...]
-rw-r--r--   1 imbezol imbezol      760 Oct  4 07:54 iomemory-vsl_3.2.16.1732.dsc
-rw-r--r--   1 imbezol imbezol 15857600 Oct  4 07:54 iomemory-vsl_3.2.16.1732.tar.gz
-rw-r--r--   1 imbezol imbezol     9508 Oct  4 07:55 iomemory-vsl-5.0.0-25-generic_3.2.16.1732_amd64.deb
-rw-r--r--   1 imbezol imbezol     9508 Oct  4 07:55 iomemory-vsl-config-5.0.0-25-generic_3.2.16.1732_amd64.deb
-rw-r--r--   1 imbezol imbezol     9512 Oct  4 07:55 iomemory-vsl-source_3.2.16.1732_amd64.deb
-rw-rw-r--   1 imbezol imbezol 10138973 Jan 20  2019 jeeptent.jpg
-rw-rw-r--   1 imbezol imbezol       76 May  8 21:48 .~lock.LoadLog.ods#
-rw-rw-r--   1 imbezol imbezol    70543 Feb 24  2019 search.html
-rw-rw-r--   1 imbezol imbezol   508071 Jan 18  2019 wk180c.jpg
[07:56:04]<imbezol@castle:~/tmp> dpkg --contents iomemory-vsl-5.0.0-25-generic_3.2.16.1732_amd64.deb
drwxr-xr-x root/root         0 2015-02-04 10:42 ./
drwxr-xr-x root/root         0 2015-02-04 10:42 ./usr/
drwxr-xr-x root/root         0 2015-02-04 10:42 ./usr/share/
drwxr-xr-x root/root         0 2015-02-04 10:42 ./usr/share/doc/
drwxr-xr-x root/root         0 2015-02-04 10:42 ./usr/share/doc/iomemory-vsl-5.0.0-25-generic/
-rw-r--r-- root/root      1850 2015-02-04 10:42 ./usr/share/doc/iomemory-vsl-5.0.0-25-generic/changelog.gz
-rw-r--r-- root/root     19255 2015-02-04 10:42 ./usr/share/doc/iomemory-vsl-5.0.0-25-generic/copyright
[07:56:12]<imbezol@castle:~/tmp> dpkg --contents iomemory-vsl-config-5.0.0-25-generic_3.2.16.1732_amd64.deb
drwxr-xr-x root/root         0 2015-02-04 10:42 ./
drwxr-xr-x root/root         0 2015-02-04 10:42 ./usr/
drwxr-xr-x root/root         0 2015-02-04 10:42 ./usr/share/
drwxr-xr-x root/root         0 2015-02-04 10:42 ./usr/share/doc/
drwxr-xr-x root/root         0 2015-02-04 10:42 ./usr/share/doc/iomemory-vsl-config-5.0.0-25-generic/
-rw-r--r-- root/root      1850 2015-02-04 10:42 ./usr/share/doc/iomemory-vsl-config-5.0.0-25-generic/changelog.gz
-rw-r--r-- root/root     19255 2015-02-04 10:42 ./usr/share/doc/iomemory-vsl-config-5.0.0-25-generic/copyright```

imbezol, 2019-10-04:
```
Linux castle 5.0.0-25-generic #26-Ubuntu SMP Thu Aug 1 12:04:58 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

snuf, 2019-10-04:
Hi,

Looking at the output it does produce the module:

  CC      /home/imbezol/tmp/iomemory-vsl/driver_source/iomemory-vsl.mod.o

  LD [M]  /home/imbezol/tmp/iomemory-vsl/driver_source/iomemory-vsl.ko

make[2]: Leaving directory '/usr/src/linux-headers-5.0.0-25-generic'

It seems there is something with the compatibility of the packaging tooling
at first glance. I’ll have a look on 19.04 to see if I can fix it. In the
meantime you  could build the module manually and copy/insert it or use
dkms if that is an option for you?

Cheers,

Funs

snuf, 2019-10-05:
Hi,

Seems like a build options issue, after checking I was initially using:
```
dpkg-buildpackage -rfakeroot -b --no-check-builddeps -us -uc
```
which hasn't resulted in a good deb file since July it turns out. Switching to:
```
dpkg-buildpackage -rfakeroot -B --no-check-builddeps -us -uc
```
However does, adding a check for the size of the package now when built, making sure this doesn't happen again.

Looked at debuild, and when adding the "-b" the binrary flag I get an empty package too, as it invokes dpkg-buildpackage with a small -b instead of a capitol -B. 
```
snuf@scipio:~/Temp/iomemory-vsl$ debuild -i -us -uc -b
 dpkg-buildpackage -rfakeroot -us -uc -ui -i -b
```
debuild -B however does translate into:
```
snuf@scipio:~/Temp/iomemory-vsl$ debuild --no-sign -B
 dpkg-buildpackage -rfakeroot -us -uc -ui -B
```
Which results in the kernel module and config deb file with the module inside.

Hope this helps,

Funs

snuf, 2019-10-10:
Closing this due to lack of feedback.

imbezol, 2019-10-10:
i was able to install the module via DKMS and it does recognize the card and I was able to create a PV and a filesystem on it. However, when I shut down my system crashes with a kernel dump. Additionally it does not come up automatically because LVM is not seeing the PV, which may just be a matter of telling it to look for that device type, but have not had time to dig into it further yet. I will test the debuild method again as suggested and give feedback, hopefully next week. Thanks for your help.

snuf, 2019-10-11:
Ah awesome. Thanks for your feedback, it's valuable information that helps to improve the quality. I'll add LVM to testing when I have some time. The DKMS route is generally easiest, and happy it worked out. The system crashing is "no bueno"....

When unloading the module in the test suite I do see a dump, it doesn't crash the system though:
```
[   90.698893] BUG: unable to handle kernel NULL pointer dereference at 0000000000000000
[   90.698945] #PF error: [WRITE]
[   90.698961] PGD 0 P4D 0 
[   90.698978] Oops: 0002 [#1] SMP PTI
[   90.698998] CPU: 7 PID: 1750 Comm: rmmod Tainted: G           OE     5.0.0-29-generic #31-Ubuntu
[   90.699031] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS 1.10.2-1ubuntu1 04/01/2014
[   90.699070] RIP: 0010:_raw_spin_lock_irqsave+0x22/0x40
[   90.699092] Code: 66 ff 5d c3 0f 1f 40 00 66 66 66 66 90 55 48 89 e5 53 9c 58 66 66 90 66 90 48 89 c3 fa 66 66 90 66 66 90 31 c0 ba 01 00 00 00 <f0> 0f b1 17 75 06 48 89 d8 5b 5d c3 89 c6 e8 bb 05 6c ff 66 90 eb
[   90.699156] RSP: 0018:ffffb2b741d13d40 EFLAGS: 00010046
[   90.699179] RAX: 0000000000000000 RBX: 0000000000000246 RCX: ffffb2b741d13d24
[   90.699207] RDX: 0000000000000001 RSI: ffff93ba3498e2e0 RDI: 0000000000000000
[   90.699234] RBP: ffffb2b741d13d48 R08: ffffb2b741d13d24 R09: ffffb2b741d13c50
[   90.699262] R10: ffff93ba2903df80 R11: 0000000000000001 R12: 0000000000000000
[   90.699289] R13: ffff93ba3498e180 R14: 0000000000000000 R15: 0000000000000000
[   90.699318] FS:  00007fe1ebabb740(0000) GS:ffff93ba37bc0000(0000) knlGS:0000000000000000
[   90.699348] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   90.699372] CR2: 0000000000000000 CR3: 00000002040c4005 CR4: 00000000000606e0
[   90.699407] Call Trace:
[   90.699473]  fusion_spin_lock_irqsave+0x12/0x20 [iomemory_vsl]
[   90.699522]  kfio_destroy_disk+0x51/0x100 [iomemory_vsl]
[   90.699576]  ifio_b1760.51deb95af7bcac43ff607265fea6779f673.3.2.16.1731+0xa5/0xe0 [iomemory_vsl]
[   90.699636]  fio_detach+0x63e/0xca0 [iomemory_vsl]
[   90.699662]  ? __wait_rcu_gp+0x114/0x140
[   90.699683]  ? synchronize_rcu+0x75/0x80
[   90.699730]  ifio_1d82e.51fa10865e9de192e97d387322d0cd5e41e.3.2.16.1731+0x4d/0xb0 [iomemory_vsl]
[   90.699766]  ? __sysrq_swap_key_ops+0x68/0x80
[   90.699816]  cleanup_fio_blk+0x9/0x10 [iomemory_vsl]
[   90.699859]  exit_fio_driver+0x1a/0x35 [iomemory_vsl]
[   90.699884]  __x64_sys_delete_module+0x13f/0x270
[   90.699907]  do_syscall_64+0x5a/0x110
[   90.699926]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[   90.699949] RIP: 0033:0x7fe1ebbfd427
[   90.699968] Code: 73 01 c3 48 8b 0d 69 5a 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 b8 b0 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 39 5a 0c 00 f7 d8 64 89 01 48
[   90.700031] RSP: 002b:00007ffc74f47728 EFLAGS: 00000206 ORIG_RAX: 00000000000000b0
[   90.700061] RAX: ffffffffffffffda RBX: 0000562adb373780 RCX: 00007fe1ebbfd427
[   90.700088] RDX: 000000000000000a RSI: 0000000000000800 RDI: 0000562adb3737e8
[   90.700115] RBP: 0000000000000000 R08: 00007ffc74f466a1 R09: 0000000000000000
[   90.700142] R10: 00007fe1ebc78ae0 R11: 0000000000000206 R12: 00007ffc74f47950
[   90.700170] R13: 00007ffc74f48daa R14: 0000562adb373260 R15: 0000562adb373780
[   90.700198] Modules linked in: iomemory_vsl(OE-) dm_multipath scsi_dh_rdac scsi_dh_emc scsi_dh_alua kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel joydev input_leds serio_raw mac_hid qemu_fw_cfg sch_fq_codel ib_iser rdma_cm iw_cm ib_cm ib_core iscsi_tcp libiscsi_tcp libiscsi scsi_transport_iscsi ip_tables x_tables autofs4 btrfs zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear cirrus aesni_intel ttm drm_kms_helper aes_x86_64 crypto_simd cryptd syscopyarea glue_helper sysfillrect sysimgblt psmouse fb_sys_fops drm virtio_net sym53c8xx net_failover failover scsi_transport_spi i2c_piix4 pata_acpi floppy
[   90.702923] CR2: 0000000000000000
[   90.703730] ---[ end trace 39188a0bc7b94672 ]---
[   90.704520] RIP: 0010:_raw_spin_lock_irqsave+0x22/0x40
[   90.705314] Code: 66 ff 5d c3 0f 1f 40 00 66 66 66 66 90 55 48 89 e5 53 9c 58 66 66 90 66 90 48 89 c3 fa 66 66 90 66 66 90 31 c0 ba 01 00 00 00 <f0> 0f b1 17 75 06 48 89 d8 5b 5d c3 89 c6 e8 bb 05 6c ff 66 90 eb
[   90.706569] RSP: 0018:ffffb2b741d13d40 EFLAGS: 00010046
[   90.707088] RAX: 0000000000000000 RBX: 0000000000000246 RCX: ffffb2b741d13d24
[   90.707616] RDX: 0000000000000001 RSI: ffff93ba3498e2e0 RDI: 0000000000000000
[   90.708125] RBP: ffffb2b741d13d48 R08: ffffb2b741d13d24 R09: ffffb2b741d13c50
[   90.708624] R10: ffff93ba2903df80 R11: 0000000000000001 R12: 0000000000000000
[   90.709122] R13: ffff93ba3498e180 R14: 0000000000000000 R15: 0000000000000000
[   90.709614] FS:  00007fe1ebabb740(0000) GS:ffff93ba37bc0000(0000) knlGS:0000000000000000
[   90.710123] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   90.710618] CR2: 0000000000000000 CR3: 00000002040c4005 CR4: 00000000000606e0
vagrant@fio:~$ 
```
kfio_destroy_disk, which calls fusion_spin_lock_irqsave, which actually calls _raw_spin_lock_irqsave seems to be the origin. Look at lsmod it also gives a -1 as its status:
    default: iomemory_vsl         1277952  -1

snuf, 2019-10-11:
Ah turns out disk queue_lock in the kfio_destroy_disk is actually NULL sometimes (often), and not checked. I've updated it with checking if it's NULL around the save and restore, which makes the above error pop up. I'm running tests now on the 5.1.28 with the fix in.

snuf, 2019-10-12:
Fix seems to work, tests are clean

About the PV, are you trying to boot from it or is it just another FS on the machine ?

closed 2019-12-12 by snuf