RemixVSL/iomemory-vsl #47: Did not work on CentOS 8

furban1, 2019-10-23:
Hi,

will be there a version working on CentOS 8. It does not work at this time now:

                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfile.c:33:
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/common-linux/commontypes.h:46:32: error: size of array '__C_ASSERT__' is negative
 #define C_ASSERT(x) extern int __C_ASSERT__ [(x)?1:-1]
                                ^~~~~~~~~~~~
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfile.c:49:1: note: in expansion of macro 'C_ASSERT'
 C_ASSERT(sizeof(fusion_file_operations_t) >= sizeof(struct file_operations));

Greetings
Frank

snuf, 2019-10-23:
Hi Frank,

What gcc version, kernel version, and branch are you compiling?

Kan you try the 5.1.28 branch and let me know if that works?

Thanks,

Funs

furban1, 2019-10-23:
Hi Funs,

Nothing special here. Only CentOS 8 everything patched to the newest version
gcc-8.2.1-3.5.el8.x86_64
kernel-core-4.18.0-80.11.2.el8_0.x86_64

I have done this and so I still used version 5.1.28 when I'm right

[root@host~]# git clone https://github.com/snuf/iomemory-vsl
Cloning into 'iomemory-vsl'...
remote: Enumerating objects: 163, done.
remote: Counting objects: 100% (163/163), done.
remote: Compressing objects: 100% (80/80), done.
remote: Total 1287 (delta 84), reused 127 (delta 60), pack-reused 1124
Receiving objects: 100% (1287/1287), 10.46 MiB | 8.07 MiB/s, done.
Resolving deltas: 100% (668/668), done.
[root@host~]# cd iomemory-vsl/
[root@host iomemory-vsl]# git checkout next_generation
Branch 'next_generation' set up to track remote branch 'next_generation' from 'origin'.
Switched to a new branch 'next_generation'
[root@host iomemory-vsl]# rpmbuild -ba fio-driver.spec

snuf, 2019-10-23:
Hi Frank,

Can you try:
```
git checkout 5.1.28 
```
instead of 
``` 
git checkout next_generation
```
it seems like you're checking out the *next_generation* branch. The *5.1.28* branch is an evolution of that branch that has not been merged back yet, and contains the fix required.

furban1, 2019-10-23:
OK,
looks different but there are still problems

                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/sysrq.c:34:
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/common-linux/commontypes.h:49:47: error: 'x' undeclared here (not in a function)
 #define C_ASSERT(X) extern int __C_ASSERT__ [(x)?1:-1]
                                               ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/gcc/align.h:40:4: note: in expansion of macro 'C_ASSERT'
    C_ASSERT(__alignof__(structure) == (align_bytes))

snuf, 2019-10-23:
Ah I see why, I made an error in the matching of the GCC version. Just pushed a fix, if you do a ```git pull``` on the branch you should get the update that matches the GCC version correctly.

furban1, 2019-10-23:
Great!
Now it looks good
/root/rpmbuild/RPMS/x86_64/iomemory-vsl-source-3.2.16.1732-1.0.el8.x86_64.rpm
Many thanks

closed 2019-10-23 by furban1