RemixVSL/iomemory-vsl #48: ioMemory utilities' support library missing

lkopczynski, 2019-10-23:
Hey! Thank you very much for your work.

I'm on Ubuntu 18.04 and after the setup I get the following:

"Error: ioMemory utilities' support library is missing.
       Make sure the ioMemory driver is properly installed and try again.

       Found 3 ioMemory cards requiring v4.x driver."

What I did:
- build and install the source in branch next_generation
- install the newest (for Ubuntu 16.04) Utilities from support.fusionio.com
- load module using modprobe

Any ideas?

Thanks!

snuf, 2019-10-23:
Hi!

Ah the iomemory 4 devices are different devices than the iomemory 3 and older devices. Can you attach the ```lspci -b -nn ``` output for your cards to make sure?

In the past I've done an adaptation of the iomemory 4 driver for someone based on iomemory-vsl4-4.3.1.836 to be specific (never checked it in though). There is someone else who did [iomemory-vsl4-4.3.5](https://github.com/naamval/iomemory-vsl4/tree/master/root/usr/src/iomemory-vsl4-4.3.5) three months ago. Did a quick compile of it and seems to go ok. The changes we've done here in the past seem to be portable to the iomemory-vsl4 driver 1<->1.

Best,

Funs

qdrop17, 2019-10-24:
> Hi!
>
> Ah the iomemory 4 devices are different devices than the iomemory 3 and older devices. Can you attach the `lspci -b -nn ` output for your cards to make sure?
>
> In the past I've done an [...]

Hi @snuf. I'm working together with @lkopczynski. As requested the output of lspci:

```
root@K8S-onpremnode001:/home/master# lspci -b -nn | grep ioMemory
81:00.0 Mass storage controller [0180]: SanDisk ioMemory HHHL [1aed:3002]
c1:00.0 Mass storage controller [0180]: SanDisk ioMemory HHHL [1aed:3002]
cb:00.0 Mass storage controller [0180]: SanDisk ioMemory HHHL [1aed:3002]
```

We'll give the other repo with the 4.x drivers a shot :-)

Thank you very much for your support so far!

snuf, 2019-10-24:
@qdrop17 you're welcome, thanks for the lspci. Please let me know if it works out with the iomemory-vsl4 driver. If you need any fixes I can assist. 

Perhaps we should put all this stuff in one place at some point.

lkopczynski, 2019-10-28:
@snuf I just tried to build the source from the repo you mentioned but keep getting this:

`make[3]: *** No rule to make target '/tmp/iomemory-vsl4-master/driver_source/kfio/x86_64_cc74_libkfio.o', needed by '/tmp/iomemory-vsl4-master/driver_source/iomemory-vsl4.o'.  Stop.
make[3]: *** Waiting for unfinished jobs....
  CC [M]  /tmp/iomemory-vsl4-master/driver_source/khotplug.o
  CC [M]  /tmp/iomemory-vsl4-master/driver_source/kcsr.o
Makefile:1577: recipe for target '_module_/tmp/iomemory-vsl4-master/driver_source' failed
make[2]: *** [_module_/tmp/iomemory-vsl4-master/driver_source] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.15.0-66-generic'
Makefile:84: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/tmp/iomemory-vsl4-master/driver_source'
ERROR:
debian/rules:100: recipe for target 'build-arch-stamp' failed
make: *** [build-arch-stamp] Error 1
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2`

Do you maybe have any input on what we could try? Any help much appreciated!

snuf, 2019-10-28:
@lkopczynski ah that is the library that shims the device operationsl. By default when you compile the library it will look for the library with the C major and C minor version. A patch on the iomemory-vsl driver [copies](https://github.com/snuf/iomemory-vsl/blob/master/root/usr/src/iomemory-vsl-3.2.15/Makefile#L64-L70) the latest one over when running make.

This can be resolved by copying the object file to the major/minor version of your C compiler:
```
cp iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.5/kfio/x86_64_cc63_libkfio.o_shipped iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.5/kfio/x86_64_cc74_libkfio.o_shipped
```

Let me know if this helps?

Best,

SwissGipfel, 2019-10-28:
Hi @snuf 
Another one working with @lkopczynski here :). 

Your copy hint for the library helped very much.

I had to use

```
cp iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.4/kfio/x86_64_cc71_libkfio.o_shipped iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.4/kfio/x86_64_cc74_libkfio.o_shipped
cp iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.4/kfio/.x86_64_cc71_libkfio.o.cmd iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.4/kfio/.x86_64_cc74_libkfio.o.cmd
```

until the build command progressed further.

However, now we got a new error. 

```
dh_missing: missing files, aborting
dh_install: dh_missing --exclude /usr/share/doc/iomemory-vsl4 --sourcedir root --fail-missing returned exit code 255
debian/rules:133: recipe for target 'install-arch' failed
make: *** [install-arch] Error 25
dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2
```

It seems that `dh_missing` reports every file under `usr/src/iometry-vsl4-4.3.4/` as `exists in root but is not installed to anywhere`

Do you have another helpful input for us?

Thanks for your time!

Best,

SwissGipfel, 2019-10-28:
Update:
Out of a hunch, I tried to follow the way "without RPM or .deb" from your README and build/install the module using make directly in `root/usr/src/iomemory-vsl4-4.3.4`

`make modules` runs trough without errors as it did with `dpkg-buildpackage -rfakeroot -b`

`make modules_install` seems to have a problem because there's no signing key, but other than that, also runs trough smoothly, as far as I can see.

```
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-66-generic'
  INSTALL /tmp/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.4/iomemory-vsl4.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-66-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-66-generic'
```

so the error I described above has nothing to do with your or naamvals repo. Seems to be a problem with our dephelper version itself. :( or a lack of my understanding.

snuf, 2019-10-28:
@SwissGipfel good to see it worked out in the end! 

Yeah the debian package build can be somewhat of a pain, I remember having to tweak some things on that in the past. Which command are you using to build the package btw?

I've never tried the ```make modules_install``` myself tbh. I most often go for ```dkms``` as it gives the comfort of recompiling when I upgrade a kernel, providing the module works with that kernel.

SwissGipfel, 2019-10-29:
>Which command are you using to build the package btw?
I was using `dpkg-buildpackage -rfakeroot -b` because I thought it was the first choice described in the original README. Maybe I should give `sudo dpkg-buildpackage` or `debuild` a chance too.

snuf, 2019-10-29:
@SwissGipfel there's several ways I've used ```dpkg-buildpackage``` in several forms, my module_builder container uses 
```
src_hdr=$(ls -1 /usr/src | grep $kernel_branch)
deb_kernel_version=$kernel_branch \
  KERNEL_SRC=/usr/src/$src_hdr \
  dpkg-buildpackage \
  -rfakeroot -B --no-check-builddeps -us -uc
```
However in the module_tester I tend to use
```
dpkg-buildpackage \
  -rfakeroot --no-check-builddeps --no-sign
```

snuf, 2019-10-29:
@qdrop17 as long as there are no changes in the linux kernel, as in impacting changes to the queue or block layer. So if you're on LTS and a newer version of the kernel pops up you should be fine. However if there is a major change, as in minor version change in the linux kernel, the impact may be that the build will fail, and you will need a newer driver with patches for the missing, or changed parts in the kernel. Also, if the GCC version changes make sure to copy the files in step 4 to the right major-minor number, I do this in the Makefile for the iomemory-vsl driver.

For example, if in Ubuntu the kernel hops from 4.18.10 to 4.18.39 there should not be a problem as none of the things the driver interfaces with are impacted. However if you the version jumps from 4.18.39 to 4.19.x or 5.x.x there are chances that the driver you have now will not work and fail to build. The background of this is that the parts of the linux kernel are actively changed, and parts the driver interfaces with get deprecated, changed or newly-introduced. With regard to GCC the major and minor version in step 4 are presented by the ```cc74```.

I hope this somewhat answers your question?

snuf, 2019-10-30:
@lkopczynski @qdrop17 @SwissGipfel am closing this one, if there's anything else don't hesitate to reach out!

Best

closed 2019-10-30 by snuf