RemixVSL/iomemory-vsl #49: Fusion-IO SX300-6400 unsupported?

flotus1, 2019-10-31:
I tried to get this working on OpenSUSE Leap 15.1, kernel 4.12.14-lp151.28.13-default
With a lot of help over at the OpenSUSE forum, we pinned down the issue to my particular card not being supported by this driver [https://forums.opensuse.org/showthread.php/537593-Fusion-IO-SSD-driver-install](https://forums.opensuse.org/showthread.php/537593-Fusion-IO-SSD-driver-install)

Now thanks to my general cluelessness, it is entirely possible that I screwed up somewhere in the process of setting up the driver. So before digging any deeper, I wanted to ask if this card is really not supported.

snuf, 2019-10-31:
Hi @flotus1, 

I went over the thread on opensuse. The SX300 cards are as far as I know iomemory-vsl4 devices, which won't work with the iomemory-vsl 3.x branch software, as you mentioned too. The enhancements in 3.2.x here are not device support fixes, but general driver fixes. When troubleshooting the loading of the kernel module looking at ```dmesg``` can be useful as it will tell you the if initialization goes according to plan. For your device have a look at #48 or give the driver referred to in the [wiki](https://github.com/snuf/iomemory-vsl/wiki) a go. 

Let me know if this helps

flotus1, 2019-10-31:
What started this whole exercise was that I read about drivers for these SSDs no longer working with newer Linux Kernel versions. Since I intended to use this drive longer than most of the other hardware in my system, I tried to search for a solution.
Will the 4.x software branch have the same problem as 3.x with Linux kernels, or could I just keep using the card with official 4.x drivers?

snuf, 2019-11-01:
**_tl;dr_**: If you're not running a hot to trot kernel or your linux distribution and version are in the download list, stick with the official driver. If you are running a newer kernel or a distribution that is not on the list that has a newer kernel, it depends....

**_long_**: Eventually every driver that is not maintained to a certain degree will have the problem that it will stop working. The Linux kernel has a healthy amount of flux, contentiously improves, and with doing so deprecates or changes interfacing subsystems drivers use.

The 4.x driver working depends on the vendor maintaining the 4.x drivers, where it seems that [naamval](https://github.com/naamval/iomemory-vsl4) encountered the driver not working with *some kernel*. Looking at #48 it seems that there are minor issues when running on newer kernels and the driver still. However, the iomemory-vsl4 driver is actively maintained for now it seems. Depending on which linux distributions they support, and the kernels those distributions run you're supported if you stick to that set. As not everyone does, they get into issues where they have to patch the driver, its build, or other things, to work with the kernel they run.

The Iomemory-vsl 3.16 driver as an example supports Red Hat 7 and Sles 12, which come with a 3.10 and 3.12 kernel, which already kind of indicates that if you're wanting to run a newer distribution and kernel you might have a tougher time.

Hope this clarifies some things?

flotus1, 2019-11-04:
Thanks for clearing up some of my misconceptions.
So let's assume I just run the SSD until the OS I use can no longer handle the driver, and then switch to different hardware, preferably something NVMe...for which OS version should I download the official drivers? I am using OpenSUSE leap 15.1, which is not one of the distributions available on the download page.

snuf, 2019-11-05:
Don't think it matters a lot, as long as you get 4.3.6. 4.3.6 supports Ubuntu 16.04 you should be ok with the kernel version in OpenSUSE. The only problem you may find is that the GCC version stamped *.so* is missing as outlined in #48.

snuf, 2019-11-08:
@flotus1 closing this one, if it didn't work out let me know!

closed 2019-11-08 by snuf