RemixVSL/iomemory-vsl #50: new_generation dkms build fails on Arch w kernel 5.3.8

sotiris-bos, 2019-11-05:
### Bug description
DKMS build fails

### How to reproduce

### Environment information
 

make log:

```
cat /var/lib/dkms/iomemory-vsl/3.2.16/build/make.log 
DKMS make.log for iomemory-vsl-3.2.16 for kernel 5.3.8-arch1-1 (x86_64)
Tue  5 Nov 21:09:02 EET 2019
if [ "92" -gt "63" ];then \
        if [ ! -f "kfio/x86_64_cc92_libkfio.o_shipped" ];then \
                cp kfio/x86_64_cc63_libkfio.o_shipped kfio/x86_64_cc92_libkfio.o_shipped; \
        fi \
fi
./kfio_config.sh -a x86_64 -o include/fio/port/linux/kfio_config.h -k /lib/modules/5.3.8-arch1-1/build -p -d /var/lib/dkms/iomemory-vsl/3.2.16/build/kfio_config -l 0
Detecting Kernel Flags
Config dir         : /var/lib/dkms/iomemory-vsl/3.2.16/build/kfio_config
Output file        : include/fio/port/linux/kfio_config.h
Kernel output dir  : /lib/modules/5.3.8-arch1-1/build
Kernel source dir  : 
Starting tests:
  1572980942.565  KFIOC_MISSING_WORK_FUNC_T...
  1572980942.575  KFIOC_WORKDATA_PASSED_IN_WORKSTRUCT...
  1572980942.586  KFIOC_HAS_PCI_ERROR_HANDLERS...
  1572980942.595  KFIOC_HAS_GLOBAL_REGS_POINTER...
  1572980942.604  KFIOC_HAS_SYSRQ_KEY_OP_ENABLE_MASK...
  1572980942.614  KFIOC_HAS_DISK_STATS_READ_WRITE_ARRAYS...
  1572980942.624  KFIOC_HAS_LINUX_SCATTERLIST_H...
  1572980942.634  KFIOC_KMEM_CACHE_CREATE_REMOVED_DTOR...
  1572980942.645  KFIOC_BIO_ENDIO_HAS_BYTES_DONE...
  1572980942.659  KFIOC_INVALIDATE_BDEV_REMOVED_DESTROY_DIRTY_BUFFERS...
  1572980942.669  KFIOC_NEEDS_VIRT_TO_PHYS...
  1572980942.680  KFIOC_HAS_BLK_UNPLUG...
  1572980942.692  KFIOC_HAS_BLK_DELAY_QUEUE...
  1572980942.705  KFIOC_REQUEST_QUEUE_HAS_UNPLUG_FN...
  1572980942.718  KFIOC_REQUEST_QUEUE_UNPLUG_FN_HAS_EXTRA_BOOL_PARAM...
  1572980942.732  KFIOC_BACKING_DEV_INFO_HAS_UNPLUG_IO_FN...
  1572980942.748  KFIOC_HAS_KMEM_CACHE...
  1572980942.763  KFIOC_HAS_MUTEX_SUBSYSTEM...
  1572980942.778  KFIOC_STRUCT_FILE_HAS_PATH...
  1572980942.793  KFIOC_HAS_PATH_LOOKUP...
  1572980942.812  KFIOC_UNREGISTER_BLKDEV_RETURNS_VOID...
  1572980942.828  KFIOC_PARTITION_STATS...
  1572980942.854  KFIOC_HAS_NEW_BLOCK_METHODS...
  1572980942.873  KFIOC_BLOCK_DEVICE_RELEASE_RETURNS_INT...
  1572980942.892  KFIOC_HAS_NEW_BLKDEV_METHODS...
  1572980942.911  KFIOC_HAS_COMPAT_IOCTL_METHOD...
  1572980942.929  KFIOC_COMPAT_IOCTL_RETURNS_LONG...
  1572980942.951  KFIOC_DISCARD...
  1572980942.974  KFIOC_DISCARD_ZEROES_IN_LIMITS...
  1572980942.995  KFIOC_HAS_REQ_UNPLUG...
  1572980943.020  KFIOC_HAS_BIO_RW_SYNC...
  1572980943.043  KFIOC_HAS_BIO_RW_UNPLUG...
  1572980943.065  KFIOC_HAS_BIOVEC_ITERATORS...
  1572980943.088  KFIOC_DISCARD_GRANULARITY_IN_LIMITS...
  1572980943.111  KFIOC_BARRIER...
  1572980943.138  KFIOC_USE_LINUX_UACCESS_H...
  1572980943.160  KFIOC_HAS_SPIN_LOCK_IRQSAVE_NESTED...
  1572980943.187  KFIOC_MODULE_PARAM_ARRAY_NUMP...
  1572980943.212  KFIOC_OWNER_IN_STRUCT_PROC_DIR_ENTRY...
  1572980943.238  KFIOC_CONFIG_PREEMPT_RT...
  1572980943.267  KFIOC_CONFIG_TREE_PREEMPT_RCU...
  1572980943.287  KFIOC_HAS_BLK_QUEUE_HARDSECT_SIZE...
  1572980943.314  KFIOC_HAS_END_REQUEST...
  1572980943.337  KFIOC_USE_IO_SCHED...
[... 424 lines cut ...]
 1784 |     if (q->request_fn)
      |            ^~~~~~~~~~
      |            make_request_fn
/var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.c: In function ‘kfio_make_request’:
/var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.c:1971:9: error: implicit declaration of function ‘bio_phys_segments’; did you mean ‘bio_segments’? [-Werror=implicit-function-declaration]
 1971 |     if (bio_phys_segments(queue, bio) > queue_max_segments(queue))
      |         ^~~~~~~~~~~~~~~~~
      |         bio_segments
/var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.c: At top level:
/var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.c:175:13: warning: ‘kfio_blk_complete_request’ declared ‘static’ but never defined [-Wunused-function]
  175 | static void kfio_blk_complete_request(struct request *req, int error);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.c:176:20: warning: ‘kfio_request_to_bio’ declared ‘static’ but never defined [-Wunused-function]
  176 | static kfio_bio_t *kfio_request_to_bio(kfio_disk_t *disk, struct request *req,
      |                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.c: In function ‘kfio_get_gd_in_flight’:
/var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.c:1020:1: warning: control reaches end of non-void function [-Wreturn-type]
 1020 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.o] Error 1
make[1]: *** [Makefile:1626: _module_/var/lib/dkms/iomemory-vsl/3.2.16/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.3.8-arch1-1/build'
make: *** [Makefile:101: modules] Error 2
```

snuf, 2019-11-05:
@sotiris-bos thanks for the report, could you check out the 5.1.28 branch, it contains the required fixes for all kernels up to 5.4.x.

Let me know if that helps?

sotiris-bos, 2019-11-05:
> @sotiris-bos thanks for the report, could you check out the 5.1.28 branch, it contains the required fixes for all kernels up to 5.4.x.
> 
> Let me know if that helps?

Yes it does work!

Your README in the 5.1.28 branch needs some updating (3.2.16 instead of 3.2.15). I thought the next_generation was the 3.2.16 version and the latest version.

Thank you so much for your work!

snuf, 2019-11-05:
Welcome, I'll merge the 5.1.28 and next_generation when I have some spare cycles. Thanks for getting back!

closed 2019-11-05 by snuf