RemixVSL/iomemory-vsl #52: Weird dmesg messages

sotiris-bos, 2019-11-11:
### Bug description
Getting some dmesg messages while trying to read lots of files from an iodrive 2 Duo (from one of the two modules)

### Environment information
1. 5.3.10-arch1-1
2. gcc (GCC) 9.2.0
3. Arch Linux 
4. 5.1.28

```
[ 7221.084625] CPU: 10 PID: 511 Comm: fct1-worker Tainted: G        W  OE     5.3.10-arch1-1 #1
[ 7221.084626] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./X399 Taichi, BIOS P3.80 08/27/2019
[ 7221.084629] Call Trace:
[ 7221.084630]  dump_stack+0x5c/0x80
[ 7221.084633]  debug_smp_processor_id.cold+0x4e/0x53
[ 7221.084644]  kfio_disk_stat_write_update+0x83/0x150 [iomemory_vsl]
[ 7221.084659]  ifio_1e5a3.5e62afc90b8d1b25ae145e583e480994677.3.2.16.1731+0xac9/0x1d00 [iomemory_vsl]
[ 7221.084662]  ? __switch_to_asm+0x40/0x70
[ 7221.084664]  ? __switch_to_asm+0x34/0x70
[ 7221.084667]  ? __switch_to_asm+0x40/0x70
[ 7221.084669]  ? __kprobes_text_end+0xfdf00/0xfdf00
[ 7221.084671]  ? irq_entries_start+0x5d8/0x660
[ 7221.084674]  ? __switch_to_asm+0x40/0x70
[ 7221.084691]  ? ifio_1e5a3.5e62afc90b8d1b25ae145e583e480994677.3.2.16.1731+0x1ccd/0x1d00 [iomemory_vsl]
[ 7221.084707]  ? ifio_fbeca.b91e0233dee7fc9112bb37f58c4e526bcd8.3.2.16.1731+0x808/0x2430 [iomemory_vsl]
[ 7221.084710]  ? preempt_count_add+0x68/0xa0
[ 7221.084713]  ? _raw_spin_lock_irqsave+0x26/0x50
[ 7221.084715]  ? wait_woken+0x70/0x70
[ 7221.084731]  ? ifio_5fb65.a9c484151da25a9eb60ef9a6e7309d1a95f.3.2.16.1731+0x10c/0x350 [iomemory_vsl]
[ 7221.084734]  ? _raw_spin_lock_irqsave+0x26/0x50
[ 7221.084737]  ? kthread+0xfb/0x130
[ 7221.084752]  ? ifio_f0cee.039ff5a1c1f314a171e2a8a425361a5875d.3.2.16.1731+0x60/0x60 [iomemory_vsl]
[ 7221.084754]  ? kthread_park+0x80/0x80
[ 7221.084756]  ? ret_from_fork+0x22/0x40
```

snuf, 2019-11-12:
@sotiris-bos thanks for logging this!

Looks like something is not going well with the stats counters there, coincidentally I found a bug where it seems to mess up the counter every now and then. This dump seems to be a locking issue of the thread. I'll see if I can reproduce, and also fix the counter bug.

Do you happen to know if it's the simultaneous access that causes it, or is it sequential, and is there a simple way to reproduce?

sotiris-bos, 2019-11-12:
Hey! I am not sure what causes it, I was trying to tar + pigz a directory with lots of subdirectories and files (100GB+) while simultaneously sending the data to my NAS with nc.

snuf, 2019-11-12:
Kay, that is good info! Are there any other *side-effects* you noticed?

sotiris-bos, 2019-11-12:
Not really, other than the file transfer stopping and starting a bunch of times. I think that was related to this but it might have been my network instead.

I tried to copy the data back from the NAS to the iodrive and compare the original to the copy to check for corruption. I ran some checksums on some files and they were all wrong but my tar + pigz + nc command was sketchy at best and the one I ran on the NAS to copy the files back was even worse, so I am not sure if it was me that caused it or not.

I will go ahead and try to properly tar the directory, copy it to the other fio (I have an iodrive 2 Duo), copy it back, untar and check again.

snuf, 2019-11-12:
@sotiris-bos  just ran a test on archlinux with the 5.3.10 kernel, and I get that error as soon as I insert the module and mount the filesystem. I've tested the module with 5.3.6 on Debian and don't see this error popping up. So it's something specific that changed in the kernel it seems. I've tested with putting a return at the top of the **kfio_disk_stat_write_update** function, to disable the stats write section and that seems to suppress the problem from occurring.

snuf, 2019-11-13:
@sotiris-bos the problem is somewhat related to #16 and I have a temporary fix I'm vetting atm. It's in the *KFIOC_BIO_HAD_BIO_SEGMENTS* area of kblock.c. It seems that when a DMA mapping/unmapping takes place above a certain size things go off, and it corrupts the write, as stated in #16. Oddly enough, there it bombs on directio, here we hit the page_cache. Am testing the boundaries of what I'm seeing, for now it seems that the bug is in:
```
if (bio_segments(bio) > queue_max_segments_queue))
``` 
Which should be
```
if (bio_segments(bio) >= queue_max_segments_queue))
```
Horrible oversight, and seems to be related to changes in the kernel, which is related to #16 conceptually. Am testing the outcome of the patch and adding two tests that cover this in the future.

snuf, 2019-11-13:
@sotiris-bos I've also fixed the cause of the *dmesg* error with the stack_dump, it was related to a lock that before 5.3.10 didn't show any issues. It's logical that it does, and should but didn't. The placement of  the PART_STAT_REQUIRES_CPU check in *kblock.c* was in the wrong place, both  ecab711230d6a279650195c6a75d43807f4c3b4c and 02cac8e145cd4126c5ae1c0499c8ea91d31ffdd5 are the changes. The latter did not show any issue in 5.3.6, nor 5.3.10, but did show up in 5.3.11.

sotiris-bos, 2019-11-13:
Nice, thank you for taking the time to work on this! I will test as soon as I find some time.

Sorry for not being able to assist, I am not really familiar with testing and troubleshooting storage devices and drivers, I just happened to stumble upon the errors.

Is this something that could be a problem on the next_generation branch as well? Just asking because I am looking to move the device to a kernel v4.xx machine (probably CentOS).

snuf, 2019-11-14:
@sotiris-bos you're welcome and no problem. It was actually good to work on as it added some things to the test suite.

This 5.1.28 branch is a fork of the next_generation branch, and will be merged back into the next_generation branch at some point. Right now the next_generation branch will supports up to 5.0 if I remember correctly. While the 5.1.28 branch in actuality should supports everything up to 5.4.

The next_generation branch contains a lot of fixes. However the kernel keeps evolving and things change for us mostly in the BIO and QUEUE sub systems. In later kernel versions things may behave as they should, or may not depending on what changed, even in minor-minor versions we see this over 5.3.6->5.3.10->5.3.11, were different problems pop up. On a 4.x kernel you have the luxury of choosing what you want to follow now. 

Ideally I need to write a script that does regression testing based on the test suite that is there now. With the components that are there it should be a matter of getting a bunch of distributions, and kernels in a list, and traverse over them with the test suite, in the hope that I won't wear out my MLC with the regression testing :)

sotiris-bos, 2019-11-14:
Updated to 5.3.11 before I updated the driver so I did not test it with 5.3.10. After updating to the latest commit, I am still getting the dmesg Call trace at boot:

```
[    6.810390] ------------[ cut here ]------------
[    6.810413] DEBUG_LOCKS_WARN_ON(val > preempt_count())
[    6.810421] WARNING: CPU: 16 PID: 423 at kernel/sched/core.c:3680 preempt_count_sub+0x5a/0x90
[    6.810469] Modules linked in: sd_mod sr_mod cdrom hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid ahci libahci crc32c_intel libata xhci_pci iomemory_vsl(OE+) xhci_hcd scsi_mod nct6775 hwmon_vid vfio_pci irqbypass vfio_virqfd vfio_iommu_type1 vfio
[    6.810553] CPU: 16 PID: 423 Comm: fio-wq/16 Tainted: G           OE     5.3.11-arch1-1 #1
[    6.810582] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./X399 Taichi, BIOS P3.80 08/27/2019
[    6.810616] RIP: 0010:preempt_count_sub+0x5a/0x90
[    6.810634] Code: e0 b5 5e c3 e8 57 0c 38 00 85 c0 74 f6 8b 15 3d c0 28 01 85 d2 75 ec 48 c7 c6 7e 80 4c a2 48 c7 c7 80 0f 4b a2 e8 a5 4a fd ff <0f> 0b c3 84 c0 75 c9 e8 2a 0c 38 00 85 c0 74 c9 8b 05 10 c0 28 01
[    6.810696] RSP: 0018:ffffa256446f7c08 EFLAGS: 00010286
[    6.810715] RAX: 0000000000000000 RBX: ffff8a242b888800 RCX: 0000000000000000
[    6.810739] RDX: 0000000000000001 RSI: ffffffffa2cb0a6a RDI: 0000000000000246
[    6.810763] RBP: 0000000000000000 R08: 0000000195eea908 R09: 000000000000002a
[    6.810788] R10: ffffffffa2cb0e20 R11: 00000000a2cb0a53 R12: ffffc2523e6819b0
[    6.810812] R13: ffffc2523e6819b0 R14: 0000000000000200 R15: ffff8a28235681f8
[    6.810837] FS:  0000000000000000(0000) GS:ffff8a242e280000(0000) knlGS:0000000000000000
[    6.810865] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    6.810885] CR2: 000055e9874fe058 CR3: 000000065940a000 CR4: 00000000003406e0
[    6.810909] Call Trace:
[    6.810937]  kfio_disk_stat_read_update+0x12b/0x150 [iomemory_vsl]
[    6.810972]  ifio_015db.f4472ed7df1b82d80c0d8d7eb68abe47d24.3.2.16.1731+0x23d/0x1260 [iomemory_vsl]
[    6.811001]  ? __switch_to_asm+0x40/0x70
[    6.811014]  ? __switch_to_asm+0x34/0x70
[    6.811027]  ? __switch_to_asm+0x40/0x70
[    6.811053]  ? ifio_015db.f4472ed7df1b82d80c0d8d7eb68abe47d24.3.2.16.1731+0xa46/0x1260 [iomemory_vsl]
[    6.811081]  ? __switch_to_asm+0x40/0x70
[    6.811093]  ? __switch_to_asm+0x34/0x70
[    6.811106]  ? __switch_to_asm+0x40/0x70
[    6.811118]  ? __switch_to_asm+0x34/0x70
[    6.811128]  ? __switch_to_asm+0x40/0x70
[    6.811138]  ? _raw_spin_unlock_irq+0x1d/0x30
[    6.811150]  ? finish_task_switch+0x80/0x280
[    6.811162]  ? __switch_to+0x158/0x460
[    6.811173]  ? __switch_to_asm+0x34/0x70
[    6.811183]  ? preempt_count_add+0x68/0xa0
[    6.811206]  ? ifio_1e5a3.5e62afc90b8d1b25ae145e583e480994677.3.2.16.1731+0xac9/0x1d00 [iomemory_vsl]
[    6.811228]  ? schedule_timeout+0x44/0x3d0
[    6.811252]  ? ifio_6cc94.ad1bb35854b8fe9d4a3108a40224ef93208.3.2.16.1731+0x495/0x640 [iomemory_vsl]
[    6.811274]  ? preempt_count_add+0x68/0xa0
[    6.811285]  ? _raw_spin_lock_irqsave+0x26/0x50
[    6.811297]  ? kthread+0xfb/0x130
[    6.811318]  ? ifio_6cc94.ad1bb35854b8fe9d4a3108a40224ef93208.3.2.16.1731+0x1b0/0x640 [iomemory_vsl]
[    6.811340]  ? kthread_park+0x80/0x80
[    6.811350]  ? ret_from_fork+0x22/0x40
[    6.811360] ---[ end trace 1135c477775e6000 ]---
```

Also, is there a way to update the Driver version build number shown under fio-status so that we don't get confused between changes?

snuf, 2019-11-14:
@sotiris-bos I can't reproduce this unfortunately. I've built the module from commit 02cac8e with DKMS on archlinux with kernel 5.3.11 and don't see this when rebooting multiple times, my dmesg and journalctl are clean. Could you double check the timestamp on your module, just to make sure? 

Unfortunately updating the version as seen in fio-status is a bit more involved, as it seems to be hardcoded in the binary *kfio/x86_64_ccXX_libkfio.o* files. I'll have a look if we can patch the kernel module itself after linking to contain it though.

sotiris-bos, 2019-11-14:
Yeah I am definitely on commit 02cac8e145cd4126c5ae1c0499c8ea91d31ffdd5 and built the package with DKMS but might not have done it properly.

I just removed the module, rebuilt it and loaded it and I did not get any (bad) messages. I am unable to reboot the system right now but I will do so once I get the chance and update.

snuf, 2019-11-15:
There are a couple of options for implementing the versioning. The easiest way would be to replace the second part of the output of */proc/fusion/version* and override the version that comes hard-coded from the *.o* libraries, which would look as follows: 
```
[vagrant@fio iomemory-vsl-3.2.16]$ cat /proc/fusion/version 
3.2.16.1731 02cac8e 
```
This would not change what's in fio-status though, changing that may be a bit harder, I'll give it a try if I have time.

sotiris-bos, 2019-11-15:
Nice! Anything that helps with identifying the versioning will be useful.

As an update, it was my fault of not upgrading the driver correctly that caused the call trace. After doing it properly, I am no longer seeing errors! Sorry for the trouble..

snuf, 2019-11-15:
Yeah it does make sense =) I've played around a bit and am going to drop the username before the *@* sign, and move the original hash forward and parse the commit hash behind it resulting in:
```
[vagrant@fio iomemory-vsl-3.2.16]$ cat /proc/fusion/version 
3.2.16.1731 4870ad45b7ea   02cac8e
```

No problem at all, better safe than sorry right? I'll keep this open till I've put the version change for the binary in.

Thanks for the help, much appreciated!

snuf, 2019-11-26:
tl;dr: When in the git repo it's easy to pick up the SHA1 sum and use it, however with DKMS that context is gone, getting that context in requires a pre-commit or post-checkout hook, still looking into that.

Hmm I've fiddled around with this a bit, it's not optimal. Either someone needs to add a post-checkout locally, or I need to have a pre-commit hook that I put in every time I newly check out. Kind of sad that Github doesn't allow you to propagate those, and that the $Id$ tag doesn't really do what you'd want it to. I've been exploring of extending the Makefile with some capability to checksum all the files and base it off of that, before it starts compiling, but don't know if that's desirable. Need to spend a bit more time on it I guess. The real problem is when it's a DKMS module, as the git context is gone then, while in a home directory you can just get the SHA1 of the commit you're on.

snuf, 2020-02-07:
fixed in [aad6d57](https://github.com/snuf/iomemory-vsl/commit/aad6d57eaf583647572a7126230cc2ecb042c22e). 

Modified the Makefile to:
1) Add the module version to the license.c before compiling, resulting in a version string on modinfo:
```
modinfo ./iomemory-vsl.ko  | grep version
version:        d663292
```

2) patch the module binary to reflect the version that is in use.
```
dmesg | grep "ioDrive driver"
old:
[174199.774438] <6>fioinf ioDrive driver 3.2.16.1731 pinnacles@4870ad45b7ea loading...
[174319.199644] <6>fioinf ioDrive driver unloading...
[174319.285798] <6>fioinf ioDrive driver unloaded...

new:
[175631.667841] <6>fioinf ioDrive driver 3.2.16.1731 4870ad45b7ea d663292 a loading...
```

The module version is derived from the commit hash HEAD is at.

closed 2020-02-07 by snuf