RemixVSL/iomemory-vsl #54: Compile against 5.0.21-5-pve Kernel (Proxmox) fails

qdrop17, 2019-11-26:
### Bug description
When compiling against 5.0.21-5-pve Kernel (Proxmox), the process fails with numerous errors. This is both the case for the official driver release (iomemory-vsl4-4.3.6.1173) as well as the driver right here (3.2.15 and 3.2.16)

I was able to compile against an up-to-date Ubuntu 18.04 Kernel though: https://github.com/snuf/iomemory-vsl/issues/48#issuecomment-547494739

### How to reproduce
Follow the instructions as documented here in the repo.

### Logfiles

[iomemory-vsl4-3.2.15.log](https://github.com/snuf/iomemory-vsl/files/3893153/iomemory-vsl4-3.2.15.log)
[iomemory-vsl4-4.3.6.1173.log](https://github.com/snuf/iomemory-vsl/files/3893155/iomemory-vsl4-4.3.6.1173.log)

Any help to get this working is highly appreciated. I just don't have to skills to debug this kind of software.

Best

qdrop

snuf, 2019-11-26:
@qdrop17 thanks for filing!

It seems like you're looking for the iomemory-vsl4 driver, what device do you have? The driver here is for iomemory-vsl devices prior to their 4th generation.

Looking at the errors you're getting the [5.1.28](https://github.com/snuf/iomemory-vsl/tree/5.1.28) branch will compile, however if your device is a 4th gen device this driver will not work with it this [iomemory-vsl4](https://github.com/naamval/iomemory-vsl4) driver should. The fixes however are reasonably portable between the two generations. Unfortunately I don't have a 4th gen device, as I would be able to test it and maintain that somewhat besides the 3rd gen driver.

Let me know if this helps

qdrop17, 2019-11-27:
@snuf thanks for your help!

I was indeed able to compile branch 5.1.28 :-) And you were right: I need a 4th-gen driver. Are there instructions on how to fix the official release (iomemory-vsl4-4.3.6.1173) so it compiles against this 5.0.21-5-pve Kernel? Unfortunately I didn't manage to build it after copying some files:

```
cp /usr/src/iomemory-vsl4-4.3.6/kfio/x86_64_cc63_libkfio.o_shipped /usr/src/iomemory-vsl4-4.3.6/kfio/x86_64_cc83_libkfio.o_shipped
cp /usr/src/iomemory-vsl4-4.3.6/kfio/.x86_64_cc63_libkfio.o.cmd /usr/src/iomemory-vsl4-4.3.6/kfio/.x86_64_cc83_libkfio.o.cmd
```
...there are still issues with building kblock.o and kcpu.o. I attached the logs here:

[vsl4-4.3.6.1173.log](https://github.com/snuf/iomemory-vsl/files/3896124/vsl4-4.3.6.1173.log)

Any help is highly appreciated.

snuf, 2019-11-29:
Generally the process is to add conditional tests through kfio_config.sh, these conditional tests drive the flags that switch between behavior in the source code. The main problem is that when creating a patch that patch needs to be tested quite thoroughly. Not testing leads to unexpected behavior (read data loss), especially when it includes changes to queueing and block io. Having said that the iomemory-vsl driver here contains most of the patches required for the iomemory-vsl 4 driver, which can be copied 1 to 1, however there are some minor differences which require changes that are unique.

I've ordered an SX300 from ebay, it however won't arrive before Dec 7. I've also obtained the iomemory-vsl4 source and will have a look. As said before, most changes are 1 to 1, there are a couple that may require some more digging.

qdrop17, 2019-11-30:
Hi @snuf 

I the meantime I went for the older Proxmox-Version (5.4) which ships with a 4.15 kernel. I managed to compile the official drivers against this kernel without any issues. This is maybe the best and most stable solution until there are drivers release that are compatible with 5.x kernels.

Thank you very much for your support. We highly appreciate it!

Fever-Wits, 2019-12-01:
Hi @snuf

I have an HP 320GB SLC PCIe ioDrive Duo for ProLiant Servers (2 x 160GB)
I was able to compile the driver you modified for the following kernel 5.3.10-1-pve and kernel 5.0.21-5-pve.
I saw something strange in the following test:

root @ nasa: ~ # hdparm -tT / dev / fioa

/ dev / drawer:
  Timing cached reads: 17640 MB in 2.00 seconds = 8829.16 MB / sec
  HDIO_DRIVE_CMD (identify) failed: Inappropriate ioctl for device
  Timing buffered disk reads: 2574 MB in 3.00 seconds = 857.99 MB / sec

Is this normal ?
Should I be worried ?

Greetings

naamval, 2019-12-03:
> @qdrop17 thanks for filing!
>
> It seems like you're looking for the iomemory-vsl4 driver, what device do you have? The driver here is for iomemory-vsl devices prior to their 4th generation.
>
> Looking [...]

Hey, just to clarify: that driver doesn't work at all, unfortunately. It was my failed attempt at adapting it to build under kernel 5.0.xx (Proxmox), without any knowledge of kernels whatsoever. I left it there in case someone else could use it somehow.

If you're going to tweak the vsl4 drivers, I would of course be happy to test it (SX350 here).

snuf, 2019-12-03:
@Fever-Wits took a bit of figuring out, sorry for the delay. *hdparm-9.58* sends a command that it doesn't get an expected result for when identifying the device. Looking at the verbose output of *hdparm-9.54* the same command is issued, but only once, where it fails too, but only shows with verbose output enabled. In both cases the *hdparm -I*, the identification command fails, however in one case not visibly. Mind you that the FusionIO driver does not use the sub-system expected by hdparm, it's not an ATA or IDE device. I hope this answers your question?

@qdrop17 appreciate the offer, I'll see if I can get some coffee thing going at some point :)

@naamval Thanks for the info and offering to test. I'll stop referring to your repo, I've put an order in for an SX300, it's here this weekend so should be able to get something going after that. Once I have something that can be tested I'll definitely reach out to you.

Best,

Fever-Wits, 2019-12-03:
Hello @snuf 
thanks for the reply.
thanks for making it possible for me to still use my memory.
For now, the driver works fine with kernel 5.3.10-1-pve.
I'll write when I have a new kernel.

Greetings,

snuf, 2020-04-05:
@naamval finally had some time to sit down. I backported the iomemory-vsl changes and had to do some other modifications to get the iomemory-vsl4 driver to work on kernel 5.3. It's only tested on 5.3 for now and still needs some "cleaning", it does however seem to work. I've checked it in here: https://github.com/snuf/iomemory-vsl4

Am closing this ticket as the iomemory-vsl4 can move to its own space now :)

closed 2020-04-05 by snuf