RemixVSL/iomemory-vsl #56: next_generation branch compile error

Cavesnakes, 2020-02-04:
I'm using F31 with up to date kernel (5.4.15-200.fc31.x86_64)

previously I'm using F31 with 5.3.x kernel and next_generation branch works well

but 5.4.x kernel isn't 

full log is here
[error.txt](https://github.com/snuf/iomemory-vsl/files/4154117/error.txt)
 
I really appreciate your works

Sincerely a guy live in korea.

snuf, 2020-02-04:
Hey there,

Looking at the error.log it seems like you're missing both bison and flex:
``` 
 LEX     scripts/kconfig/lexer.lex.c
/bin/sh: flex: command not found
  YACC    scripts/kconfig/parser.tab.[ch]
/bin/sh: bison: command not found 
```
For the rest everything seems to compile.

Best,

Funs

Cavesnakes, 2020-02-05:
after I see your comment I install bison and flex package and trying compile but this errors occur

[error.txt](https://github.com/snuf/iomemory-vsl/files/4157149/error.txt)

snuf, 2020-02-05:
Yeah it seems like you're missing some more bits and pieces here and there, when I have some time I'll spin up a fedora31 test machine and see what I can figure out.

Cavesnakes, 2020-02-06:
I'll try when i get back home thanks for fast response

always, thanks about your work

Cavesnakes, 2020-02-06:
all packages that you say is already installed 

 full log is here
[error.txt](https://github.com/snuf/iomemory-vsl/files/4162978/error.txt)

Cavesnakes, 2020-02-06:
I installed newer kernel 5.4.17 and selinux library error is finally gone

 but this error remains

> /usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Makefile.build:37: /usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Build.include: No such file or directory

I investigated some fact about this error
looks like some missing file(or wrong location)

so I've entered this command

rpm -ql kernel-headers

/usr/include/asm
/usr/include/asm-generic
/usr/include/asm-generic/auxvec.h
/usr/include/asm-generic/bitsperlong.h
/usr/include/asm-generic/bpf_perf_event.h
/usr/include/asm-generic/errno-base.h
/usr/include/asm-generic/errno.h
/usr/include/asm-generic/fcntl.h
/usr/include/asm-generic/hugetlb_encode.h
/usr/include/asm-generic/int-l64.h
...(etc.)

kernel-headers package's file location is /usr/include/

what should i do?

snuf, 2020-02-06:
The kernel-devel package is required for the /usr/src/<kernel> stuff. 
```
[vagrant@fio ~]$ rpm -ql kernel-devel | grep usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Makefile
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Makefile.build
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Makefile.feature
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Makefile.include
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/feature
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/feature/Makefile
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/tests
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/tests/ex
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/tests/ex/Makefile
```

I'm using a clean Fedora 31 Cloud install in libvirt, and am unable to reproduce your problem [build.log](https://github.com/snuf/iomemory-vsl/files/4165437/build.log). The full command set and packages that get installed prior to rebooting and then checking out the module and using rpmbuild are:
```
    yum update && yum upgrade
    yum install -y kernel-headers kernel-devel gcc git make dkms fio \
      jq python2-pip rpm-build mlocate
```
I'm curious what's different about your install from a standard Fedora 31 Cloud image?

Oddly your message below did end up in my email, but not here
```
 installed newer kernel 5.4.17 and selinux library error is finally gone

but this error remains

/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Makefile.build:37: /usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Build.include: No such file or directory

I investigated some fact about this error
looks like some missing file(or wrong location)

so I've entered this command

rpm -ql kernel-headers

/usr/include/asm
/usr/include/asm-generic
/usr/include/asm-generic/auxvec.h
/usr/include/asm-generic/bitsperlong.h
/usr/include/asm-generic/bpf_perf_event.h
/usr/include/asm-generic/errno-base.h
/usr/include/asm-generic/errno.h
/usr/include/asm-generic/fcntl.h
/usr/include/asm-generic/hugetlb_encode.h
/usr/include/asm-generic/int-l64.h
...(etc.)

kernel-headers package's file location is /usr/include/

what should i do?
```

snuf, 2020-02-07:
@Cavesnakes Haven't heard back from you yet, all ok? Did it work out?

jbwyatt4, 2020-02-09:
Hi @snuf, not the person you asked, but also on Fedora 31 with 5.4.17-200.fc31.x86_64 Fedora kernel. Got past flex, bison, and elfutils-libelf-devel

Having the following issue after executing:

rpmbuild -ba fio-driver.spec

`  DESCEND  objtool
  HOSTCC  scripts/unifdef
  HOSTCC  scripts/selinux/genheaders/genheaders
  HOSTCC  scripts/selinux/mdp/mdp
scripts/selinux/genheaders/genheaders.c:18:10: fatal error: classmap.h: No such file or directory
   18 | #include "classmap.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[4]: *** [scripts/Makefile.host:107: scripts/selinux/genheaders/genheaders] Error 1
make[3]: *** [scripts/Makefile.build:509: scripts/selinux/genheaders] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/selinux/mdp/mdp.c:36:10: fatal error: classmap.h: No such file or directory
   36 | #include "classmap.h"
      |          ^~~~~~~~~~~~
compilation terminated.
/usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Makefile.build:37: /usr/src/kernels/5.4.17-200.fc31.x86_64/tools/build/Build.include: No such file or directory
`

I am new to Fedora so I may not have everything set up.

Cavesnakes, 2020-02-09:
> 
> 
> @Cavesnakes Haven't heard back from you yet, all ok? Did it work out?

No it's not but i'm trying...

jbwyatt4, 2020-02-09:
@snuf I also tried the 5.1.28 branch.

Got the following error:

Processing files: iomemory-vsl-source-3.2.16.1732-1.0.fc31.x86_64
Provides: config(iomemory-vsl-source) = 3.2.16.1732-1.0.fc31 iomemory-vsl iomemory-vsl-3.2.16.1732 iomemory-vsl-source = 3.2.16.1732-1.0.fc31 iomemory-vsl-source(x86-64) = 3.2.16.1732-1.0.fc31 libvsl libvsl.so()(64bit)
Requires(rpmlib): rpmlib(CompressedFileNames) <= 3.0.4-1 rpmlib(FileDigests) <= 4.6.0-1 rpmlib(PayloadFilesHavePrefix) <= 4.0-1
Requires: /usr/bin/sh libc.so.6()(64bit) libc.so.6(GLIBC_2.14)(64bit) libc.so.6(GLIBC_2.2.5)(64bit) libc.so.6(GLIBC_2.3)(64bit) libc.so.6(GLIBC_2.4)(64bit) libc.so.6(GLIBC_2.7)(64bit) libgcc_s.so.1()(64bit) libgcc_s.so.1(GCC_3.0)(64bit) libm.so.6()(64bit) libpthread.so.0()(64bit) libpthread.so.0(GLIBC_2.2.5)(64bit) librt.so.1()(64bit) libstdc++.so.6()(64bit) libstdc++.so.6(CXXABI_1.3)(64bit) libstdc++.so.6(CXXABI_1.3.1)(64bit) libstdc++.so.6(GLIBCXX_3.4)(64bit) libstdc++.so.6(GLIBCXX_3.4.11)(64bit) libstdc++.so.6(GLIBCXX_3.4.15)(64bit) libstdc++.so.6(GLIBCXX_3.4.9)(64bit) libuuid.so.1()(64bit) libuuid.so.1(UUID_1.0)(64bit) rtld(GNU_HASH)
Obsoletes: fio-driver fio-driver-source iodrive-driver iodrive-driver-source
Checking for unpackaged file(s): /usr/lib/rpm/check-files /root/rpmbuild/BUILDROOT/iomemory-vsl-3.2.16.1732-1.0.fc31.x86_64
error: Installed (but unpackaged) file(s) found:
   /usr/src/iomemory-vsl-3.2.16/patch_module_version.sh

RPM build errors:
    line 38: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-kmod, iodrive-driver, fio-driver
    Macro expanded in comment on line 42: %{name}-%{fio_tar_version}.tar.gz

    line 38: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-kmod, iodrive-driver, fio-driver
    Macro expanded in comment on line 42: %{name}-%{fio_tar_version}.tar.gz

    line 133: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver, fio-driver
    line 205: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-source, fio-driver-source
    line 207: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver, fio-driver
    Installed (but unpackaged) file(s) found:
   /usr/src/iomemory-vsl-3.2.16/patch_module_version.sh

snuf, 2020-02-09:
@Cavesnakes which branch are you using? Since of now next_generation, and 5.1.28 are merged. next_generation did not compile on 5.x kernels.

@jbwyatt4 that is a bug I just fixed, it is related to setting the versions in the binary and the module information. It's fixed in next_generation, as well as 5.1.28 (which will be removed shortly)

Cavesnakes, 2020-02-10:
> @Cavesnakes which branch are you using? Since of now next_generation, and 5.1.28 are merged. next_generation did not compile on 5.x kernels.
>
> @jbwyatt4 that is a bug I just fixed, it is related to [...]

i'm using next_generation branch which is compatible with 5.3.x kernel era (w/ F31)

Cavesnakes, 2020-02-10:
all error is disappeared after new updated next_generation branch

thanks for your great work :D

jbwyatt4, 2020-02-10:
@snuf compiled and installed successfully. Will test out the card once I put back into this system.

Btw, I know C and took operating systems in college. Do you have any newbie friendly Linux dev tasks you need to get done?

snuf, 2020-02-11:
@jbwyatt4 awesome, and thanks for the offer! 

I've made a project board, don't know if anything that is interesting to you is in there, but you can have a look if you want. 

Something that is not in there is the test framework which I've pushed to another repository today, it's a bunch of shell and some ruby that uses Vagrant to build and test PCI devices, in this case specifically the IOMemory stuff. It contains some container bits allowing you to fetch and build kernels, test multiple Linux varieties and devices simultaneous depending on available resources. This needs massive work on the comparing data output part, and some usability work probably, as I'm the only nutcase using it :)

closed 2020-02-11 by snuf

Cavesnakes, 2020-04-12:
I reopened this issue because simillar error occurs.

I've already installed glibc-static and glibc-devel
might something is missing that i can't find out

<script src="https://gist.github.com/Cavesnakes/fc6ee459e04ea4ee3e17639eab0406e7.js"></script>

snuf, 2020-04-12:
Can you please provide an error log, otherwise it’s fairly impossible to
help you.

Best

Cavesnakes, 2020-04-12:
where i could find the error log?

snuf, 2020-04-12:
@Cavesnakes was on mobile, problem is not that it doesn't compile it actually has a module:
```
  LD [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/iomemory-vsl.ko
```
The problem is the RPM build, which is a different problem.
```
RPM build errors:
    line 38: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-kmod, iodrive-driver, fio-driver
    Macro expanded in comment on line 42: %{name}-%{fio_tar_version}.tar.gz

    line 38: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-kmod, iodrive-driver, fio-driver
    Macro expanded in comment on line 42: %{name}-%{fio_tar_version}.tar.gz
    line 133: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver, fio-driver
    line 205: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-source, fio-driver-source
    line 207: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver, fio-driver
    Installed (but unpackaged) file(s) found:
   /usr/src/iomemory-vsl-3.2.16/kfio_config_add.sh
```

So the module compiles fine, rpmbuild is just unhappy.

Best,

Funs

snuf, 2020-04-12:
@Cavesnakes also in github it seams that placing the following breaks the lookup oddly enough :) : <script src="
 https://gist.github.com/Cavesnakes/fc6ee459e04ea4ee3e17639eab0406e7.js"></script>

snuf, 2020-04-12:
@Cavesnakes seems like ```/usr/src/iomemory-vsl-3.2.16/kfio_config_add.sh``` was missing from the spec file. I added it so should be able to rpmbuild again.

```
snuf@scipio:~/Documents/Sources/iomemory-vsl$ rpmbuild -ba fio-driver.spec
[-snip-]
Wrote: /home/snuf/rpmbuild/SRPMS/iomemory-vsl-3.2.16.1732-1.0.src.rpm
Wrote: /home/snuf/rpmbuild/RPMS/x86_64/iomemory-vsl-5.3.0-42-generic-3.2.16.1732-1.0.x86_64.rpm
Wrote: /home/snuf/rpmbuild/RPMS/x86_64/iomemory-vsl-config-5.3.0-42-generic-3.2.16.1732-1.0.x86_64.rpm
Wrote: /home/snuf/rpmbuild/RPMS/x86_64/iomemory-vsl-source-3.2.16.1732-1.0.x86_64.rpm
[-snip-]
```

Cavesnakes, 2020-04-12:
okay I'll try out soon 

thanks for fast response I always appreciate your great work :D

Cavesnakes, 2020-04-12:
It works! thank you!

snuf, 2020-04-12:
Welcome! and enjoy. For 5.6 there will probably be some work. Am working with @Tourman36 on a refactor of the [iomemory-vsl4](https://github.com/snuf/iomemory-vsl4) library at the moment. Hopefully that will all make it back to the iomemory-vsl driver.