RemixVSL/iomemory-vsl #58: Kernel 5.6 drops ioremap_nocache

MariusSchiffer, 2020-04-16:
### Bug description
Starting from kernel 5.6, ioremap_nocache was removed. 
According to https://www.virtualbox.org/ticket/19312 ioremap is uncached by default.

### How to reproduce
Normal compilation with kernel 5.6 headers.

### Possible solution
Replace ioremap_nocache with ioremap.

### Environment information
1. Linux kernel compiled against (uname -a)
5.6.4
2. The C compiler version used (gcc --version)
gcc 9.3.0
3. distribution, and version 
Archlinux
4. Branch of iomemory-vsl that is being compiled
next-generation

snuf, 2020-04-16:
Hey @MariusSchiffer thanks for the bug report, there are fixes in https://github.com/snuf/iomemory-vsl4 code-cleanup which I've been working on with @Tourman36, I'll take those and put them in.

snuf, 2020-04-16:
@MariusSchiffer fix is merged to next_generation in #60, tests ran clean.

Thanks for reporting!

closed 2020-04-18 by snuf