RemixVSL/iomemory-vsl #61: Build failure on CentOS 7

plappermaul, 2020-05-02:
Hi,

coming back here after a long time. For some reason CentOS 7 compilation is broken in the fio-3.2.16.1732 branch.

```
In file included from /root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/linux/ktypes.h:48:0,
                 from /root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/ktypes.h:76,
                 from /root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/kfio.h:37,
                 from /root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/dbgset.h:33,
                 from /root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/errno.c:37:
/root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/common-linux/commontypes.h:54:47: error: 'x' undeclared here (not in a function)
 #define C_ASSERT(X) extern int __C_ASSERT__ [(x)?1:-1]
                                               ^
/root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/gcc/align.h:40:4: note: in expansion of macro 'C_ASSERT'
    C_ASSERT(__alignof__(structure) == (align_bytes))
    ^
/root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/kcondvar.h:109:1: note: in expansion of macro 'FUSION_STRUCT_ALIGN_VERIFY'
 FUSION_STRUCT_ALIGN_VERIFY(8, fusion_condvar_t);
```

```
# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/4.8.5/lto-wrapper
Target: x86_64-redhat-linux
Configured with: ../configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-bootstrap --enable-shared --enable-threads=posix --enable-checking=release --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-linker-hash-style=gnu --enable-languages=c,c++,objc,obj-c++,java,fortran,ada,go,lto --enable-plugin --enable-initfini-array --disable-libgcj --with-isl=/builddir/build/BUILD/gcc-4.8.5-20150702/obj-x86_64-redhat-linux/isl-install --with-cloog=/builddir/build/BUILD/gcc-4.8.5-20150702/obj-x86_64-redhat-linux/cloog-install --enable-gnu-indirect-function --with-tune=generic --with-arch_32=x86-64 --build=x86_64-redhat-linux
Thread model: posix
gcc version 4.8.5 20150623 (Red Hat 4.8.5-36) (GCC)
```

plappermaul, 2020-05-02:
fixed in 3.12.16 branch. maybe update of other branches needed

snuf, 2020-05-02:
cool, yes clearly a typo there, will cherry pick it to master

snuf, 2020-05-02:
cherry picked and put in master!

closed 2020-05-02 by snuf