RemixVSL/iomemory-vsl #62: KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS determination fails

plappermaul, 2020-05-02:
I have a gcc 4.8.5 and want to compile the 3.2.16 branch for kernel 4.14. This results in

```
/root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfile.c: In function 'kfio_create_proc_fops_entry':
/root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfile.c:240:55: warning: passing argument 4 of 'proc_create_data' from incompatible pointer type [enabled by default]
     return proc_create_data(name, mode, base, (struct proc_ops *)fops, data);
                                                       ^
In file included from /root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfile.c:37:0:
./include/linux/proc_fs.h:27:31: note: expected 'const struct file_operations *' but argument is of type 'struct proc_ops *'
 extern struct proc_dir_entry *proc_create_data(const char *, umode_t,
                               ^
/root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfile.c: In function 'kfio_set_file_ops_llseek_handler':
/root/rpmbuild1/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfile.c:327:30: error: dereferencing pointer to incomplete type
     ((struct proc_ops *)pops)->proc_lseek = (file_ops_llseek_fn)llseek;

```
It seems as if test KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS does not work as expected.

snuf, 2020-05-02:
This can be removed for fio-3.2.16, as it's part of the support for 5.x and up.

plappermaul, 2020-05-02:
fixed in last commit for branch 3.2.16

thanks for your hard work.

snuf, 2020-05-02:
check will have a look, the check is however correct as the behavior checks for 5.4 and up, whereas 5.3 and down this check should fail to go down the correct ifdef path.

Appreciate the fixes btw, and happy to see you here again :)

snuf, 2020-05-02:
oh shit I just read the error, correctly aaaaah got it....

snuf, 2020-05-02:
The breaking of the check on 3.16 is fine, on 5.3 and higher the test will now complain about an unused variable ops, which is not relevant though as we're planning for only master to be maintained. The failure is however now on `proc_ops ops`, and not on the `proc_create_data` on pre 5.4 kernels:
```
ANTS_PROC_OPS/kfioc_test.c: In function ‘kfioc_has_proc_create_data’:
/home/snuf/Documents/Sources/fusion/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.c:8:27: error: storage size of ‘ops’ isn’t known
     const struct proc_ops ops;
                           ^~~
/home/snuf/Documents/Sources/fusion/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.c:10:44: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Werror=incompatible-pointer-types]
     return proc_create_data(NULL, 0, NULL, pops, NULL);
                                            ^~~~
```
Effectively testing two things instead of the input for `proc_create_data`, whereas the 5.6 kernel warns on both of these, and previously only on one:
```
/home/vagrant/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.c: In function ‘kfioc_has_proc_create_data’:
/home/vagrant/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.c:8:27: warning: unused variable ‘ops’ [-Wunused-variable]
    8 |     const struct proc_ops ops;
      |                           ^~~
/home/vagrant/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.c:10:12: warning: ‘pops’ is used uninitialized in this function [-Wuninitialized]
   10 |     return proc_create_data(NULL, 0, NULL, pops, NULL);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  MODPOST 1 modules
```

closed 2020-05-02 by snuf