RemixVSL/iomemory-vsl #64: CentOS 8.1 failed rpmbuild with all branches

Cavesnakes, 2020-05-15:
I think it something go wrong hmm..

Tourman36, 2020-05-15:
can you post the build output? what fails?

Cavesnakes, 2020-05-15:
https://gist.github.com/Cavesnakes/de6c0b1008d01ff61b87d4eab8655409

Cavesnakes, 2020-05-15:
fio-3.2.16 branch: https://gist.github.com/Cavesnakes/e7c5f982106f3a739bd0bb3afe947c55

snuf, 2020-05-15:
@Cavesnakes in the fio-3.2.16 branch this issue has been fixed with commit https://github.com/snuf/iomemory-vsl/commit/83f87b2ac44700efafb95e86d0a3442f53262e6a 3 days ago, did you do a git pull before building ?

On master 94bd6684ab8c8c30c1ad73620a74f5e19d4eb215 fixes the other issue.

Cavesnakes, 2020-05-15:
@snuf yes i purged older source and rpmbuild folder

snuf, 2020-05-15:
@Cavesnakes what does `git branch -v` say?

Cavesnakes, 2020-05-15:
@snuf sorry I just updated to 94bd668 i cannot identify branch's version

Cavesnakes, 2020-05-15:
compiling with 94bd668 is works well 
thanks for your fast response i always appreciate your great work

snuf, 2020-05-15:
@Cavesnakes no problem, thanks for the bug report and happy you're helping to improve the overall experience by creating issue reports :)

closed 2020-05-15 by snuf