RemixVSL/iomemory-vsl #65: Debian 10.4, kernel 4.19.98-1+deb10u1, build error

GARMTECH, 2020-05-15:
Dear Funs,

I have a kblock.c problem building latest release (v5.6.0) on Debian 10.4 with kernel 4.19.98-1+deb10u1. Any suggestions how to overcome this?

`kblock.c: In function ‘kfio_fetch_next_bio’:
kblock.c:222:16: error: passing argument 1 of ‘spin_lock_irq’ from incompatible pointer type [-Werror=incompatible-pointer-types] #define Q_LOCK &q->queue_lock`

`22:39:11 [root@node ~/iodrive/snuf/iomemory-vsl]$ make dpkg
cd root/usr/src/iomemory-vsl-3.2.16 && \
  	make clean
make[1]: Entering directory '/root/iodrive/snuf/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
make \
        -j8 \
	-C /lib/modules/4.19.0-8-amd64/build \
	FIO_DRIVER_NAME=iomemory-vsl \
	FUSION_DRIVER_DIR=/root/iodrive/snuf/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
	M=/root/iodrive/snuf/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
	EXTRA_CFLAGS+="-I/root/iodrive/snuf/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall -Werror" \
	KFIO_LIB=kfio/x86_64_cc83_libkfio.o_shipped \
	clean
make[2]: Entering directory '/usr/src/linux-headers-4.19.0-8-amd64'
  CLEAN   /root/iodrive/snuf/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/.tmp_versions
make[2]: Leaving directory '/usr/src/linux-headers-4.19.0-8-amd64'
rm -rf include/fio/port/linux/kfio_config.h kfio_config license.c
make[1]: Leaving directory '/root/iodrive/snuf/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
cd root/usr/src/iomemory-vsl-3.2.16 && \
	make patch_module_version
make[1]: Entering directory '/root/iodrive/snuf/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
grep: license.c: No such file or directory

Adding module version v5.6.0-3.2.16 to source license.c

make[1]: Leaving directory '/root/iodrive/snuf/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
# patch fio_version, fio_short_version in debian/fio_values
cd /root/iodrive/snuf/iomemory-vsl && \
	dpkg-buildpackage -rfakeroot --no-check-builddeps --no-sign
dpkg-buildpackage: warning: using a gain-root-command while being root
dpkg-buildpackage: info: source package iomemory-vsl
dpkg-buildpackage: info: source version 3.2.16.1732
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Support <support@fusionio.com>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source --before-build .
 fakeroot debian/rules clean
make[1]: Entering directory '/root/iodrive/snuf/iomemory-vsl'
dh_testdir
dh_testroot
rm -f build-arch-stamp build-indep-stamp configure-stamp
# Add here commands to clean up after the build process.
#/usr/bin/make clean
rm -rf \
	"/root/iodrive/snuf/iomemory-vsl/driver_source" \
	"/root/iodrive/snuf/iomemory-vsl/debian/fio" \
	/root/iodrive/snuf/iomemory-vsl/debian/iomemory-vsl-4.19.0-8-amd64 \
	/root/iodrive/snuf/iomemory-vsl/debian/iomemory-vsl-config-4.19.0-8-amd64 \
	/root/iodrive/snuf/iomemory-vsl/debian/iomemory-vsl-4.19.0-8-amd64.* \
	/root/iodrive/snuf/iomemory-vsl/debian/iomemory-vsl-config-4.19.0-8-amd64.* 
sed -r -i -e 's/^Package: iomemory-vsl(-config)?(-[.0-9]+.*)*$/Package: iomemory-vsl\1/;' debian/control
dh_clean
	rm -f debian/debhelper-build-stamp
	rm -rf debian/.debhelper/
	rm -f -- debian/iomemory-vsl.substvars debian/iomemory-vsl-config.substvars debian/iomemory-vsl-source.substvars debian/files
	rm -fr -- debian/iomemory-vsl/ debian/tmp/ debian/iomemory-vsl-config/ debian/iomemory-vsl-source/
	find .  \( \( \
		\( -path .\*/.git -o -path .\*/.svn -o -path .\*/.bzr -o -path .\*/.hg -o -path .\*/CVS -o -path .\*/.pc -o -path .\*/_darcs \) -prune -o -type f -a \
	        \( -name '#*#' -o -name '.*~' -o -name '*~' -o -name DEADJOE \
		 -o -name '*.orig' -o -name '*.rej' -o -name '*.bak' \
		 -o -name '.*.orig' -o -name .*.rej -o -name '.SUMS' \
		 -o -name TAGS -o \( -path '*/.deps/*' -a -name '*.P' \) \
		\) -exec rm -f {} + \) -o \
[... 164 lines cut ...]
         struct gendisk *gd = fgd->gd;
                         ^~
/root/iodrive/snuf/iomemory-vsl/driver_source/kblock.c: In function ‘kfio_alloc_queue’:
/root/iodrive/snuf/iomemory-vsl/driver_source/kblock.c:960:10: error: too few arguments to function ‘blk_alloc_queue_node’
     rq = blk_alloc_queue_node(GFP_NOIO, node);
          ^~~~~~~~~~~~~~~~~~~~
In file included from /root/iodrive/snuf/iomemory-vsl/driver_source/port-internal.h:63,
                 from /root/iodrive/snuf/iomemory-vsl/driver_source/kblock.c:32:
/usr/src/linux-headers-4.19.0-8-common/include/linux/blkdev.h:1282:23: note: declared here
 struct request_queue *blk_alloc_queue_node(gfp_t gfp_mask, int node_id,
                       ^~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[6]: *** [/usr/src/linux-headers-4.19.0-8-common/scripts/Makefile.build:308: /root/iodrive/snuf/iomemory-vsl/driver_source/kblock.o] Error 1
make[6]: *** Waiting for unfinished jobs....
make[5]: *** [/usr/src/linux-headers-4.19.0-8-common/Makefile:1537: _module_/root/iodrive/snuf/iomemory-vsl/driver_source] Error 2
make[4]: *** [Makefile:146: sub-make] Error 2
make[3]: *** [Makefile:8: all] Error 2
make[3]: Leaving directory '/usr/src/linux-headers-4.19.0-8-amd64'
make[2]: *** [Makefile:132: modules] Error 2
make[2]: Leaving directory '/root/iodrive/snuf/iomemory-vsl/driver_source'
ERROR:
make[1]: *** [debian/rules:96: build-arch-stamp] Error 1
make[1]: Leaving directory '/root/iodrive/snuf/iomemory-vsl'
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
make: *** [Makefile:11: dpkg] Error 2`

snuf, 2020-05-15:
@GARMTECH please use the [fio-3.2.16.1732](https://github.com/snuf/iomemory-vsl/tree/fio-3.2.16.1732) branch, the 5.6.0 release and master are only for 5 kernels and upwards. [fio-3.2.16.1732](https://github.com/snuf/iomemory-vsl/tree/fio-3.2.16.1732) is meant for kernels below 5.0, as stated in the README.md under [current_version](https://github.com/snuf/iomemory-vsl#current_version)

snuf, 2020-05-15:
@GARMTECH for clarity we've added a v4.20.0 tag which supports all kernels before 5.0. Also changed the wording in the README.md a bit which hopefully clarifies the difference between the tags / branches.  Hope this helps, and let us know if this solves your problem?

GARMTECH, 2020-05-15:
Thank you! Compiled with "git checkout fio-3.2.16.1732".

Single unrelated issue, but maybe you have faced such issue before and can help me right on spot (Unable to open CSR for device 'fct0'):
```
23:45:20 [root@node ~]$ fio-update-iodrive /usr/share/fio/firmware/fusion_3.2.16-20180821.fff
WARNING: DO NOT TURN OFF POWER OR RUN ANY IODRIVE UTILITIES WHILE THE FIRMWARE UPDATE IS IN PROGRESS
  Please wait...this could take a while

Updating: [                    ] (  0%) /
Error: Unable to open CSR for device 'fct0'.
```

Tourman36, 2020-05-15:
Swing by discord if you have questions, happy to help there.

https://discord.gg/hyfuXk

closed 2020-05-15 by Tourman36