RemixVSL/iomemory-vsl #66: Build failure on kernel 4.19 Centos 7.5

sibersoft, 2020-06-05:
Hello

I have problem with build package under CentOS 7.5 , kernel 4.19 
Use fio-3.2.16.1732 branch.

`make[1]: Entering directory `/usr/src/kernels/4.19.0+1-x86_64'
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/main.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/license.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/pci.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/sysrq.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/driver_init.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfio.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/errno.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/state.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfile.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kcache.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kmem.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfio_common.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kcpu.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kscatter.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/ktime.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/sched.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/cdev.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kcondvar.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kinfo.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kexports.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/khotplug.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kcsr.o
  SHIPPED /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfio/x86_64_cc48_libkfio.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/module_param.o
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c: In function 'kfio_create_disk':
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c:655:9: error: implicit declaration of function 'queue_flag_set_unlocked' [-Werror=implicit-function-declaration]
         queue_flag_set_unlocked(QUEUE_FLAG_DISCARD, rq);
         ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c:683:5: error: implicit declaration of function 'queue_flag_set' [-Werror=implicit-function-declaration]
     queue_flag_set(QUEUE_FLAG_WC, rq);
     ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c:705:5: error: implicit declaration of function 'queue_flag_clear_unlocked' [-Werror=implicit-function-declaration]
     queue_flag_clear_unlocked(QUEUE_FLAG_ADD_RANDOM, rq);
     ^
cc1: some warnings being treated as errors
make[2]: *** [/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [_module_/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source] Error 2
make[1]: Leaving directory `/usr/src/kernels/4.19.0+1-x86_64'
make: *** [modules] Error 2
make: Leaving directory `/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source'
+ set +x
ERROR:

error: Bad exit status from /var/tmp/rpm-tmp.3RzroA (%build)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.3RzroA (%build)`

snuf, 2020-06-05:
@sibersoft thanks for your report. I'm missing some details from it. Could you run a `git branch -v` and `find . -type f | xargs grep PORT_WERROR` and paste both outputs for me? 

The line in your original report that was important is `cc1: some warnings being treated as errors`. This is a compiler warning that it treats warnings as errors. The warnings in this case are "harmless". The compile fails due to the Makefile having this behavior enabled when PORT_WERROR is set. In v4.20.0 the release which is based on fio-3.2.16-1732 this was addressed at some point. With a fresh checkout I can't reproduce the issue. My guess is you were on an older checkout and the pull has merged and left this flag behind. The above commands are for trying to figure out if my guess is right in some way, if so the easiest way around is to do a fresh clone, and use build the package.

snuf, 2020-06-05:
Hold on I'm a derp, it should actually do that as the warnings were resolved for most cases and -Werror was switched back on again. You can disable the "treat warnings as errors" by setting PORT_WERROR=0. I'm unable to get the 4.19 kernel for CentOS anymore, elrepo only has 4.4.224-1 (which breaks), or 5.6.14. I do have a 4.19 kernel for debian / ubuntu which I'll have a look at.

snuf, 2020-06-05:
@sibersoft I can't reproduce this on the 4.19 debian / ubuntu, all goes smooth there. As mentioned before please provide `git branch -v` and `find . -type f | xargs grep PORT_WERROR` to see if we can figure out what the issue is.

sibersoft, 2020-06-08:
Hey, thanks for replay

git branch -v
`* fio-3.2.16.1732 2a18ca0 updated the README.md to be consistent when building normally
  master          6c1388d Update README.md`

find . -type f | xargs grep PORT_WERROR
`./root/usr/src/iomemory-vsl-3.2.16/Makefile:# Set PORT_WERROR=0 to disable treating warnings as errors
./root/usr/src/iomemory-vsl-3.2.16/Makefile:ifeq ($(PORT_WERROR),1)`

sibersoft, 2020-06-08:
git checkout
`M       root/usr/src/iomemory-vsl-3.2.16/check_target_kernel.conf
M       root/usr/src/iomemory-vsl-3.2.16/license.c`

On fresh checkout I did rpmbuild -ba fio-driver.spec but also same error

snuf, 2020-06-08:
@sibersoft I'm unable to get the 4.19 kernel for CentOS. Debian 4.19 compiles, so I'm not able to reproduce the problem. However if you turn off `-Werror` in the Makefile you should be able to compile (commenting out line 36).

Tourman36, 2020-06-08:
@sibersoft are you able to test with centos 8, or a different OS with a newer Kernel?

sibersoft, 2020-06-15:
@snuf Sorry for delay. I'v tried with disabling -Werror and didn't solved issue.

`/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c: In function kfio_create_disk:
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c:655:9: error: implicit declaration of function 'queue_flag_set_unlocked' [-Werror=implicit-function-declaration]
         queue_flag_set_unlocked(QUEUE_FLAG_DISCARD, rq);
         ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c:683:5: error: implicit declaration of function 'queue_flag_set' [-Werror=implicit-function-declaration]
     queue_flag_set(QUEUE_FLAG_WC, rq);
     ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c:705:5: error: implicit declaration of function 'queue_flag_clear_unlocked' [-Werror=implicit-function-declaration]
     queue_flag_clear_unlocked(QUEUE_FLAG_ADD_RANDOM, rq);`

@Tourman36 I can't test other kernel because it's xcp-ng virtualization system so thats a problem.

Tourman36, 2020-06-15:
@sibersoft Please post uname -r. Which version of XCP-ng are you running?

snuf, 2020-06-15:
@sibersoft please for next time state the distribution you're running. I've installed xcp-ng 8.1.0 and can reproduce the issue. The problem comes from the KFIOC_X_HAS_BLK_QUEUE_FLAG_OPS that fails on xcp-ng:
```
[08:32 xcp-ng-evngfewc KFIOC_X_HAS_BLK_QUEUE_FLAG_OPS]# make
make -C /lib/modules/4.19.0+1/build M=/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_HAS_BLK_QUEUE_FLAG_OPS EXTRA_CFLAGS='-Wall -Werror -Werror=frame-larger-than=4096' modules
make[1]: Entering directory `/usr/src/kernels/4.19.0+1-x86_64'
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_HAS_BLK_QUEUE_FLAG_OPS/kfioc_test.o
/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_HAS_BLK_QUEUE_FLAG_OPS/kfioc_test.c: In function ‘has_queue_blk_queue_flag_ops’:
/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_HAS_BLK_QUEUE_FLAG_OPS/kfioc_test.c:8:1: error: the frame size of 2464 bytes is larger than 1 bytes [-Werror=frame-larger-than=]
 }
 ^
cc1: all warnings being treated as errors
make[2]: *** [/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_HAS_BLK_QUEUE_FLAG_OPS/kfioc_test.o] Error 1
make[1]: *** [_module_/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_HAS_BLK_QUEUE_FLAG_OPS] Error 2
make[1]: Leaving directory `/usr/src/kernels/4.19.0+1-x86_64
```

snuf, 2020-06-15:
@sibersoft found the issue, it's related the way a flag doesn't work for a test with the GCC version that XCP-ng uses. It's fixed in v4.20.1 and in fio-3.2.16.1732.
```

make \
    -j4 \
-C /lib/modules/4.19.0+1/build \
FIO_DRIVER_NAME=iomemory-vsl \
FIO_SCSI_DEVICE=0 \
FUSION_DRIVER_DIR=/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
KBUILD_EXTMOD=/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
EXTRA_CFLAGS+="-I/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/include -DBUILDING_MODULE -DLINUX_IO_SCHED" \
INSTALL_MOD_DIR=extra/fio \
INSTALL_MOD_PATH= \
KFIO_LIB=kfio/x86_64_cc48_libkfio.o_shipped \
modules
make[1]: Entering directory `/usr/src/kernels/4.19.0+1-x86_64'
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/main.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/pci.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/license.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/sysrq.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/driver_init.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/errno.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/state.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kcache.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfile.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kmem.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_common.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kcpu.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kscatter.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/ktime.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/sched.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/cdev.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kcondvar.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kinfo.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kexports.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/khotplug.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kcsr.o
  SHIPPED /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio/x86_64_cc48_libkfio.o
  CC [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/module_param.o
  LD [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/iomemory-vsl.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: could not find /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio/.x86_64_cc48_libkfio.o.cmd for /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio/x86_64_cc48_libkfio.o
  CC      /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/iomemory-vsl.mod.o
  LD [M]  /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/iomemory-vsl.ko
make[1]: Leaving directory `/usr/src/kernels/4.19.0+1-x86_64'

Ok: altered version for iomemory-vsl.ko appended f3bb361

[09:35 xcp-ng-evngfewc iomemory-vsl-3.2.16]# 
```

snuf, 2020-06-16:
@sibersoft I'm closing this for now, let us know if it works if you use v4.20.1, or if you run into problems.

closed 2020-06-16 by snuf

sibersoft, 2020-06-16:
@snuf Works thanks a lot.