RemixVSL/iomemory-vsl #67: [INFO] Debian 10 with Kernel 4.19.0-9-amd64 adn PX600

nerdalertdk, 2020-06-23:
Hi

I just created this issue for other people that is in the same boat as me.
I have an PX600 (cisco branded) and they use the 4.X driver so i can not use this driver.

I ported some of you fixes to the official driver so it will install on Debian 10

Feel free to delete this issue if its not relevant, I found all the fixes her :)
thanks for you great work

https://github.com/nerdalertdk/iomemory-vsl4

nerdalertdk, 2020-06-23:
Search Optimized

ticks replaced with nsecs

has no member named ‘ticks’
current_kernel_time replaced with ktime_get_coarse_real_ts64

implicit declaration of function ‘current_kernel_time’; did you mean ‘current_kernel_time64’? [-Werror=implicit-function-declaration]
struct timespec ts = current_kernel_time();
MODULE_LICENSE "Proprietary" repalced with "GPL"

FATAL: modpost: GPL-incompatible module iomemory-vsl.ko uses GPL-only symbol 'ktime_get_real_seconds'

closed 2020-06-23 by nerdalertdk

snuf, 2020-06-23:
@nerdalertdk thanks for the update! There is a repository for [iomemory-vsl4](https://github.com/snuf/iomemory-vsl4). `make gpl` will let you compile the module with the GPL symbols on the default driver from Sandisk/WD up until you run out of support runway. The iomemory-vsl4 repo only contains fixes and cleanups from 5.x onwards.