RemixVSL/iomemory-vsl #71: kernel 4.19.0+1-x86_64 not support

learn0208, 2020-09-29:
env: xenserver 8.0 ddk
kernerl-version: 4.19.0+1-x86_64
make rpm error message:
`make[2]: Entering directory `/usr/src/kernels/4.19.0+1-x86_64'
printf '#include "linux/module.h"\nMODULE_LICENSE("GPL");\n' >/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/license.c
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/main.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/license.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/pci.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/sysrq.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/driver_init.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfio.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/errno.o
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/state.o
In file included from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/linux/ktypes.h:48:0,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/ktypes.h:76,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/state.h:37,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/state.c:30:
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/common-linux/commontypes.h:54:47: error: 'x' undeclared here (not in a function)
 #define C_ASSERT(X) extern int __C_ASSERT__ [(x)?1:-1]
                                               ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/gcc/align.h:40:4: note: in expansion of macro 'C_ASSERT'
    C_ASSERT(__alignof__(structure) == (align_bytes))
    ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/kcondvar.h:109:1: note: in expansion of macro 'FUSION_STRUCT_ALIGN_VERIFY'
 FUSION_STRUCT_ALIGN_VERIFY(8, fusion_condvar_t);
 ^
In file included from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/linux/ktypes.h:48:0,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/ktypes.h:76,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/kfio.h:37,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/dbgset.h:33,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/errno.c:37:
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/common-linux/commontypes.h:54:47: error: 'x' undeclared here (not in a function)
 #define C_ASSERT(X) extern int __C_ASSERT__ [(x)?1:-1]
                                               ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/gcc/align.h:40:4: note: in expansion of macro 'C_ASSERT'
    C_ASSERT(__alignof__(structure) == (align_bytes))
    ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/kcondvar.h:109:1: note: in expansion of macro 'FUSION_STRUCT_ALIGN_VERIFY'
 FUSION_STRUCT_ALIGN_VERIFY(8, fusion_condvar_t);
 ^
make[3]: *** [/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/errno.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/state.o] Error 1
In file included from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/linux/ktypes.h:48:0,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/ktypes.h:76,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/sysrq.c:34:
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/common-linux/commontypes.h:54:47: error: 'x' undeclared here (not in a function)
 #define C_ASSERT(X) extern int __C_ASSERT__ [(x)?1:-1]
                                               ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/gcc/align.h:40:4: note: in expansion of macro 'C_ASSERT'
    C_ASSERT(__alignof__(structure) == (align_bytes))
    ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/kcondvar.h:109:1: note: in expansion of macro 'FUSION_STRUCT_ALIGN_VERIFY'
 FUSION_STRUCT_ALIGN_VERIFY(8, fusion_condvar_t);
 ^
make[3]: *** [/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/sysrq.o] Error 1
In file included from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/linux/ktypes.h:48:0,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/ktypes.h:76,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/kfio.h:37,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/driver_init.c:32:
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/common-linux/commontypes.h:54:47: error: 'x' undeclared here (not in a function)
 #define C_ASSERT(X) extern int __C_ASSERT__ [(x)?1:-1]
                                               ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/gcc/align.h:40:4: note: in expansion of macro 'C_ASSERT'
    C_ASSERT(__alignof__(structure) == (align_bytes))
    ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/kcondvar.h:109:1: note: in expansion of macro 'FUSION_STRUCT_ALIGN_VERIFY'
 FUSION_STRUCT_ALIGN_VERIFY(8, fusion_condvar_t);
 ^
make[3]: *** [/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/driver_init.o] Error 1
[... 30 lines cut ...]
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/dbgset.h:33,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfio.c:51:
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/common-linux/commontypes.h:54:47: error: 'x' undeclared here (not in a function)
 #define C_ASSERT(X) extern int __C_ASSERT__ [(x)?1:-1]
                                               ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/gcc/align.h:40:4: note: in expansion of macro 'C_ASSERT'
    C_ASSERT(__alignof__(structure) == (align_bytes))
    ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/include/fio/port/kcondvar.h:109:1: note: in expansion of macro 'FUSION_STRUCT_ALIGN_VERIFY'
 FUSION_STRUCT_ALIGN_VERIFY(8, fusion_condvar_t);
 ^
make[3]: *** [/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kfio.o] Error 1
make[2]: *** [_module_/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source] Error 2
make[2]: Leaving directory `/usr/src/kernels/4.19.0+1-x86_64'
make[1]: *** [modules] Error 2
make[1]: Leaving directory `/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source'
+ set +x
ERROR:

error: Bad exit status from /var/tmp/rpm-tmp.Igm3Qw (%build)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.Igm3Qw (%build)
make: *** [rpm] Error 1
`

snuf, 2020-09-29:
@learn0208 which branch is this on, is this the v4.20.1 release?

snuf, 2020-10-13:
stale.

closed 2020-10-13 by snuf