RemixVSL/iomemory-vsl #72: Can't seem to make it work

PouletteMC, 2020-10-23:
Hey ! 

first of all thank you for keeping thath up to date :)

I have a fusion ioDrive2 (410gb) and tried to install this on a fresh Ubuntu server 20.04 lts.
I had it working once previously but had to reinstall

I get no errors and yet the drive does not show up

snuf, 2020-10-23:
Hey @PouletteMC can you elaborate on what you've tried/done? Can you share your `lsmod` and dmesg?

PouletteMC, 2020-10-23:
thanks for answering so fast

So far I've followed the guide and tried doing it from the package I downloaded from the website

here is my lsmod https://pastebin.com/Fs6QkzQw

and my dmesg https://pastebin.com/HGZF1JUW

I'm kinda new to linux so sorry if I send useless stuff

snuf, 2020-10-23:
@PouletteMC the driver is not installed on the host, have you followed the installation steps?

PouletteMC, 2020-10-23:
yes I have, I followed the 4 setps you have for installing on debian

snuf, 2020-10-23:
did you `dpkg -i` the debian package that came out of it ? There is caveat when using .deb packages that if the kernel is updated you have to rebuild the .deb package and remove the old package and install the new package.

PouletteMC, 2020-10-23:
it creates 3 packages, I did `sudo` dpkg -i iomemory-vsl-5.4.0-52-generic_3.2.16.1732_amd64.deb`

snuf, 2020-10-23:
and you're running 5.4.0-52 (`uname -a`) ?

PouletteMC, 2020-10-23:
yes i am

snuf, 2020-10-23:
can you do a `find /lib/modules/5.4.0-52-generic/ -name iomemory-vsl.ko`, btw do you have discord? you can join [here](https://discord.gg/TeWPbfM), might be easier than this.

PouletteMC, 2020-10-23:
the command doesn't return anything

and discord says the link is invalid

PouletteMC, 2020-10-23:
I tried using the commands listed for "Source" instead of ".deb" and it worked

snuf, 2020-10-23:
Sorry Link is [updated](https://discord.gg/NebrDJb),could you run the .deb sequence and paste the output?

closed 2020-10-23 by PouletteMC