RemixVSL/iomemory-vsl #75: [BUG] rpm build errors on fedora 33 - kernel 5.9.16

dougalII, 2020-12-30:
### Bug description
_I know my kernel version is not supported here.. but.._
I have tried`fio-3.2.16.1732` i get:
`error: 'struct mm_struct' has no member named 'mmap_sem'; did you mean 'mmap_base'?`
from a quick google, there seems to have been a change from kernel ~5.8 which breaks this, others online seem to switch from `mmap_sem` to `mm`?

On `master`  and  `v5.6.0` 
i am getting errors with:
`error: 'enum pci_channel_state' declared inside parameter list will not be visible outside of this definition or declaration [-Werror]`

### How to reproduce
Branch `fio-3.2.16.1732` 
```
[david@localhost iomemory-vsl]$ git checkout fio-3.2.16.1732
Switched to branch 'fio-3.2.16.1732'
Your branch is up to date with 'origin/fio-3.2.16.1732'.
[david@localhost iomemory-vsl]$ rpmbuild -ba fio-driver.spec
warning: line 38: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-kmod, iodrive-driver, fio-driver
warning: Macro expanded in comment on line 42: %{name}-%{fio_tar_version}.tar.gz

warning: line 38: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-kmod, iodrive-driver, fio-driver
warning: Macro expanded in comment on line 42: %{name}-%{fio_tar_version}.tar.gz

warning: line 133: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver, fio-driver
warning: line 205: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-source, fio-driver-source
warning: line 207: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver, fio-driver
setting SOURCE_DATE_EPOCH=1491782400
Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.SK0Oye
+ umask 022
+ cd /home/david/rpmbuild/BUILD
+ cd /home/david/rpmbuild/BUILD
+ rm -rf iomemory-vsl-3.2.16.1732
+ /usr/bin/mkdir -p iomemory-vsl-3.2.16.1732
+ cd iomemory-vsl-3.2.16.1732
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ rsync -rv /home/david/iomemory/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/ driver_source/
sending incremental file list
created directory driver_source
./
Kbuild
Makefile
cdev.c
check_target_kernel.conf
check_target_kernel.sh
dkms.conf
driver_init.c
errno.c
kblock.c
kcache.c
kcondvar.c
kcpu.c
kcsr.c
kexports.c
kfile.c
kfio.c
kfio_common.c
kfio_config.sh
kfio_config_add.sh
khotplug.c
kinfo.c
kmem.c
kscatter.c
kscsi.c
kscsi_host.c
ktime.c
license.c
main.c
module_param.c
patch_module_version.sh
pci.c
[... 1160 lines cut ...]
RPM build errors:
    line 41: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-kmod, iodrive-driver, fio-driver
    Macro expanded in comment on line 45: %{name}-%{fio_tar_version}.tar.gz

    line 41: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-kmod, iodrive-driver, fio-driver
    Macro expanded in comment on line 45: %{name}-%{fio_tar_version}.tar.gz

    line 136: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver, fio-driver
    line 208: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-source, fio-driver-source
    line 210: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver, fio-driver
    Bad exit status from /var/tmp/rpm-tmp.RNr0DO (%build)
make: *** [Makefile:17: rpm] Error 1
```

### Possible solution
I know from looking at other projects with the same mm_struct issue the switch to `mm` over `mmap_sem` but no clue what that actually means..

### Environment information
kernel: `5.9.16-200.fc33.x86_64`
distribution: `Fedora 33`
gcc: `gcc (GCC) 10.2.1 20201125 (Red Hat 10.2.1-9)`
card: `01:00.0 Mass storage controller [0180]: SanDisk ioDrive2 [1aed:2001] (rev 04)`

Other notes:
I do have this machine setup for VFIO with IOMMU groups enabled, may have something to do with the pci errors but not sure if the rpmbuild scripts actually attempt to talk to the card during build stage.....

snuf, 2020-12-30:
Hey @dougalII thanks for the bug report. 

The original driver has too many issues and is something I don't look at all at anymore, the semaphore fix you mention and a lot of other fixes are already present in master and v5.6.0. The here maintained driver has issues with the 5.9 kernel as there are too many changes to the kernel sub system for handling IO, which means an overhaul of that part of the sub-system in the driver here (also for `iomemory-vsl4`). These changes are not very simple and straightforward. I have something that compiles but breaks miserably when attempting to do IO, so until that is fixed the `iomemory-vsl` driver will not work beyond 5.8.x.

Best

dougalII, 2020-12-30:
Hey @snuf 
Understood, knew it was a long shot. Let me know if you ever get something i can test with!

Thank you again for your awesome work!

snuf, 2020-12-31:
@dougalII it seems like I have something that compiles and doesn't BUG/RIP out in the kernel. I've done some small IO on it to check and it seems *OK*. I'm running more elaborate tests and need to make sure I can integrate it and remain backwards compatible with all of 5.x. `request_fn` is the branch I'm working on atm.

snuf, 2020-12-31:
@dougalII test came through clean on a normal ext4 fs, I'm going to check on LVM too and do some other tests I have to make sure that works, I'll report back here.

dougalII, 2020-12-31:
@snuf, 
Confirmed my end as well for base ext4! Migrated a few VMs onto it and so far no issues.
Thank you!

snuf, 2021-01-01:
@dougalII cool, tests with VMs and containers came back clean here too, so it seems we have a path forward. I'll make a nice patch and see if we can do a new release for the v5.9.x. I'll leave this open till I'm done with the patch for the new release. Thanks for testing!

snuf, 2021-01-07:
#78 contains fixes for 5.9.x, among other things, and has been merged to master.

closed 2021-01-07 by snuf