RemixVSL/iomemory-vsl #81: Ununtu 20.04 - driver not attaching -  "could not find canonical value across 24 pads"

ian-p-johnson, 2021-04-21:
### Bug description
Describe the issue, or paste the full error encountered here.
6.348332] <6>fioinf ioDrive 0000:28:00.0.0: Firmware version 7.1.17 116786 (0x700411 0x1c832)
[    6.348334] <6>fioinf ioDrive 0000:28:00.0.0: Platform version 16
[    6.348335] <6>fioinf ioDrive 0000:28:00.0.0: Firmware VCS version 116786 [0x1c832]
[    6.348347] <6>fioinf ioDrive 0000:28:00.0.0: Firmware VCS uid 0xaeb15671994a45642f91efbb214fa428e4245f8a
[    6.351546] <6>fioinf ioDrive 0000:28:00.0.0: Powercut flush: Enabled
[    6.621756] EDAC amd64: F17h_M70h detected (node 0).
[    6.621803] EDAC amd64: Node 0: DRAM ECC disabled.
[    6.648130] <6>fioinf ioDrive 0000:28:00.0.0: PCIe power monitor enabled (master). Limit set to 75.0 watts.
[    6.648132] <6>fioinf ioDrive 0000:28:00.0.0: Thermal monitoring: Enabled
[    6.648134] <6>fioinf ioDrive 0000:28:00.0.0: Hardware temperature alarm set for 100C.
[    6.705381] EDAC amd64: F17h_M70h detected (node 0).
[    6.705429] EDAC amd64: Node 0: DRAM ECC disabled.
[    6.732125] <6>fioinf ioDrive 0000:28:00.0: Found device fct0 (Dell ioDrive2 Duo 2410GB MLC 0000:28:00.0) on pipeline 0
[    6.732335] <3>fioerr Dell ioDrive2 Duo 2410GB MLC 0000:28:00.0: failed to map append request
[    6.732338] <3>fioerr Dell ioDrive2 Duo 2410GB MLC 0000:28:00.0: request page program 00000000248b8592 failed -22
[    7.061307] EDAC amd64: F17h_M70h detected (node 0).
[    7.061359] EDAC amd64: Node 0: DRAM ECC disabled.
[    7.116394] <6>fioinf ioDrive 0000:28:00.0.0: stuck flush request on startup detected, retry iteration 1 of 3...
[    7.116395] <6>fioinf ioDrive 0000:28:00.0.0: Starting master controller
[    7.117819] EDAC amd64: F17h_M70h detected (node 0).
[    7.117864] EDAC amd64: Node 0: DRAM ECC disabled.
[    7.168684] EDAC amd64: F17h_M70h detected (node 0).
[    7.168733] EDAC amd64: Node 0: DRAM ECC disabled.
[    7.172213] <6>fioinf ioDrive 0000:28:00.0.0: PMP Address: 1 1 2 1
[    7.256295] <6>fioinf ioDrive 0000:28:00.0.0: SMP Controller Firmware APP  version 1.0.29 0
[    7.256297] <6>fioinf ioDrive 0000:28:00.0.0: SMP Controller Firmware BOOT version 0.0.9 1
[    7.256298] <6>fioinf ioDrive 0000:28:00.0.0: SMP Adapter    Firmware APP  version 1.0.35 0
[    7.256298] <6>fioinf ioDrive 0000:28:00.0.0: SMP Adapter    Firmware BOOT version 0.0.8 1
[    7.704476] <6>fioinf ioDrive 0000:28:00.0.0: Required PCIE bandwidth 2.000 GBytes per sec
[    7.704479] <6>fioinf ioDrive 0000:28:00.0.0: Board serial number is 1231D1219-1121
[    7.704479] <6>fioinf ioDrive 0000:28:00.0.0: Adapter serial number is 1231D1219
[    7.704482] <6>fioinf ioDrive 0000:28:00.0.0: Default capacity        1205.000 GBytes
[    7.704483] <6>fioinf ioDrive 0000:28:00.0.0: Default sector size     512 bytes
[    7.704483] <6>fioinf ioDrive 0000:28:00.0.0: Rated endurance         17.00 PBytes
[    7.704484] <6>fioinf ioDrive 0000:28:00.0.0: 100C temp range hardware found
[    7.704485] <6>fioinf ioDrive 0000:28:00.0.0: Maximum capacity        1294.000 GBytes
[    8.712330] <6>fioinf ioDrive 0000:28:00.0.0: Firmware version 7.1.17 116786 (0x700411 0x1c832)
[    8.712332] <6>fioinf ioDrive 0000:28:00.0.0: Platform version 16
[    8.712333] <6>fioinf ioDrive 0000:28:00.0.0: Firmware VCS version 116786 [0x1c832]
[    8.712345] <6>fioinf ioDrive 0000:28:00.0.0: Firmware VCS uid 0xaeb15671994a45642f91efbb214fa428e4245f8a
[    8.715565] <6>fioinf ioDrive 0000:28:00.0.0: Powercut flush: Enabled
[    8.972559] <3>fioerr ioDrive 0000:28:00.0.0: could not find canonical value across 24 pads
[    9.270283] <3>fioerr ioDrive 0000:28:00.0.0: MINIMAL MODE DRIVER: hardware failure.
[    9.352255] <6>fioinf ioDrive 0000:28:00.0: Found device fct0 (Dell ioDrive2 Duo 2410GB MLC 0000:28:00.0) on pipeline 0
[    9.352542] <6>fioinf fct0: stuck flush request got better on retry.
[    9.352543] <6>fioinf Dell ioDrive2 Duo 2410GB MLC 0000:28:00.0: probed fct0
[    9.352546] <6>fioinf Dell ioDrive2 Duo 2410GB MLC 0000:28:00.0: Attaching explicitly disabled
[    9.352548] <3>fioerr Dell ioDrive2 Duo 2410GB MLC 0000:28:00.0: auto attach failed with error EINVAL: Invalid argument
[    9.352593] iodrive 0000:29:00.0: enabling device (0000 -> 0002)
[    9.352681] <6>fioinf ioDrive 0000:29:00.0: mapping controller on BAR 5
[    9.352779] <6>fioinf ioDrive 0000:29:00.0: MSI enabled
[    9.352781] <6>fioinf ioDrive 0000:29:00.0: using MSI interrupts
[    9.383015] <6>fioinf ioDrive 0000:29:00.0.0: Starting master controller
[    9.440040] <6>fioinf ioDrive 0000:29:00.0.0: PMP Address: 1 1 1 1
[    9.524294] <6>fioinf ioDrive 0000:29:00.0.0: SMP Controller Firmware APP  version 1.0.29 0
[    9.524296] <6>fioinf ioDrive 0000:29:00.0.0: SMP Controller Firmware BOOT version 0.0.9 1
[    9.524297] <6>fioinf ioDrive 0000:29:00.0.0: SMP Adapter    Firmware APP  version 1.0.35 0
[    9.524298] <6>fioinf ioDrive 0000:29:00.0.0: SMP Adapter    Firmware BOOT version 0.0.8 1
[    9.972456] <6>fioinf ioDrive 0000:29:00.0.0: Required PCIE bandwidth 2.000 GBytes per sec
[    9.972458] <6>fioinf ioDrive 0000:29:00.0.0: Board serial number is 1231D1219-1111
[    9.972459] <6>fioinf ioDrive 0000:29:00.0.0: Adapter serial number is 1231D1219
[    9.972461] <6>fioinf ioDrive 0000:29:00.0.0: Default capacity        1205.000 GBytes
[    9.972461] <6>fioinf ioDrive 0000:29:00.0.0: Default sector size     512 bytes
[    9.972462] <6>fioinf ioDrive 0000:29:00.0.0: Rated endurance         17.00 PBytes
[    9.972463] <6>fioinf ioDrive 0000:29:00.0.0: 100C temp range hardware found
[    9.972464] <6>fioinf ioDrive 0000:29:00.0.0: Maximum capacity        1294.000 GBytes
[   10.952329] <6>fioinf ioDrive 0000:29:00.0.0: Firmware version 7.1.17 116786 (0x700411 0x1c832)
[   10.952330] <6>fioinf ioDrive 0000:29:00.0.0: Platform version 16
[... 81 lines cut ...]
I managed to get the 25w limit lifted and the card working on Windows 10 fine (eventually) so a hardware problem is unlikely 

### Possible solution

### Environment information

gcc (Ubuntu 10.2.0-13ubuntu1) 10.2.0
Linux flasha 5.8.0-50-generic #56-Ubuntu SMP Mon Apr 12 17:18:36 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
Distributor ID: Ubuntu
Description:    Ubuntu 20.10
Release:        20.10
Codename:       groovy

Tag: v5.10.0

Ryzen 9 3900X - Asus B450 Tomahawk MAX
....
25:00.0 PCI bridge [0604]: SanDisk Device [1aed:2100] (rev bb)
26:04.0 PCI bridge [0604]: SanDisk Device [1aed:2100] (rev bb)
26:05.0 PCI bridge [0604]: SanDisk Device [1aed:2100] (rev bb)
26:06.0 PCI bridge [0604]: SanDisk Device [1aed:2100] (rev bb)
28:00.0 Mass storage controller [0180]: SanDisk ioDrive2 [1aed:2001] (rev 04)
29:00.0 Mass storage controller [0180]: SanDisk ioDrive2 [1aed:2001] (rev 04)
...
Dell iodrive ii Duo 2.4TB -

snuf, 2021-04-21:
Thanks for reporting @ian-p-johnson ! 

Recently someone reported this on a dual socket machine with Fedora 33, a 5.11 kernel and a 1.2TB IODrive2. They had no problem on kernel 5.10 (or any prior), however booting 5.11 caused the exact same problem. When the card was moved to a single socket machine the problem was gone and it attached fine on kernel 5.11. This was on the same firmware version as you are on 7.1.17, though I remember them downgrading to 7.1.15 on the dual socket machine with no effect. Looking at your setup the problem is probably not related to the fact that the other host they used had two sockets, or the firmware version.

The two cards I have I tested on Ubuntu 20.10 with 5.4, 5.8.0-50, and 5.11.0-16 on a 3900x (aorus pro wifi x570 1.0). One 1.2TB IODrive2 (fw 7.1.15) and a Duo 640GB (fw 7.1.17), unfortunately I can't reproduce the issue (I don't run on full PCIe power also). My cards are in the x8 and x4 slot, and the x16 slot is not used atm. Do you have above 4G decoding on in the bios btw?

So far troubleshooting seems to point at the scatter gather mapping going wrong, there is a 5.11-debug branch that shows the SGL output when it tries to attach showing the SGL number of entries not being what it should, resulting in the append failure and the -22 error, causing the drive not to be attached further as it just bails. (this could point to get_user_pages somewhere down the line or something that is harder to reach in the .so)

I revived a dual socket machine to do testing based on the findings earlier. However looking at your problem, which seems to be the same as the other person, but with a 5.8.0 kernel, and looking at my results so far I doubt it will give the expected result of triggering the failure, though I won't know till late tomorrow.

The fio tools you pulled are fine, we have the same pinned on the Discord channel.

Are these cards new to you, or have they worked on linux in your machine before?

Best

ian-p-johnson, 2021-04-21:
They are "new" (to me at least) just a day or two ago - 95% of TBW still remains and my planned usage is heavy - that is why they caught my interest 

Eventually i ended up with 2000+MB read/write on Windows 10  (CrystalDiskMark) on a single CPU LGA1150 which am happy with but the eventual destinations for the cards are the DIY built 3900X and a set of Dell Precision T5610  Dual E5-2650v2 LGA-2011 with 64GB ram running Ubuntu 20 (I have some leeway to change the distro version) - i have only tested on these 2 machines so far and as yet only have one card but was planning to order more once i get them working

I have not yet tested on the Dual Precision T5610 (Ubuntu 20) - I can't test on them until later today 

The B450 Tomahawk 3900X exhibiting the problem has only 2 16xPCIE slots and am pretty sure I tried both but I can  confirm that in a few hours - spec says Gen3(PCI_E1), Gen2(PCI_E4) and is running 32GB (4x8GB) 3600/16

"Do you have above 4G decoding on in the bios btw?" - will check in a couple of hours 

I also have a couple of HP DL380p G8 I can test on later (Ubuntu 20.10) - Dual E5-2640v2 LGA-2011-2)

So my todo list is:
1. Check 4G decoding
2. Check BOTH Tomahawk 16x slots
3. Check Dell Precision T5610/Dual 
4. Test on DL-380p Dual on Ubuntu 20.10

ian-p-johnson, 2021-04-21:
lshw shows: 

              *-pci:2
                   description: PCI bridge
                   product: 400 Series Chipset PCIe Port
                   vendor: Advanced Micro Devices, Inc. [AMD]
                   physical id: 4
                   bus info: pci@0000:20:04.0
                   version: 01
                   width: 32 bits
                   clock: 33MHz
                   capabilities: pci normal_decode bus_master cap_list
                   configuration: driver=pcieport
                   resources: irq:39 memory:f7200000-f76fffff ioport:e0000000(size=3145728)
                 *-pci
                      description: PCI bridge
                      product: SanDisk
                      vendor: SanDisk
                      physical id: 0
                      bus info: pci@0000:25:00.0
                      version: bb
                      width: 32 bits
                      clock: 33MHz
                      capabilities: pci normal_decode bus_master cap_list
                      configuration: driver=pcieport
                      resources: irq:40 memory:f7600000-f761ffff memory:f7200000-f75fffff ioport:e0000000(size=3145728)
                    *-pci:0
                         description: PCI bridge
                         product: SanDisk
                         vendor: SanDisk
                         physical id: 4
                         bus info: pci@0000:26:04.0
                         version: bb
                         width: 32 bits
                         clock: 33MHz
                         capabilities: pci normal_decode bus_master cap_list
                         configuration: driver=pcieport
                         resources: irq:41
                    *-pci:1
                         description: PCI bridge
                         product: SanDisk
                         vendor: SanDisk
                         physical id: 5
                         bus info: pci@0000:26:05.0
                         version: bb
                         width: 32 bits
                         clock: 33MHz
                         capabilities: pci normal_decode bus_master cap_list
                         configuration: driver=pcieport
                         resources: irq:42 memory:f7400000-f75fffff ioport:e0000000(size=2097152)
                      *-storage
                            description: Mass storage controller
                            product: ioDrive2
                            vendor: SanDisk
                            physical id: 0
                            bus info: pci@0000:28:00.0
                            version: 04
                            width: 32 bits
                            clock: 33MHz
                            capabilities: storage bus_master cap_list rom
                            configuration: driver=iodrive latency=0
                            resources: irq:82 memory:f7520000-f7521fff memory:f7500000-f751ffff memory:f7400000-f74fffff
                    *-pci:2
                         description: PCI bridge
                         product: SanDisk
                         vendor: SanDisk
                         physical id: 6
                         bus info: pci@0000:26:06.0
                         version: bb
                         width: 32 bits
                         clock: 33MHz
                         capabilities: pci normal_decode bus_master cap_list
                         configuration: driver=pcieport
                         resources: irq:43 memory:f7200000-f73fffff
                       *-storage
                            description: Mass storage controller
                            product: ioDrive2
                            vendor: SanDisk
                            physical id: 0
                            bus info: pci@0000:29:00.0
                            version: 04
                            width: 32 bits
                            clock: 33MHz
                            capabilities: storage bus_master cap_list rom
                            configuration: driver=iodrive latency=0
                            resources: irq:86 memory:f7320000-f7321fff memory:f7300000-f731ffff memory:f7200000-f72fffff

ian-p-johnson, 2021-04-21:
lspci -vv  shows:
25:00.0 PCI bridge: SanDisk Device 2100 (rev bb) (prog-if 00 [Normal decode])
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 40
        IOMMU group: 17
        Region 0: Memory at f7600000 (32-bit, non-prefetchable) [size=128K]
        Bus: primary=25, secondary=26, subordinate=29, sec-latency=0
        I/O behind bridge: [disabled]
        Memory behind bridge: f7200000-f75fffff [size=4M]
        Prefetchable memory behind bridge: 00000000e0000000-00000000e02fffff [size=3M]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16+ MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: <access denied>
        Kernel driver in use: pcieport

26:04.0 PCI bridge: SanDisk Device 2100 (rev bb) (prog-if 00 [Normal decode])
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 41
        IOMMU group: 17
        Bus: primary=26, secondary=27, subordinate=27, sec-latency=0
        I/O behind bridge: [disabled]
        Memory behind bridge: [disabled]
        Prefetchable memory behind bridge: [disabled]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16+ MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: <access denied>
        Kernel driver in use: pcieport

26:05.0 PCI bridge: SanDisk Device 2100 (rev bb) (prog-if 00 [Normal decode])
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 42
        IOMMU group: 17
        Bus: primary=26, secondary=28, subordinate=28, sec-latency=0
        I/O behind bridge: [disabled]
        Memory behind bridge: f7400000-f75fffff [size=2M]
        Prefetchable memory behind bridge: 00000000e0000000-00000000e01fffff [size=2M]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16+ MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: <access denied>
        Kernel driver in use: pcieport
26:06.0 PCI bridge: SanDisk Device 2100 (rev bb) (prog-if 00 [Normal decode])
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 43
        IOMMU group: 17
        Bus: primary=26, secondary=29, subordinate=29, sec-latency=0
        I/O behind bridge: [disabled]
        Memory behind bridge: f7200000-f73fffff [size=2M]
        Prefetchable memory behind bridge: [disabled]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16+ MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: <access denied>
        Kernel driver in use: pcieport

28:00.0 Mass storage controller: SanDisk ioDrive2 (rev 04)
        Subsystem: Dell ioDrive2
        Physical Slot: 117
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 82
        IOMMU group: 17
        Region 4: Memory at f7520000 (32-bit, non-prefetchable) [size=8K]
        Region 5: Memory at f7500000 (32-bit, non-prefetchable) [size=128K]
        Expansion ROM at f7400000 [disabled] [size=1M]
        Capabilities: <access denied>
        Kernel driver in use: iodrive
        Kernel modules: iomemory_vsl

29:00.0 Mass storage controller: SanDisk ioDrive2 (rev 04)
        Subsystem: Dell ioDrive2
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 86
        IOMMU group: 17
        Region 4: Memory at f7320000 (32-bit, non-prefetchable) [size=8K]
        Region 5: Memory at f7300000 (32-bit, non-prefetchable) [size=128K]
        Expansion ROM at f7200000 [disabled] [size=1M]
        Capabilities: <access denied>
        Kernel driver in use: iodrive
        Kernel modules: iomemory_vsl

ian-p-johnson, 2021-04-21:
The 3900X/B450 Tomahawk had above 4G disabled but it did not help the problem at all when i enabled it 

I will be installing Windows 10 on this machine as a test/interim solution pending some progress - at least it can be used .

bplein, 2021-04-21:
Unrelated to the bug, you will want to get your temperatures down in that system. Running 75-79C at idle will mean you will hit throttling as soon as you start writing to the cards with any intensity (possibly even with reads).  If possible, increase fan speeds, keep the card away from other cards in adjacent slots, and generally look at the airflow inside the system and make sure the card is getting plenty of airflow.

ian-p-johnson, 2021-04-21:
Yeah i plan on doing so - the case side is open - i am just reconfiguring the internals and the fans are not hooked up yet - the cooling will change radically

snuf, 2021-04-22:
Hmm I've had issues with my sx300 not being able to get attached due to cooling issues. The machine at that time did not seem too warm, but the card gets really hot. The sx300 uses the vsl4 driver, and did not show the exact same error though, failed to attach with pad issues. If you could double check with the fans attached that would be worth something perhaps.

The vsl cards have been benching all day with 7.1.17 on two machines, with temps of 74-79 on the fpga. I'll disconnect the fans in one machine tomorrow to see if I can get it to the point that it's unhappy. I've tested in a circular fashion, stomp over the drive with fio a bunch of times, unmount, rmmod, insmod, mount, fio stomp, rinse repeat. I'll turn the same thing on tomorrow morning sans fans.

ian-p-johnson, 2021-04-22:
I dont think its heat related - that problem was specifically after i had disconnected all case fans, had no air flow and backed up against a stack of hot spinning rust (which had just fitted) - temps are nearer 50c most of the time when not under a heavy load. It is running a continuous decent load NOW (high read & write) in windows and has been for the last 6-7 hours without problems around 62c, max 65c - and that is without a dedicated  fan (which i might add if it helps) 

I will try temps again  some time tomorrow - just to be sure - and check temps

When i tested yesterday earlier in Linux it was around 52c

I will also try it in the Dell Precision T5610 (2xE5-2650v2) to see if its a drive/Linux/MB combo problem

snuf, 2021-04-22:
@ian-p-johnson yeah that would be interesting to see. I'm running cards on the 3900x and a 2xE5-2650v2 on a X9DRi-LN4+/X9DR3-LN4+. Would it be possible to paste a full `fio-status -a` here?

With the 3900x and x570 I've had other interesting issues though with 5.8.0 and Ubuntu. Starting two VMs while having a ZFS filesystem somewhere in the box would make the kernel hit a BUG. this was prior to 5.8.0-50-generic though. It seemed to be aggravated by the "beta" BIOS and the kernel at that point, so I reverted the beta bios to a normal bios and stuck with 5.4.0 until I moved to 5.11.

ian-p-johnson, 2021-04-22:
I also have two more modules hopefully popping up tomorrow so i should be
able to try a few options later on Friday (UK hours)

ian-p-johnson, 2021-04-23:
Two FRESH Dell ioDrive II 2.4Tb modules thrown into the back of a 
  Dell Precision T5600 - Dual E5-2670 v1 64GB(8x8GB 12800)  DDR, running Ubuntu 20.10 

Setup in Raid 0 - working fine 
Odd timings but that is not a priority right now 
Slots are good for 75w so disabled checks 

sudo hdparm -t /dev/fio?

/dev/fioa: Timing buffered disk reads: 4478 MB in  3.00 seconds = 1492.50 MB/sec
/dev/fiob: Timing buffered disk reads: 4428 MB in  3.00 seconds = 1475.51 MB/sec
/dev/fioc: Timing buffered disk reads: 4508 MB in  3.00 seconds = 1502.22 MB/sec
/dev/fiod: Timing buffered disk reads: 4414 MB in  3.00 seconds = 1471.31 MB/sec

sudo mdadm --create --verbose /dev/md0 --level=0 --raid-devices=4 /dev/fio[abcd]
mdadm: chunk size defaults to 512K
mdadm: Defaulting to version 1.2 metadata
mdadm: array /dev/md0 started.
ianj@flashb:~$ sudo hdparm -t /dev/md0
/dev/md0: Timing buffered disk reads: 6482 MB in  3.00 seconds = 2159.83 MB/sec

I'll be testing the original module in the T5610 later to ascertain the likely-hood of it being MB or a module problem but it is promising so far - anything but windoze .. lol

Any clues on that 2159.83 MB/sec ? I just setup as raid 0 with no options - should i be expecting something more ?

snuf, 2021-04-23:
Happy to see something works :). Hdparm is not a greatest tool to test performance as it's not really a performance tool. Fio or IOZone are better for that. Also mind you that for simple access with one thread that writing/reading FIO devices are not rocket ships, they're fast but not great. For multi threads/process access they are great though.
```
snuf@scipio:~/Documents/Sources/Vagrant/ubuntu2004$ sudo hdparm -t /dev/fioa1 

/dev/fioa1:
 Timing buffered disk reads: 6546 MB in  3.00 seconds = 2181.93 MB/sec
snuf@scipio:~/Documents/Sources/Vagrant/ubuntu2004$
```
Though with FIO I get really different results, which are closer to what my workloads are like :)

ian-p-johnson, 2021-04-23:
That is why i am not too bothered - my usage will include memory mapped files accessed from Java and the other use values TBW and single larger FS. 

BTW is there any way to access remaining TBW from the basic Linux utilities - I just want to check they have decent life before it gets to late to return (or i get more) - I used the windows HP gui utility to check the first module

ian-p-johnson, 2021-04-23:
Does it require some specific fio-status version - this is all -a gives and that is everything ?
Does the "Driver not loaded" message indicate a version mismathc between utility and drivers - the modules are working fine 

Found 4 ioMemory devices in this system with 2 ioDrive Duos
Driver version: Driver not loaded

Adapter: Dual Controller Adapter
        Dell ioDrive2 Duo 2410GB MLC, Product Number:7F6JV, SN:US07F6JV760512890023
        ioDrive2 Adapter Controller, PN:F4K5G
        SMP(AVR) Versions: App Version: 1.0.33.0, Boot Version: 0.0.8.1
        External Power: NOT connected
        PCIe Power limit threshold: Disabled
        PCIe slot available power: 25.00W
        PCIe negotiated link: 4 lanes at 5.0 Gt/sec each, 2000.00 MBytes/sec total
        Connected ioMemory modules:
          0c:00.0:      Product Number:7F6JV, SN:1231D1314-1111
          0b:00.0:      Product Number:7F6JV, SN:1231D1314-1121

0c:00.0 SN:1231D1314-1111
        SMP(AVR) Versions: App Version: 1.0.29.0, Boot Version: 0.0.9.1
        Located in slot 0 Upper of ioDrive2 Adapter Controller SN:1231D1314
        PCI:0c:00.0
        Vendor:1aed, Device:2001, Sub vendor:1028, Sub device:1f71
        Firmware v7.1.17, rev 116786 Public
        PCIe slot available power: 25.00W
        PCIe negotiated link: 4 lanes at 5.0 Gt/sec each, 2000.00 MBytes/sec total
        Internal temperature: 57.09 degC, max 57.09 degC
        Internal voltage: avg 1.01V, max 1.02V
        Aux voltage: avg 2.48V, max 2.48V

0b:00.0 SN:1231D1314-1121
        SMP(AVR) Versions: App Version: 1.0.29.0, Boot Version: 0.0.9.1
        Located in slot 1 Lower of ioDrive2 Adapter Controller SN:1231D1314
        PCI:0b:00.0
        Vendor:1aed, Device:2001, Sub vendor:1028, Sub device:1f71
        Firmware v7.1.17, rev 116786 Public
        PCIe slot available power: 25.00W
        PCIe negotiated link: 4 lanes at 5.0 Gt/sec each, 2000.00 MBytes/sec total
        Internal temperature: 55.61 degC, max 57.58 degC
        Internal voltage: avg 1.02V, max 1.03V
        Aux voltage: avg 2.48V, max 2.49V

Adapter: Dual Controller Adapter
        Fusion-io ioDrive2 Duo 2410GB MLC, Product Number:7F6JV, SN:US07F6JV760513CR0022
        ioDrive2 Adapter Controller, PN:F4K5G
        SMP(AVR) Versions: App Version: 1.0.33.0, Boot Version: 0.0.8.1
        External Power: NOT connected
        PCIe Power limit threshold: Disabled
        PCIe slot available power: 0.24W
        PCIe negotiated link: 8 lanes at 5.0 Gt/sec each, 4000.00 MBytes/sec total
        Connected ioMemory modules:
          07:00.0:      Product Number:7F6JV, SN:1347D0B0F-1111
          06:00.0:      Product Number:7F6JV, SN:1347D0B0F-1121

07:00.0 SN:1347D0B0F-1111
        SMP(AVR) Versions: App Version: 1.0.29.0, Boot Version: 0.0.9.1
        Located in slot 0 Upper of ioDrive2 Adapter Controller SN:1347D0B0F
        PCI:07:00.0
        Vendor:1aed, Device:2001, Sub vendor:1028, Sub device:1f71
        Firmware v7.1.17, rev 116786 Public
        PCIe slot available power: 25.00W
        PCIe negotiated link: 4 lanes at 5.0 Gt/sec each, 2000.00 MBytes/sec total
        Internal temperature: 64.47 degC, max 64.97 degC
        Internal voltage: avg 1.01V, max 1.02V
        Aux voltage: avg 2.50V, max 2.50V

06:00.0 SN:1347D0B0F-1121
        SMP(AVR) Versions: App Version: 1.0.29.0, Boot Version: 0.0.9.1
        Located in slot 1 Lower of ioDrive2 Adapter Controller SN:1347D0B0F
        PCI:06:00.0
        Vendor:1aed, Device:2001, Sub vendor:1028, Sub device:1f71
        Firmware v7.1.17, rev 116786 Public
        PCIe slot available power: 25.00W
        PCIe negotiated link: 4 lanes at 5.0 Gt/sec each, 2000.00 MBytes/sec total
        Internal temperature: 60.04 degC, max 60.54 degC
        Internal voltage: avg 1.01V, max 1.01V
        Aux voltage: avg 2.50V, max 2.50V

snuf, 2021-04-23:
Yeah you probably want a newer version of the tools, which version do you have now `fio-status -v` ?
I run `3.2.16.1731 pinnacles@4870ad45b7ea` and get:
```
vagrant@fio:~/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16$ sudo fio-status -a

Found 1 ioMemory device in this system
Driver version: 3.2.16 build 1731

Adapter: Single Controller Adapter
	Fusion-io ioDrive2 1.205TB, Product Number:F00-001-1T20-CS-0001, SN:1241D0259, FIO SN:1241D0259
	ioDrive2 Adapter Controller, PN:PA004137009
	External Power: NOT connected
	PCIe Bus voltage: avg 12.02V
	PCIe Bus current: avg 0.76A
	PCIe Bus power: avg 9.10W
	PCIe Power limit threshold: 24.75W
	PCIe slot available power: unavailable
	Connected ioMemory modules:
	  fct0:	Product Number:F00-001-1T20-CS-0001, SN:1241D0259

fct0	Attached
	ioDrive2 Adapter Controller, Product Number:F00-001-1T20-CS-0001, SN:1241D0259
	ioDrive2 Adapter Controller, PN:PA004137009
	SMP(AVR) Versions: App Version: 1.0.35.0, Boot Version: 0.0.9.1
	Located in slot 0 Center of ioDrive2 Adapter Controller SN:1241D0259
	Powerloss protection: protected
	PCI:00:04.0, Slot Number:3
	Vendor:1aed, Device:2001, Sub vendor:1aed, Sub device:2001
	Firmware v7.1.17, rev 116786 Public
	UEFI Option ROM v3.2.16, rev 1731 Enabled
	1205.00 GBytes device size
	Format: v500, 2353515625 sectors of 512 bytes
	PCIe slot available power: unavailable
	PCIe negotiated link: 4 lanes at 5.0 Gt/sec each, 2000.00 MBytes/sec total
	Internal temperature: 64.47 degC, max 65.95 degC
	Internal voltage: avg 1.01V, max 1.03V
	Aux voltage: avg 2.48V, max 2.50V
	Reserve space status: Healthy; Reserves: 100.00%, warn at 10.00%
	Active media: 100.00%
	Rated PBW: 17.00 PB, 99.44% remaining
	Lifetime data volumes:
	   Physical bytes written: 94,629,387,524,152
	   Physical bytes read   : 84,089,924,419,024
	RAM usage:
	   Current: 417,562,880 bytes
	   Peak   : 417,562,880 bytes
	Contained VSUs:
	  fioa:	ID:0, UUID:0f402a7e-9948-7a4f-a2d0-7b0cb4d1ecda

fioa	State: Online, Type: block device
	ID:0, UUID:0f402a7e-9948-7a4f-a2d0-7b0cb4d1ecda
	1205.00 GBytes device size
	Format: 2353515625 sectors of 512 bytes
```

ian-p-johnson, 2021-04-23:
I was on 3.11 - just pulled down 3.16 and :

        Rated PBW: 17.00 PB, 96.29% remaining
        Rated PBW: 17.00 PB, 97.30% remaining
        Rated PBW: 17.00 PB, 96.48% remaining
        Rated PBW: 17.00 PB, 96.59% remaining

Am happy with that :)

snuf, 2021-04-23:
Some troubleshooting has been going on in the background on the Discord. Do you per chance have `amd_iommu=on` on the Ryzen box?

snuf, 2021-04-23:
@Aqua9 figured out that if you have `intel_iommu=on` set as a kernel boot cmdline option in 5.11 the driver fails to load with the exact same error. Adding `iommu=pt` causes the driver to load again. I've validated this behavior on a test machine and it worked exactly the same. `intel_iommu=on` alone causes the driver to bomb adding the `iommu=pt` causes it to load again. 

After looking at my Ryzen machine I have `amd_iommu=on iommu=pt` set for my boot cmdline options, as I've had some ACPI issues in the past, which is probably why I haven't encountered it on that machine.

Can you check what your cmdline settings are and if it's possible to try these?

snuf, 2021-04-26:
@ian-p-johnson any news on this or all good?

ian-p-johnson, 2021-04-27:
Sorry I don't have access to that machine at the moment - its working ok on a Dell Precision T5610 and a T5600 on Ubuntu 20.10 so far though - as soon as I get the Asus Tomahawk MAX & 3900X back I will be testing it

snuf, 2021-04-27:
np, just making sure we get to a resolving this report point :)

snuf, 2021-05-18:
@ian-p-johnson haven't heard back from you =), am closing this one, please reopen if required.

closed 2021-05-18 by snuf

waverly360, 2021-06-10:
I can confirm that by adding `amd_iommu=on iommu=pt` as options to my kernel, I was able to get my fusion card working.  I'm running Fedora 34 with a Ryzen processor.  (Thanks for the tip arrjaybee!!)

It was previously alerting that the card was running in minimal mode:
`Driver version: 3.2.16 build 1731

Adapter: Dual Controller Adapter
	Fusion-io ioDrive2 Duo 2410GB MLC, Product Number:7F6JV, SN:US07F6JV760512A40013
	ioDrive2 Adapter Controller, PN:F4K5G
	SMP(AVR) Versions: App Version: 1.0.33.0, Boot Version: 0.0.8.1
	External Power: NOT connected
	PCIe Power limit threshold: 24.75W
	PCIe slot available power: 75.00W
	PCIe negotiated link: 8 lanes at 5.0 Gt/sec each, 4000.00 MBytes/sec total
	Connected ioMemory modules:
	  fct0:	Product Number:7F6JV, SN:1239D2479-1121
	  fct1:	Product Number:7F6JV, SN:1239D2479-1111

fct0	Status unknown: Driver is in MINIMAL MODE:
		Device has a hardware failure
	SN:1239D2479-1121
!! --->	There are active errors or warnings on this device!  Read below for details.
	SMP(AVR) Versions: App Version: 1.0.29.0, Boot Version: 0.0.9.1
	Located in slot 1 Lower of ioDrive2 Adapter Controller SN:1239D2479
	Powerloss protection: not available
	PCI:0c:00.0, Slot Number:12
	Vendor:1aed, Device:2001, Sub vendor:1028, Sub device:1f71
	Firmware v7.1.17, rev 116786 Public
	Geometry and capacity information not available.
	Format: not low-level formatted
	PCIe slot available power: 25.00W
	PCIe negotiated link: 4 lanes at 5.0 Gt/sec each, 2000.00 MBytes/sec total
	Internal temperature: 71.86 degC, max 71.86 degC
	Internal voltage: avg 1.01V, max 1.02V
	Aux voltage: avg 2.48V, max 2.48V
	Rated PBW: 17.00 PB
	Lifetime data volumes:
	   Physical bytes written: 0
	   Physical bytes read   : 0
	RAM usage:
	   Current: 0 bytes
	   Peak   : 0 bytes

	ACTIVE WARNINGS:
	    The ioMemory is currently running in a minimal state. `