RemixVSL/iomemory-vsl #89: [BUG]

mtmongoose, 2021-05-07:
### Bug description
Describe the issue, or paste the full error encountered here.

When running "make dkms" on Proxmox host I get the following output:

Building module:
cleaning build area...
'make' DKMS_KERNEL_VERSION=5.11.7-1-pve....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.11.7-1-pve (x86_64)
Consult /var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/make.log for more information.
make[1]: *** [Makefile:119: dkms] Error 10
make[1]: Leaving directory '/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
make: *** [Makefile:8: dkms] Error 2

### How to reproduce
What are the steps to reproduce the reported issue.
```
git clone https://github.com/snuf/iomemory-vsl.git
cd iomemory-vsl
git checkout v5.12.1
make dkms 

** poof, broken token **
```

### Possible solution
Is a solution know, or type any plausible suggestions here, if none leave clear.

### Environment information
Information about the system the module is used on
1. Linux kernel compiled against (uname -a) -> 5.11.7-1-pve
2. The C compiler version used (gcc --version) -> gcc (Debian 8.3.0-6) 8.3.0
3. distribution, and version (cat /etc/os-release) -> Debian GNU/Linux 10 (buster)
4. Tag or Branch of iomemory-vsl that is being compiled -> v5.12.1
5. FIO device used, if applicable
   * fio-status -> ioDrive Duo

snuf, 2021-05-07:
@mtmongoose thanks for the report, can you add the output of `/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/make.log` ?

mtmongoose, 2021-05-07:
Well thank you. @snuf for this project. I just realized I just left the title of this issue "Bug", sorry about that. Below is the output of /var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/make.log.

DKMS make.log for iomemory-vsl-5.11.7-1-ecc66f1 for kernel 5.11.7-1-pve (x86_64)
Fri 07 May 2021 04:25:49 PM CDT
make[2]: Entering directory '/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build'
sed -i 's/Proprietary/GPL/g' Kbuild

Change found in target kernel: KERNELVER KERNEL_SRC
Running clean before building driver

make[3]: Entering directory '/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build'
make \
        -j16 \
        -C /lib/modules/5.11.7-1-pve/build \
        FIO_DRIVER_NAME=iomemory-vsl \
        FUSION_DRIVER_DIR=/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build \
        M=/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build \
        EXTRA_CFLAGS+="-I/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall -Werror" \
        KFIO_LIB=kfio/x86_64_cc83_libkfio.o_shipped \
        clean
make[4]: Entering directory '/usr/src/linux-headers-5.11.7-1-pve'
make[4]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
rm -rf include/fio/port/linux/kfio_config.h kfio_config license.c
make[3]: Leaving directory '/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build'
if [ "83" -gt "83" ];then \
        if [ ! -f "kfio/x86_64_cc83_libkfio.o_shipped" ];then \
                cp kfio/x86_64_cc83_libkfio.o_shipped kfio/x86_64_cc83_libkfio.o_shipped; \
        fi \
fi
./kfio_config.sh -a x86_64 -o include/fio/port/linux/kfio_config.h -k /lib/modules/5.11.7-1-pve/build -p -d /var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/kfio_config -l 0 -s /lib/modules/5.11.7-1-pve/source
Detecting Kernel Flags
Config dir         : /var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/kfio_config
Output file        : include/fio/port/linux/kfio_config.h
Kernel output dir  : /lib/modules/5.11.7-1-pve/build
Kernel source dir  : /lib/modules/5.11.7-1-pve/build
Starting tests:
  1620422750.037  KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS...
  1620422750.052  KFIOC_X_TASK_HAS_CPUS_MASK...
  1620422750.066  KFIOC_X_LINUX_HAS_PART_STAT_H...
  1620422750.079  KFIOC_X_BLK_ALLOC_QUEUE_NODE_EXISTS...
  1620422750.091  KFIOC_X_HAS_MAKE_REQUEST_FN...
  1620422750.105  KFIOC_X_GENHD_PART0_IS_A_POINTER...
  1620422750.121  KFIOC_X_BIO_HAS_BI_BDEV...
Started tests, waiting for completions...
  1620422756.164  KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS=0
  1620422756.176  KFIOC_X_TASK_HAS_CPUS_MASK=0
  1620422756.188  KFIOC_X_LINUX_HAS_PART_STAT_H=0
  1620422756.200  KFIOC_X_BLK_ALLOC_QUEUE_NODE_EXISTS=0
  1620422756.212  KFIOC_X_HAS_MAKE_REQUEST_FN=0
  1620422756.224  KFIOC_X_GENHD_PART0_IS_A_POINTER=0
  1620422756.236  KFIOC_X_BIO_HAS_BI_BDEV=0
Finished
1620422756.242  Exiting
Preserving configdir due to '-p' option: /var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/kfio_config
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
make \
    -j16 \
-C /lib/modules/5.11.7-1-pve/build \
FIO_DRIVER_NAME=iomemory-vsl \
FUSION_DRIVER_DIR=/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build \
M=/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build \
EXTRA_CFLAGS+="-I/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall -Werror" \
INSTALL_MOD_DIR=extra/fio \
INSTALL_MOD_PATH= \
KFIO_LIB=kfio/x86_64_cc83_libkfio.o_shipped \
modules
make[3]: Entering directory '/usr/src/linux-headers-5.11.7-1-pve'
printf '#include "linux/module.h"\nMODULE_LICENSE("GPL");\n' >/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/license.c
[... 59 lines cut ...]
                                 ^~~
                                 insl
/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/kblock.c:526:33: error: ‘sectors’ undeclared (first use in this function); did you mean ‘sector_t’?
         part_stat_add(GD_PART0, sectors[0], totalsize >> 9);
                                 ^~~~~~~
                                 sector_t
/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/kblock.c:527:33: error: ‘nsecs’ undeclared (first use in this function)
         part_stat_add(GD_PART0, nsecs[0],   duration * 1000);
                                 ^~~~~
/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/kfile.c: In function ‘kfio_create_proc_fops_entry’:
/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/kfile.c:225:47: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Werror=incompatible-pointer-types]
     return proc_create_data(name, mode, base, (struct file_operations *)fops, data);
                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/kfile.c:37:
./include/linux/proc_fs.h:104:31: note: expected ‘const struct proc_ops *’ but argument is of type ‘struct file_operations *’
 extern struct proc_dir_entry *proc_create_data(const char *, umode_t,
                               ^~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[4]: *** [scripts/Makefile.build:287: /var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/kblock.o] Error 1
cc1: all warnings being treated as errors
make[4]: *** [scripts/Makefile.build:287: /var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build/kfile.o] Error 1
make[3]: *** [Makefile:1834: /var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build] Error 2
make[3]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
make[2]: *** [Makefile:134: modules] Error 2
make[2]: Leaving directory '/var/lib/dkms/iomemory-vsl/5.11.7-1-ecc66f1/build'

snuf, 2021-05-07:
@mtmongoose welcome and no problem! 

This is the same issue as https://github.com/snuf/iomemory-vsl4/issues/29, it seems like proxmox is not nice with its kernel headers/tools and updates. Please have a look at that report and let me know if it helps you?

mtmongoose, 2021-05-07:
I tried that. I'll try again though, should have put that in my previous comment.

snuf, 2021-05-07:
@mtmongoose the more you tell me the more I know and the easier it is to help :)

mtmongoose, 2021-05-07:
There's no doubt about that and I do appreciate your help. I'll be more clear going forward :-)

I removed the cloned project:
`rm -r iomemory-vsl/`

I cp the file as identified:
`cp /usr/src/linux-headers-5.11.7-1-pve/scripts/module.lds.S /usr/src/linux-headers-5.11.7-1-pve/scripts/module.lds`

Cloned the project again and checked out release v5.12.1:
`git clone https://github.com/snuf/iomemory-vsl`
'cd iomemory-vsl/`
`git checkout v5.12.1`

Then tried to build via:
`make module`

And it looks to me like I get the same output as last time.

cd root/usr/src/iomemory-vsl-3.2.16 && \
        make clean
make[1]: Entering directory '/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
make \
        -j16 \
        -C /lib/modules/5.11.7-1-pve/build \
        FIO_DRIVER_NAME=iomemory-vsl \
        FUSION_DRIVER_DIR=/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
        M=/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
        EXTRA_CFLAGS+="-I/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall -Werror" \
        KFIO_LIB=kfio/x86_64_cc83_libkfio.o_shipped \
        clean
make[2]: Entering directory '/usr/src/linux-headers-5.11.7-1-pve'
make[2]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
rm -rf include/fio/port/linux/kfio_config.h kfio_config license.c
make[1]: Leaving directory '/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
cd root/usr/src/iomemory-vsl-3.2.16 && \
        make gpl
make[1]: Entering directory '/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
sed -i 's/Proprietary/GPL/g' Kbuild

Change found in target kernel: KERNELVER KERNEL_SRC
Running clean before building driver

make[2]: Entering directory '/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
make \
        -j16 \
        -C /lib/modules/5.11.7-1-pve/build \
        FIO_DRIVER_NAME=iomemory-vsl \
        FUSION_DRIVER_DIR=/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
        M=/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
        EXTRA_CFLAGS+="-I/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall -Werror" \
        KFIO_LIB=kfio/x86_64_cc83_libkfio.o_shipped \
        clean
make[3]: Entering directory '/usr/src/linux-headers-5.11.7-1-pve'
make[3]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
rm -rf include/fio/port/linux/kfio_config.h kfio_config license.c
make[2]: Leaving directory '/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
if [ "83" -gt "74" ];then \
        if [ ! -f "kfio/x86_64_cc83_libkfio.o_shipped" ];then \
                cp kfio/x86_64_cc74_libkfio.o_shipped kfio/x86_64_cc83_libkfio.o_shipped; \
        fi \
fi
./kfio_config.sh -a x86_64 -o include/fio/port/linux/kfio_config.h -k /lib/modules/5.11.7-1-pve/build -p -d /home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config -l 0 -s /lib/modules/5.11.7-1-pve/source
Detecting Kernel Flags
Config dir         : /home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config
Output file        : include/fio/port/linux/kfio_config.h
Kernel output dir  : /lib/modules/5.11.7-1-pve/build
Kernel source dir  : /lib/modules/5.11.7-1-pve/build
Starting tests:
  1620425058.956  KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS...
  1620425058.970  KFIOC_X_TASK_HAS_CPUS_MASK...
[... 93 lines cut ...]
         part_stat_add(GD_PART0, nsecs[1],   duration * 1000);
                                 ^~~~~
/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:515:9: error: implicit declaration of function ‘part_stat_unlock’; did you mean ‘bit_spin_unlock’? [-Werror=implicit-function-declaration]
         part_stat_unlock();
         ^~~~~~~~~~~~~~~~
         bit_spin_unlock
/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_disk_stat_read_update’:
/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:525:33: error: ‘ios’ undeclared (first use in this function); did you mean ‘insl’?
         part_stat_inc(GD_PART0, ios[0]);
                                 ^~~
                                 insl
/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:526:33: error: ‘sectors’ undeclared (first use in this function); did you mean ‘sector_t’?
         part_stat_add(GD_PART0, sectors[0], totalsize >> 9);
                                 ^~~~~~~
                                 sector_t
/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:527:33: error: ‘nsecs’ undeclared (first use in this function)
         part_stat_add(GD_PART0, nsecs[0],   duration * 1000);
                                 ^~~~~
cc1: all warnings being treated as errors
make[3]: *** [scripts/Makefile.build:287: /home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.o] Error 1
make[2]: *** [Makefile:1834: /home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
make[1]: *** [Makefile:134: modules] Error 2
make[1]: Leaving directory '/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
make: *** [Makefile:29: module] Error 2

snuf, 2021-05-07:
yeah the output tells me that the header files are not found/used correctly. Can you go to the `/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS` dir and type `make` and get the output from that?

mtmongoose, 2021-05-07:
Below is the output from `make`. I'm trying to wrap my head around it. Looks like this step in the build is creating an error in kfioc_test.ko on line 29. I'm going to have a look at that file and see if I can figure out what's causing the error:

make -C /lib/modules/5.11.7-1-pve/build M=/home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS EXTRA_CFLAGS='-Wall -Werror-implicit-function-declaration' modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.7-1-pve'
  LD [M]  /home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.ko
ld:scripts/module.lds:29: ignoring invalid character `#' in expression
ld:scripts/module.lds:29: syntax error
make[3]: *** [scripts/Makefile.modfinal:58: /home/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.ko] Error 1
make[2]: *** [scripts/Makefile.modpost:117: __modpost] Error 2
make[1]: *** [Makefile:1738: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
make: *** [Makefile:12: modules] Error 2

snuf, 2021-05-07:
@mtmongoose there is something wrong with the `scripts/module.lds` file in your headers, that is what it's complaining about:
```
ld:scripts/module.lds:29: ignoring invalid character `#' in expression
ld:scripts/module.lds:29: syntax error
```
is the giveaway for this. Seems like the file that did work in the other case is not working for you and there is something wrong with it. If I look at my files I can see that the files are different:
```
snuf@scipio:~/Documents/Sources$ diff -y  /usr/src/linux-headers-5.11.0-16-generic/scripts/module.lds.S /usr/src/linux-headers-5.11.0-16-generic/scripts/module.lds
/*							      <
 * Common module linker script, always used when linking a mo <
 * Archs are free to supply their own linker scripts.  ld wil <
 * combine them automatically.				      <
 */							      <
SECTIONS {							SECTIONS {
	/DISCARD/ : {					      |	 /DISCARD/ : {
		*(.discard)				      |	  *(.discard)
		*(.discard.*)				      |	  *(.discard.*)
	}						      |	 }
							      |	 __ksymtab 0 : { *(SORT(___ksymtab+*)) }
	__ksymtab		0 : { *(SORT(___ksymtab+*)) } |	 __ksymtab_gpl 0 : { *(SORT(___ksymtab_gpl+*)) }
	__ksymtab_gpl		0 : { *(SORT(___ksymtab_gpl+* |	 __ksymtab_unused 0 : { *(SORT(___ksymtab_unused+*)) }
	__ksymtab_unused	0 : { *(SORT(___ksymtab_unuse |	 __ksymtab_unused_gpl 0 : { *(SORT(___ksymtab_unused_gpl+*)) 
	__ksymtab_unused_gpl	0 : { *(SORT(___ksymtab_unuse |	 __ksymtab_gpl_future 0 : { *(SORT(___ksymtab_gpl_future+*)) 
	__ksymtab_gpl_future	0 : { *(SORT(___ksymtab_gpl_f |	 __kcrctab 0 : { *(SORT(___kcrctab+*)) }
	__kcrctab		0 : { *(SORT(___kcrctab+*)) } |	 __kcrctab_gpl 0 : { *(SORT(___kcrctab_gpl+*)) }
	__kcrctab_gpl		0 : { *(SORT(___kcrctab_gpl+* |	 __kcrctab_unused 0 : { *(SORT(___kcrctab_unused+*)) }
	__kcrctab_unused	0 : { *(SORT(___kcrctab_unuse |	 __kcrctab_unused_gpl 0 : { *(SORT(___kcrctab_unused_gpl+*)) 
	__kcrctab_unused_gpl	0 : { *(SORT(___kcrctab_unuse |	 __kcrctab_gpl_future 0 : { *(SORT(___kcrctab_gpl_future+*)) 
	__kcrctab_gpl_future	0 : { *(SORT(___kcrctab_gpl_f |	 .init_array 0 : ALIGN(8) { *(SORT(.init_array.*)) *(.init_ar
							      |	 __jump_table 0 : ALIGN(8) { KEEP(*(__jump_table)) }
	.init_array		0 : ALIGN(8) { *(SORT(.init_a <
							      <
	__jump_table		0 : ALIGN(8) { KEEP(*(__jump_ <
}								}
							      <
/* bring in arch-specific sections */			      <
#include <asm/module.lds.h>				      <
```
So you might want to try and remove line 29 from that file.

nhidog, 2021-05-07:
Remove last line the include then try to compile

mtmongoose, 2021-05-07:
I owe you. Running `diff -y /usr/src/linux-headers-5.11.7-1-pve/scripts/module.lds.S /usr/src/linux-headers-5.11.7-1-pve/scripts/module.lds` showed that my files were identical. I removed the last line "#include <asm/module.lds.h>" and cloned the project again the `make dkms` ran successfully.

snuf, 2021-05-07:
@nhidog @mtmongoose you might want to file a bug with proxmox for this :)

mtmongoose, 2021-05-07:
Good idea. I'll work on getting one submitted. Thank you again for your help! :-)

snuf, 2021-05-07:
np anytime! Feel free to join the discord channel too btw, also don't forget to star the project :P

I'll close this one for now!

closed 2021-05-07 by snuf