RemixVSL/iomemory-vsl #90: compile error on rocky linux

PPisz, 2021-07-05:
hi i try to compile libvsl on rocky linux and i end up with the following error:

In file included from /root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/kblock.c:50:
/root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_alloc_queue’:
/root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/include/kblock_meta.h:27:45: error: passing argument 1 of ‘blk_alloc_queue’ makes integer from pointer without a cast [-Werror=int-conversion]
     #define BLK_ALLOC_QUEUE blk_alloc_queue(kfio_make_request, node);
                                             ^~~~~~~~~~~~~~~~~
/root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/kblock.c:960:10: note: in expansion of macro ‘BLK_ALLOC_QUEUE’
     rq = BLK_ALLOC_QUEUE;
          ^~~~~~~~~~~~~~~
In file included from /root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/port-internal.h:63,
                 from /root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/kblock.c:32:
./include/linux/blkdev.h:1208:23: note: expected ‘gfp_t’ {aka ‘unsigned int’} but argument is of type ‘unsigned int (*)(struct request_queue *, struct bio *)’
 struct request_queue *blk_alloc_queue(gfp_t);
                       ^~~~~~~~~~~~~~~
In file included from /root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/kblock.c:50:
/root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/include/kblock_meta.h:27:29: error: too many arguments to function ‘blk_alloc_queue’
     #define BLK_ALLOC_QUEUE blk_alloc_queue(kfio_make_request, node);
                             ^~~~~~~~~~~~~~~
/root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/kblock.c:960:10: note: in expansion of macro ‘BLK_ALLOC_QUEUE’
     rq = BLK_ALLOC_QUEUE;
          ^~~~~~~~~~~~~~~
In file included from /root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/port-internal.h:63,
                 from /root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/kblock.c:32:
./include/linux/blkdev.h:1208:23: note: declared here
 struct request_queue *blk_alloc_queue(gfp_t);
                       ^~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[3]: *** [scripts/Makefile.build:315: /root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16/kblock.o] Błąd 1
make[3]: *** Oczekiwanie na niezakończone zadania....
make[2]: *** [Makefile:1563: _module_/root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16] Błąd 2
make[2]: Opuszczenie katalogu '/usr/src/kernels/4.18.0-305.7.1.el8_4.x86_64'
make[1]: *** [Makefile:134: modules] Błąd 2
make[1]: Opuszczenie katalogu '/root/iomemory-vsl-5.12.1/root/usr/src/iomemory-vsl-3.2.16'
make: *** [Makefile:29: module] Błąd 2

I would be grateful for help!
Regards!

snuf, 2021-07-05:
@PPisz thank you for your bug submission. Can you please provide:
1) what kernel your Rocky Linux install uses
2) Do you have the kernel headers installed

Cheers

PPisz, 2021-07-06:
Hi snuf, fresh Rocky Linux 8.4:

[root@psc1 ~]# uname -a
Linux psc1.piszki.lab 4.18.0-305.7.1.el8_4.x86_64 #1 SMP Tue Jun 29 19:22:12 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
kernel-headers-4.18.0-305.7.1.el8_4.x86_64 
kernel-devel-4.18.0-305.7.1.el8_4.x86_64 and full "Development Tools"

snuf, 2021-07-06:
@PPisz ```Releases are tagged, and should be checked out by their tag. The release tags follow Linux Kernel versions. E.g. v5.12.1 (A Fine Boi) will work on all 5.x kernels that are 5.12 and lower, but is not guaranteed to work on 5.13. v4.20.2 - Grey Dawn supports most kernels that pre-date 5.0.```. You are running a 4.18 kernel but using the driver that only works with 5 series kernels. Please use the 4.20.2 tag/branch.

PPisz, 2021-07-06:
Ok now I understand, the error is still there:

[root@psc1 iomemory-vsl]# git checkout v4.20.2
HEAD is now at df066e4 take the odd bits out
[root@psc1 iomemory-vsl]# make module
...
In file included from ././include/linux/compiler_types.h:58,
                 from <command-line>:
/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_disk_stat_write_update’:
./include/linux/compiler-gcc.h:16:19: error: expected expression before ‘__asm__’
 #define barrier() __asm__ __volatile__("": : :"memory")
                   ^~~~~~~
./include/linux/preempt.h:242:29: note: in expansion of macro ‘barrier’
 #define preempt_disable()   barrier()
                             ^~~~~~~
./include/linux/genhd.h:321:26: note: in expansion of macro ‘preempt_disable’
 #define part_stat_lock() preempt_disable()
                          ^~~~~~~~~~~~~~~
/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:930:15: note: in expansion of macro ‘part_stat_lock’
         cpu = part_stat_lock();
               ^~~~~~~~~~~~~~
/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_disk_stat_read_update’:
./include/linux/compiler-gcc.h:16:19: error: expected expression before ‘__asm__’
 #define barrier() __asm__ __volatile__("": : :"memory")
                   ^~~~~~~
./include/linux/preempt.h:242:29: note: in expansion of macro ‘barrier’
 #define preempt_disable()   barrier()
                             ^~~~~~~
./include/linux/genhd.h:321:26: note: in expansion of macro ‘preempt_disable’
 #define part_stat_lock() preempt_disable()
                          ^~~~~~~~~~~~~~~
/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:989:15: note: in expansion of macro ‘part_stat_lock’
         cpu = part_stat_lock();
               ^~~~~~~~~~~~~~
/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c: At top level:
/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:178:13: error: ‘kfio_blk_complete_request’ declared ‘static’ but nev                                                           er defined [-Werror=unused-function]
 static void kfio_blk_complete_request(struct request *req, int error);
             ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:179:20: error: ‘kfio_request_to_bio’ declared ‘static’ but never def                                                           ined [-Werror=unused-function]
 static kfio_bio_t *kfio_request_to_bio(kfio_disk_t *disk, struct request *req,
                    ^~~~~~~~~~~~~~~~~~~
/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:429:13: error: ‘kfio_restart_queue’ defined but not used [-Werror=un                                                           used-function]
 static void kfio_restart_queue(struct request_queue *q)
             ^~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[3]: *** [scripts/Makefile.build:315: /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.o] Błąd 1
make[3]: *** Oczekiwanie na niezakończone zadania....
make[2]: *** [Makefile:1563: _module_/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16] Błąd 2
make[2]: Opuszczenie katalogu '/usr/src/kernels/4.18.0-305.7.1.el8_4.x86_64'
make[1]: *** [Makefile:134: modules] Błąd 2
make[1]: Opuszczenie katalogu '/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
make: *** [Makefile:29: module] Błąd 2

I'm going to check with kernel 5

PPisz, 2021-07-06:
OK successfully compiled v5.12.1 branch witch kernel 5.13.0-1.el8.elrepo.x86_64

Could you check why v4.20.2 branch doesn't compile with the 4.18.0-305.7.1.el8_4.x86_64 kernel ?

Tourman36, 2021-07-06:
Piotr, please use the newer kernels and releases going forward. 4.x is from
2018 and not actively maintained.

closed 2021-07-06 by Tourman36