RemixVSL/iomemory-vsl #94: [BUG] More an I need help building for 5.14.0 kernel

4nanook, 2021-08-31:
### Bug description
Describe the issue, or paste the full error encountered here.

I am trying to build for kernel 5.14.0 but it fails.  I suspect the issue is not knowing what to put in to the release tag in order
to get the proper version for this kernel.

### How to reproduce
What are the steps to reproduce the reported issue.
```
git clone https://github.com/snuf/iomemory-vsl.git
cd iomemory-vsl
git checkout <tag or some-branch>
make module
** poof, broken token **
```

### Possible solution
Is a solution know, or type any plausible suggestions here, if none leave clear.

### Environment information
Information about the system the module is used on
1. Linux kernel compiled against (uname -a)
2. The C compiler version used (gcc --version)
3. distribution, and version (cat /etc/os-release)
4. Tag or Branch of iomemory-vsl that is being compiled
I do not know the correct tag to put in for 5.14.0 kernel.  I can see there was a commit for this four days ago but I am not very
familiar with git and do not know how to pull the code with this applied.

5. FIO device used, if applicable
   * fio-status
   * lspci -b -nn

snuf, 2021-08-31:
5.14 is not "released" yet. The code is in `main` (what you checkout by default), does compile/work/run and was tested on ubuntu 21.04 with 5.14-rc6.

4nanook, 2021-09-01:
I do not know what to specify for <release-tag> as given in these instructions:
## Releases
Releases are tagged, and should be checked out by their tag. The release tags fo
llow Linux Kernel versions. E.g. **v5.12.1 (A Fine Boi)** will work on all 5.x k
ernels that are 5.12 and lower, but is not guaranteed to work on 5.13. **v4.20.2
 - Grey Dawn** supports most kernels that pre-date 5.0.
If I specify v5.12.1 as the <release-tag> the compile fails under 5.14.0 as follows:

In file included from /usr/src/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:50:
/usr/src/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c: In function ‘kfio_alloc_queue’:
/usr/src/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/include/kblock_meta.h:33:27: error: implicit declaration of function ‘blk_alloc_queue’; did you mean ‘kfio_alloc_queue’? [-Werror=implicit-function-declaration]
   33 |   #define BLK_ALLOC_QUEUE blk_alloc_queue(node);
      |                           ^~~~~~~~~~~~~~~
/usr/src/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:960:10: note: in expansion of macro ‘BLK_ALLOC_QUEUE’
  960 |     rq = BLK_ALLOC_QUEUE;
      |          ^~~~~~~~~~~~~~~
/usr/src/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.c:960:8: error: assignment to ‘struct request_queue *’ from ‘int’ makes pointer from integer without a cast [-Werror=int-conversion]
  960 |     rq = BLK_ALLOC_QUEUE;
      |        ^
cc1: all warnings being treated as errors
make[3]: *** [scripts/Makefile.build:271: /usr/src/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1851: /usr/src/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.14.0'
make[1]: *** [Makefile:134: modules] Error 2
make[1]: Leaving directory '/usr/src/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
make: *** [Makefile:29: module] Error 2

snuf, 2021-09-01:
When you clone the repository and do a `git branch` you're on the `main` branch. As 5.14 is not released yet, there is not branch for it. So you have to use `main`. The branch you are on when you freshly clone the repo. If you type `git branch` it shows you which repo you are on.

snuf, 2021-09-02:
@4nanook closing this one as I assume it worked out?

closed 2021-09-02 by snuf