RemixVSL/iomemory-vsl #96: Build problem on CentOS 8.5 kernel 4.18.0-348.2.1.el8_5.x86_64

vladsol, 2021-11-18:
Tried different branches, but no luck :)

```
make \
    -j4 \
-C /lib/modules/4.18.0-348.2.1.el8_5.x86_64/build \
FIO_DRIVER_NAME=iomemory-vsl \
FUSION_DRIVER_DIR=/var/lib/dkms/iomemory-vsl/v5.10.0/build \
M=/var/lib/dkms/iomemory-vsl/v5.10.0/build \
EXTRA_CFLAGS+="-I/var/lib/dkms/iomemory-vsl/v5.10.0/build/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall -Werror" \
INSTALL_MOD_DIR=extra/fio \
INSTALL_MOD_PATH= \
KFIO_LIB=kfio/x86_64_cc85_libkfio.o_shipped \
modules
make[3]: Entering directory '/usr/src/kernels/4.18.0-348.2.1.el8_5.x86_64'
printf '#include "linux/module.h"\nMODULE_LICENSE("GPL");\n' >/var/lib/dkms/iomemory-vsl/v5.10.0/build/license.c
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/main.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/license.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/pci.o

  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/sysrq.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/driver_init.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/kfio.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/errno.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/state.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/kcache.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/kfile.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/kmem.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/kfio_common.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/kcpu.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/kscatter.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/ktime.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/sched.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/cdev.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/kblock.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/kcondvar.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/kinfo.o
  CC [M]  /var/lib/dkms/iomemory-vsl/v5.10.0/build/kexports.o
In file included from /var/lib/dkms/iomemory-vsl/v5.10.0/build/kblock.c:50:
/var/lib/dkms/iomemory-vsl/v5.10.0/build/kblock.c: In function ‘kfio_alloc_queue’:
/var/lib/dkms/iomemory-vsl/v5.10.0/build/include/kblock_meta.h:21:45: error: passing argument 1 of ‘blk_alloc_queue’ makes integer from pointer without a cast [-Werror=int-conversion]
     #define BLK_ALLOC_QUEUE blk_alloc_queue(kfio_make_request, node);
                                             ^~~~~~~~~~~~~~~~~
/var/lib/dkms/iomemory-vsl/v5.10.0/build/kblock.c:961:10: note: in expansion of macro ‘BLK_ALLOC_QUEUE’
     rq = BLK_ALLOC_QUEUE;
          ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/iomemory-vsl/v5.10.0/build/port-internal.h:63,
                 from /var/lib/dkms/iomemory-vsl/v5.10.0/build/kblock.c:32:
./include/linux/blkdev.h:1219:23: note: expected ‘gfp_t’ {aka ‘unsigned int’} but argument is of type ‘unsigned int (*)(struct request_queue *, struct bio *)’
 struct request_queue *blk_alloc_queue(gfp_t);
                       ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/iomemory-vsl/v5.10.0/build/kblock.c:50:
/var/lib/dkms/iomemory-vsl/v5.10.0/build/include/kblock_meta.h:21:29: error: too many arguments to function ‘blk_alloc_queue’
     #define BLK_ALLOC_QUEUE blk_alloc_queue(kfio_make_request, node);
                             ^~~~~~~~~~~~~~~
/var/lib/dkms/iomemory-vsl/v5.10.0/build/kblock.c:961:10: note: in expansion of macro ‘BLK_ALLOC_QUEUE’
     rq = BLK_ALLOC_QUEUE;
          ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/iomemory-vsl/v5.10.0/build/port-internal.h:63,
                 from /var/lib/dkms/iomemory-vsl/v5.10.0/build/kblock.c:32:
./include/linux/blkdev.h:1219:23: note: declared here
 struct request_queue *blk_alloc_queue(gfp_t);
                       ^~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[4]: *** [scripts/Makefile.build:315: /var/lib/dkms/iomemory-vsl/v5.10.0/build/kblock.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [Makefile:1571: _module_/var/lib/dkms/iomemory-vsl/v5.10.0/build] Error 2
make[3]: Leaving directory '/usr/src/kernels/4.18.0-348.2.1.el8_5.x86_64'
make[2]: *** [Makefile:134: modules] Error 2
make[2]: Leaving directory '/var/lib/dkms/iomemory-vsl/v5.10.0/build'
```

snuf, 2021-11-18:
Have you tried main?

vladsol, 2021-11-19:
> Have you tried main?

Yes. It works :-) Thanks!

closed 2022-01-09 by snuf