RemixVSL/iomemory-vsl #99: vsl drivers for Linux 5.16?

4nanook, 2022-03-12:
I am trying to compile drivers for 5.16 not having luck with the existing package, says can't find stddef.h even though it is in the headers tree.  I just grabbed the source using git clone and did not commit any particular release as this was the instructions I was given for 5.13, because all version of Linux prior to 5.16 have a bug that can result in file corruption and/or privilege escalation, it is desirable to be on the latest kernel, 5.16.14, which has addressed these issues.  Previously I was on 5.13.19 and it compiled okay under that kernel. Kernels are self-compiled, header files get stuck in /usr/src/linux-headers-5.13.19 or /usr/src/linux-headers-5.16.14.  Thanks in advance for any help you can offer.

bplein, 2022-03-12:
> I am trying to compile drivers for 5.16 not having luck with the existing package, says can't find stddef.h even though it is in the headers tree.  I just grabbed the source using git clone and did [...]

Significant Linux kernel changes in 5.15 have broken this driver for now.  The primary developer will eventually work on it but it's broken for now.  Suggest going back to 5.14 or newer if you can.

4nanook, 2022-03-12:
Thanks, but because I need my machines to be secure and 5.14 does not
provide this and the exploit is now out in the public, I'm going to replace
this drive with a western digital drive that is supported by Linux kernel
out of the box.  I will lose the atomic write capability but the database is
capable of 10,000 TPS with this drive, where it maxed at about 200 TPS with
rotory drives, however when it maxes at 10,000 the machine is maxed on CPU
at that point and realistically the load normally does not exceed about 400 TPS
so I think I'll be fine with the WD.

-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-
  Eskimo North Linux Friendly Internet Access, Shell Accounts, and Hosting.
    Knowledgeable human assistance, not telephone trees or script readers.
  See our web site: http://www.eskimo.com/ (206) 812-0051 or (800) 246-6874.

sakurachan00, 2022-03-13:
> Thanks, but because I need my machines to be secure and 5.14 does not provide this and the exploit is now out in the public, I'm going to replace this drive with a western digital drive that is [...]

I'd use LTS kernel branches as they'd be patched for stuff like this

bplein, 2022-03-13:
@4nanook Is this for a production environment in a work environment? These cards are a decade old, and that should be taken into consideration in production environments as well.

Even at home I have moved my primary data to NVMe U.2 drives because I know a kernel change won't render them useless, and use them instead for high performance temporary workspace and backups.

4nanook, 2022-03-13:
Unfortunately the kernels chosen for LTS releases have all provided crap
performance for my applications.  I'm really stretching the hardware to the
limit which is why I'm bothering with self-compiled kernels to begin with.
5.13 provided good performance, but not LTS, 5.14 and 5.15 both spent more time 
in system space than user and did not work at all well, 5.15 was the most 
recent LTS, 5.16 again good performance.  And so it goes.

-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-
  Eskimo North Linux Friendly Internet Access, Shell Accounts, and Hosting.
    Knowledgeable human assistance, not telephone trees or script readers.
  See our web site: http://www.eskimo.com/ (206) 812-0051 or (800) 246-6874.

 Thanks in advance for any help you can offer. Significant Linux kernel changes in 5.15 have broken this driver for now. The primary developer will eventually work on it but it's broken for now. Suggest going back to 5.14 or newer if you can.

4nanook, 2022-03-13:
Yes I realize that.  The reason I went with these cards was specifically
a recommendation of the MariaDB engineers.  And it did get me a database that
handled more than 10,000 TPS, so can't complain where on rotory drives I was
maxing out at a little more than 200 TPS, but I think even a modern regular
flash drive will get me where I need to be on database performance so thats
what I'm going to attempt.

-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-
  Eskimo North Linux Friendly Internet Access, Shell Accounts, and Hosting.
    Knowledgeable human assistance, not telephone trees or script readers.
  See our web site: http://www.eskimo.com/ (206) 812-0051 or (800) 246-6874.

Tourman36, 2022-03-13:
Honestly while these cards were good and performance can be good, we are
limited by only having access to the kernel source code.

I just at this point wouldn’t recommend these in any sort of critical
infrastructure. You risk the potential from data loss due to unforeseen
complications with the hw driver and kernel. You can poke around in our
discord channel and talk to us there.

You would really be better served by using NVME drives which are natively
supported in all 5.X kernels. They can be had for relatively cheap these
days and should offer similar if not better performance.

dbartelmus, 2022-03-13:
So consumer NVMe will be better than IO Drive 2 nowadays? IOPS are ok but what about access latency and small file throughput?

4nanook, 2022-03-13:
The NVMe drives don't offer atomic writes which was why the engineer at 
MariaDB recommended the Fusion I/O, however, it turned out atomic writes were 
not supported by the ext4 file system I was using anyway so never utilized that 
feature and still achieved 10,000 TPS, so I've bought WD Black 750's to replace 
the I/O fusion drives, been using one on my workstation for several years and 
so far no errors and smartctl still indicates another three years projected 
lifetime, so that's where I'm going for now.  And this is all just a stop-gap
while I build a newer server with quite a lot more resources.  This particular
server does many things but the largest portion of the load on the database is
a friendica social media site.  It is part of the fediverse and so handles a
large message volume.  We make daily backups to a RAIDED rotary drive file
system so worst that will happen is we'll lose a day of messages and/or 
wordpress updates.  The Linux folks have been tweaking the schedular and the
block I/O code quite a lot and both of those things can very heavily impact
the peformance, and the existing machine is a six core box running at 4.2 Ghz,
not enough, so building a 14-core 28-thread machine with twice the RAM that I
hope will give us a bit more overhead.

-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-
  Eskimo North Linux Friendly Internet Access, Shell Accounts, and Hosting.
    Knowledgeable human assistance, not telephone trees or script readers.
  See our web site: http://www.eskimo.com/ (206) 812-0051 or (800) 246-6874.

4nanook, 2022-03-13:
It probably is not going to perform as well as the IO-Fusion drive but
presently the Friendica load averages around 400 TPS, with rotary I was only
about to obtain about 250 TPS, so if the WD does even half that we are fine,
and the 10,000 TPS was CPU maxing out not the drive.  The majority of the
transactions are reads but seek performance on rotarys made even reads slow.
Because this machine is very much in the public eye it gets a lot more hacker
attention than I would like so security is particularly important.  One of the
reasons I'm putting a new machine together aside from more cores is more RAM
(it will have 256GB as opposed to 128GB of the existing machine) and more
PCI lanes so more I/O bandwidth.  More memory will enable it to cache more of
the data and more I/O will maximize the ability to utilize whatever hardware
it has attached.

-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-
  Eskimo North Linux Friendly Internet Access, Shell Accounts, and Hosting.
    Knowledgeable human assistance, not telephone trees or script readers.
  See our web site: http://www.eskimo.com/ (206) 812-0051 or (800) 246-6874.

bplein, 2022-03-14:
> So consumer NVMe will be better than IO Drive 2 nowadays? IOPS are ok but what about access latency and small file throughput?

I don't use consumer NVMe drives as a replacement for Fusion-io. I use Enterprise/Datacenter devices. https://ark.intel.com/content/www/us/en/ark/products/122579/intel-ssd-dc-p4510-series-4-0tb-2-5in-pcie-3-1-x4-3d2-tlc.html

4nanook, 2022-03-14:
That's great but the app doesn't justify a $5000 device.

-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-
  Eskimo North Linux Friendly Internet Access, Shell Accounts, and Hosting.
    Knowledgeable human assistance, not telephone trees or script readers.
  See our web site: http://www.eskimo.com/ (206) 812-0051 or (800) 246-6874.

bplein, 2022-03-14:
4TB NVMe U.2 drives are under $400 used.

I'm going to close this issue and re-open one without the chatter.

closed 2022-03-14 by bplein

4nanook, 2022-03-20:
> @4nanook Take a look at https://github.com/RemixVSL/iomemory-vsl/issues/100 and you can try out the code on `main`

I've ended up extracting the card from the machine as it conflicted with
a network card that I wanted to use to fix a bug in hardware offloading so
at the moment do not have a platform with which to participate.

-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-
  Eskimo North Linux Friendly Internet Access, Shell Accounts, and Hosting.
    Knowledgeable human assistance, not telephone trees or script readers.
  See our web site: http://www.eskimo.com/ (206) 812-0051 or (800) 246-6874.

4nanook, 2022-10-11:
I will give it a try but not until I get the new flash in place.  I
can't boot the machine into the new kernel without interrupting service and
I don't wish to do that more than I need to so going to get things moved,
on the new kernel, then can try the drivers again.

-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-
  Eskimo North Linux Friendly Internet Access, Shell Accounts, and Hosting.
    Knowledgeable human assistance, not telephone trees or script readers.
  See our web site: http://www.eskimo.com/ (206) 812-0051 or (800) 246-6874.

Tourman36, 2022-10-11:
I did not say consumer nvme. Enterprise grade sure - stay away from
consumer gear. Many of them have a low WPD amount and would not survive if
you need lots of small writes.

If budget is an issue, I’d go scour eBay. Lots of good second hand gear.
Make sure you know what the drive health is before purchasing.

IoDrive is really good for the price point, but is it relatively old
technology. We are basically enthusiasts trying to make it functional.

I believe we will still happily accept changes if anyone wants to fix
compatibility, but my points remain.