RemixVSL/iomemory-vsl #105: make dkms for specific kernel?

sotiris-bos, 2022-05-13:
Hi, I updated my system before updating the driver and dkms failed to compile for the new kernel.

After pulling the latest master and trying to build with `make dkms`, make tries to build for the current kernel and fails.

How can I force it to build for the new kernel that is not running yet? (I updated systemd and have not rebooted yet)

snuf, 2022-05-13:
Hi @sotiris-bos I think you're looking for this: https://man.archlinux.org/man/extra/dkms/dkms.8.en#k to add to the `dkms install -m <module> -v <version>` which you can get with `dkms status`.

sotiris-bos, 2022-05-14:
> Hi @sotiris-bos I think you're looking for this: https://man.archlinux.org/man/extra/dkms/dkms.8.en#k to add to the `dkms install -m <module> -v <version>` which you can get with `dkms status`.

Wouldn't that install the older module version to the new kernel version though? (that failed in my case)

What I am looking for is building the new module source against the new kernel that is going to run on the next reboot.

Ultimately I rebooted and ran make dkms after the new kernel was loaded but that meant that there was no driver for the IODrive on the first reboot.

Tourman36, 2022-05-14:
I've always had to reboot prior to compiling for a new kernel. Yes
unfortunately you will come up without ioDrive on first boot.

snuf, 2022-05-15:
@sotiris-bos I linked to the `-k` flag in the manual. The `-k` flag allows you to state the kernel you want `dkms` to do something for. So installing for a newer kernel is quite possible. 

The way to do it would be to remove the existing module from `dkms`. This does not unload the module, it just removes it from `dkms`. After which you install the module for your current kernel with `make dkms`, and manually add the kernel you're not running atm with `dkms install -m <module> -v <version> -k <kernel>`.

snuf, 2022-05-17:
@sotiris-bos @Tourman36 am closing this one, as it is possible, just beyond the scope of what we do. Otherwise the bells and whistles need to be added to the already somewhat bloated `module_operations.sh` script for dkms installation.

closed 2022-05-17 by snuf