RemixVSL/iomemory-vsl #107: [BUG] Function depreciated in kernel 5.17.11

gordon-zhao, 2022-06-03:
### Bug description
Several functions are renamed/depreciated from kernel 5.17.11
1. PDE_DATA() in kfile.c renamed to pde_data(). Ref: https://github.com/openzfs/zfs/issues/13004
2. GENHD_FL_EXT_DEVT flag removed. Ref: https://lore.kernel.org/all/20211122130625.1136848-12-hch@lst.de/

### How to reproduce
```
git clone https://github.com/snuf/iomemory-vsl.git
cd iomemory-vsl
git checkout <tag or some-branch>
make dkms
```

### Possible solution
1. Replacing PDE_DATA() with pde_data() seems to work
2. Unset GENHD_FL_EXT_DEVT flag according to the change

The driver compiles and able to detect and mount the drive.

### Environment information
System: Ubuntu 22.04 LTS w/ kernel 5.17.11 generic

#### Log 1:
```
  DKMS make.log for iomemory-vsl-5.17.11-051711-8fb7b04 for kernel 5.17.11-051711-generic (x86_64)
  Fri Jun  3 07:49:24 AM UTC 2022
  make[2]: Entering directory '/var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build'
  sed -i 's/Proprietary/GPL/g' Kbuild
  
  Change found in target kernel: KERNELVER KERNEL_SRC
  Running clean before building driver
  
  make[3]: Entering directory '/var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build'
  make \
          -j8 \
          -C /lib/modules/5.17.11-051711-generic/build \
          FIO_DRIVER_NAME=iomemory-vsl \
          FUSION_DRIVER_DIR=/var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build \
          M=/var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build \
          EXTRA_CFLAGS+="-I/var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall -Werror" \
          KFIO_LIB=kfio/x86_64_cc112_libkfio.o_shipped \
          clean
  make[4]: Entering directory '/usr/src/linux-headers-5.17.11-051711-generic'
  make[4]: Leaving directory '/usr/src/linux-headers-5.17.11-051711-generic'
  rm -rf include/fio/port/linux/kfio_config.h kfio_config license.c
  make[3]: Leaving directory '/var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build'
  if [ "112" -gt "74" ];then \
          if [ ! -f "kfio/x86_64_cc112_libkfio.o_shipped" ];then \
                  cp kfio/x86_64_cc74_libkfio.o_shipped kfio/x86_64_cc112_libkfio.o_shipped; \
          fi \
  fi
  ./kfio_config.sh -a x86_64 -o include/fio/port/linux/kfio_config.h -k /lib/modules/5.17.11-051711-generic/build -p -d /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kfio_config -l 0 -s /lib/modules/5.17.11-051711-generic/source
  Detecting Kernel Flags
  Config dir         : /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kfio_config
  Output file        : include/fio/port/linux/kfio_config.h
  Kernel output dir  : /lib/modules/5.17.11-051711-generic/build
  Kernel source dir  : /lib/modules/5.17.11-051711-generic/build
  Starting tests:
    1654242564.709  KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS...
    1654242564.739  KFIOC_X_TASK_HAS_CPUS_MASK...
    1654242564.769  KFIOC_X_LINUX_HAS_PART_STAT_H...
    1654242564.793  KFIOC_X_BLK_ALLOC_QUEUE_NODE_EXISTS...
    1654242564.817  KFIOC_X_BLK_ALLOC_DISK_EXISTS...
    1654242564.846  KFIOC_X_HAS_MAKE_REQUEST_FN...
    1654242564.875  KFIOC_X_GENHD_PART0_IS_A_POINTER...
    1654242564.908  KFIOC_X_BIO_HAS_BI_BDEV...
    1654242564.936  KFIOC_X_SUBMIT_BIO_RETURNS_BLK_QC_T...
    1654242564.964  KFIOC_X_VOID_ADD_DISK...
    1654242564.997  KFIOC_X_DISK_HAS_OPEN_MUTEX...
  Started tests, waiting for completions...
[... 153 lines cut ...]
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kfile.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kmem.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kfio_common.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kcpu.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kscatter.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/ktime.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/sched.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/cdev.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kblock.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kcondvar.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kinfo.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kexports.o
    CC [M]  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/khotplug.o
  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kblock.c: In function ‘kfio_expose_disk’:
  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kblock.c:370:17: error: ‘GENHD_FL_EXT_DEVT’ undeclared (first use in this function)
    370 |     gd->flags = GENHD_FL_EXT_DEVT;
        |                 ^~~~~~~~~~~~~~~~~
  /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kblock.c:370:17: note: each undeclared identifier is reported only once for each function it appears in
  make[4]: *** [scripts/Makefile.build:288: /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build/kblock.o] Error 1
  make[4]: *** Waiting for unfinished jobs....
  make[3]: *** [Makefile:1837: /var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build] Error 2
  make[3]: Leaving directory '/usr/src/linux-headers-5.17.11-051711-generic'
  make[2]: *** [Makefile:136: modules] Error 2
  make[2]: Leaving directory '/var/lib/dkms/iomemory-vsl/5.17.11-051711-8fb7b04/build'
```

snuf, 2022-06-03:
Hi @gordon-zhao thanks for the bug report and the sleuthing on a fix, highly appreciated!!!

I'll make a patch that will most probably go into `kblock_meta.h` as a macro. I'll leave this bug open till done.

snuf, 2022-12-15:
sorry forgot to update this a while ago. @gordon-zhao there is a #5.19-dev branch that contains all the fixes forward till that point, this will probably move to main soon.

snuf, 2022-12-22:
fixed, up to 6.1.1

closed 2022-12-22 by snuf