RemixVSL/iomemory-vsl #110: [BUG] Cannot build on 4.19.0+1

timolow, 2022-07-03:
### Bug description
Unable to build v4.20.3 on 4.19.0+1 (XCP-NG) Centos 7.5 host.
`[09:37 r720-xcp-ng iomemory-vsl]# uname -a
Linux r720-xcp-ng 4.19.0+1 #1 SMP Thu Jan 13 12:55:45 CET 2022 x86_64 x86_64 x86_64 GNU/Linux
[09:37 r720-xcp-ng iomemory-vsl]# git status
# On branch v4.20.3
nothing to commit, working directory clean
[09:37 r720-xcp-ng iomemory-vsl]# rpmbuild -ba fio-driver.spec
Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.ShKhKb
+ umask 022
+ cd /root/rpmbuild/BUILD
+ cd /root/rpmbuild/BUILD
+ rm -rf iomemory-vsl-3.2.16.1732
+ /usr/bin/mkdir -p iomemory-vsl-3.2.16.1732
+ cd iomemory-vsl-3.2.16.1732
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ rsync -rv /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/ driver_source/
sending incremental file list
created directory driver_source
./
Kbuild
Makefile
cdev.c
check_target_kernel.conf
check_target_kernel.sh
dkms.conf
driver_init.c
errno.c
kblock.c
kcache.c
kcondvar.c
kcpu.c
kcsr.c
kexports.c
kfile.c
kfio.c
kfio_common.c
kfio_config.sh
khotplug.c
kinfo.c
kmem.c
kscatter.c
kscsi.c
kscsi_host.c
ktime.c
main.c
module_operations.sh
module_param.c
pci.c
port-internal.h
sched.c
state.c
sysrq.c
include/
include/fio/
include/fio/port/
include/fio/port/align.h
include/fio/port/atomic_list.h
include/fio/port/bitops.h
include/fio/port/byteswap.h
include/fio/port/cdev.h
include/fio/port/commontypes.h
include/fio/port/compiler.h
include/fio/port/csr_simulator.h
include/fio/port/dbgset.h
include/fio/port/errno.h
include/fio/port/fio-poppack.h
include/fio/port/fio-port.h
include/fio/port/fio-pshpack1.h
include/fio/port/fio-stat.h
[... 487 lines cut ...]
### Environment information
Information about the system the module is used on
1. Linux kernel compiled against (uname -a)
Linux r720-xcp-ng 4.19.0+1 #1 SMP Thu Jan 13 12:55:45 CET 2022 x86_64 x86_64 x86_64 GNU/Linux

2. The C compiler version used (gcc --version)
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)

3. distribution, and version (cat /etc/os-release)
NAME="XCP-ng"
VERSION="8.2.1"
ID="xenenterprise"
ID_LIKE="centos rhel fedora"
VERSION_ID="8.2.1"
PRETTY_NAME="XCP-ng 8.2.1"
ANSI_COLOR="0;31"
HOME_URL="http://xcp-ng.org/"
BUG_REPORT_URL="https://github.com/xcp-ng/xcp"

4. Tag or Branch of iomemory-vsl that is being compiled
# On branch v4.20.3

5. FIO device used, if applicable
   * fio-status
   * lspci -b -nn

bplein, 2022-07-03:
Have you tried compiling from main? Main is far more up to date.

timolow, 2022-07-03:
Results from main branch:
'''[10:57 r720-xcp-ng ~]# git clone https://github.com/snuf/iomemory-vsl
Cloning into 'iomemory-vsl'...
remote: Enumerating objects: 2834, done.
remote: Counting objects: 100% (267/267), done.
remote: Compressing objects: 100% (54/54), done.
remote: Total 2834 (delta 243), reused 219 (delta 211), pack-reused 2567
Receiving objects: 100% (2834/2834), 12.14 MiB | 22.66 MiB/s, done.
Resolving deltas: 100% (1554/1554), done.
[10:57 r720-xcp-ng ~]# cd iomemory-vsl/
[10:57 r720-xcp-ng iomemory-vsl]# git status
# On branch main
nothing to commit, working directory clean
[10:57 r720-xcp-ng iomemory-vsl]# ls
debian  fio-driver.spec  Makefile  README  README.md  root  tools
[10:57 r720-xcp-ng iomemory-vsl]# rpmbuild -ba fio-driver.spec
Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.XzYrai
+ umask 022
+ cd /root/rpmbuild/BUILD
+ cd /root/rpmbuild/BUILD
+ rm -rf iomemory-vsl-3.2.16.1732
+ /usr/bin/mkdir -p iomemory-vsl-3.2.16.1732
+ cd iomemory-vsl-3.2.16.1732
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ rsync -rv /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/ driver_source/
sending incremental file list
created directory driver_source
./
Kbuild
Makefile
cdev.c
check_target_kernel.conf
check_target_kernel.sh
dkms.conf
driver_init.c
errno.c
kblock.c
kcache.c
kcondvar.c
kcpu.c
kcsr.c
kexports.c
kfile.c
kfio.c
kfio_common.c
kfio_config.sh
khotplug.c
kinfo.c
kmem.c
kscatter.c
kscsi.c
kscsi_host.c
ktime.c
main.c
module_operations.sh
module_param.c
pci.c
port-internal.h
sched.c
state.c
sysrq.c
include/
include/kblock_meta.h
include/fio/
include/fio/port/
include/fio/port/align.h
include/fio/port/atomic_list.h
include/fio/port/bitops.h
include/fio/port/byteswap.h
include/fio/port/cdev.h
[... 295 lines cut ...]
./include/linux/blkdev.h:1277:23: note: expected 'gfp_t' but argument is of type 'unsigned int (*)(struct request_queue *, struct bio *)'
 struct request_queue *blk_alloc_queue(gfp_t);
                       ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c:940:5: error: too many arguments to function 'blk_alloc_queue'
     rq = BLK_ALLOC_QUEUE;
     ^
In file included from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/port-internal.h:63:0,
                 from /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c:32:
./include/linux/blkdev.h:1277:23: note: declared here
 struct request_queue *blk_alloc_queue(gfp_t);
                       ^
cc1: all warnings being treated as errors
make[2]: *** [/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.o] Error 1
make[1]: *** [_module_/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source] Error 2
make[1]: Leaving directory `/usr/src/kernels/4.19.0+1-x86_64'
make: *** [modules] Error 2
make: Leaving directory `/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source'
+ set +x
ERROR:

error: Bad exit status from /var/tmp/rpm-tmp.KpbWfV (%build)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.KpbWfV (%build)
[10:58 r720-xcp-ng iomemory-vsl]#'''

snuf, 2022-07-03:
@timolow please try the fix mentioned in #106, branch bug-#106. The fix is not merged yet as I need to swap out cards to test with vsl, instead of vsl4.

timolow, 2022-07-04:
Results on bug-#106

```
[14:33 r720-xcp-ng iomemory-vsl]# git status
# On branch bug-#106
nothing to commit, working directory clean
[14:33 r720-xcp-ng iomemory-vsl]# rpmbuild -ba fio-driver.spec
Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.wfUOrH
+ umask 022
+ cd /root/rpmbuild/BUILD
+ cd /root/rpmbuild/BUILD
+ rm -rf iomemory-vsl-3.2.16.1732
+ /usr/bin/mkdir -p iomemory-vsl-3.2.16.1732
+ cd iomemory-vsl-3.2.16.1732
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ rsync -rv /root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/ driver_source/
sending incremental file list
created directory driver_source
./
Kbuild
Makefile
cdev.c
check_target_kernel.conf
check_target_kernel.sh
dkms.conf
driver_init.c
errno.c
kblock.c
kcache.c
kcondvar.c
kcpu.c
kcsr.c
kexports.c
kfile.c
kfio.c
kfio_common.c
kfio_config.sh
khotplug.c
kinfo.c
kmem.c
kscatter.c
kscsi.c
kscsi_host.c
ktime.c
main.c
module_operations.sh
module_param.c
pci.c
port-internal.h
sched.c
state.c
sysrq.c
include/
include/kblock_meta.h
include/fio/
include/fio/port/
include/fio/port/align.h
include/fio/port/atomic_list.h
include/fio/port/bitops.h
include/fio/port/byteswap.h
include/fio/port/cdev.h
include/fio/port/commontypes.h
include/fio/port/compiler.h
include/fio/port/csr_simulator.h
include/fio/port/dbgset.h
include/fio/port/errno.h
include/fio/port/fio-poppack.h
include/fio/port/fio-port.h
include/fio/port/fio-pshpack1.h
include/fio/port/fio-stat.h
[... 243 lines cut ...]
         ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c:508:60: error: macro "part_stat_add" requires 4 arguments, but only 3 given
         part_stat_add(GD_PART0, nsecs[0],   duration * 1000);
                                                            ^
/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.c:504:25: error: unused variable 'gd' [-Werror=unused-variable]
         struct gendisk *gd = fgd->gd;
                         ^
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/module_param.o
cc1: all warnings being treated as errors
  CC [M]  /root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/license.o
make[2]: *** [/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source/kblock.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [_module_/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source] Error 2
make[1]: Leaving directory `/usr/src/kernels/4.19.0+1-x86_64'
make: *** [modules] Error 2
make: Leaving directory `/root/rpmbuild/BUILD/iomemory-vsl-3.2.16.1732/driver_source'
+ set +x
ERROR:

error: Bad exit status from /var/tmp/rpm-tmp.wVjIfz (%build)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.wVjIfz (%build)
[14:34 r720-xcp-ng iomemory-vsl]#
```

gjsmo, 2022-07-08:
Just want to add that I have the same issue, on the same platform (XCP-ng 8.2.1, kernel 4.19.0+1). I'm considering using PCIe passthrough to get around this, but it would be nice to not have the overhead.

closed 2022-09-08 by snuf