RemixVSL/iomemory-vsl #124: Can't successfully compile when using 6.6 & 6.7 kernels, tested on Fedora 39.

PanosPetrou, 2023-11-29:
### Bug description
Can't successfully compile when using 6.6 & 6.7 kernels, tested on Fedora 39.
 
### How to reproduce
git clone https://github.com/snuf/iomemory-vsl
cd iomemory-vsl
make module

** poof, broken token **
iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/sysrq.c:63:16: error: initialization of ‘void (*)(u8)’ {aka ‘void (*)(unsigned char)’} from incompatible pointer type ‘void (*)(int)’ [-Werror=incompatible-pointer-types]

iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/sysrq.c:70:16: error: initialization of ‘void (*)(u8)’ {aka ‘void (*)(unsigned char)’} from incompatible pointer type ‘void (*)(int)’ [-Werror=incompatible-pointer-types]

### Possible solution
Edited line 49 of the file root/usr/src/iomemory-vsl-3.2.16/sysrq.c
From: void iodrive_handle_sysrq(int key)
To:      void iodrive_handle_sysrq(unsigned char key)

Compiles successfully now.

### Environment information
Fedora 39, kernel-6.6.2-201.fc39.x86_64
Also tested on kernels 6.5 & 6.7 from rawhide

fio-status:
Found 1 ioMemory device in this system
Driver version: 3.2.16 build 1731

Adapter: Single Controller Adapter
	HP 1205GB MLC PCIe ioDrive2 for ProLiant Servers, Product Number:<omitted>, SN:<omitted>
	External Power: NOT connected
	PCIe Power limit threshold: 24.75W
	Connected ioMemory modules:
	  fct0:	Product Number:<omitted>, SN:<omitted>

fct0	Attached
	ioDrive2 Adapter Controller, Product Number:<omitted>, SN:<omitted>
	Located in slot 0 Center of ioDrive2 Adapter Controller SN:<omitted>
	PCI:06:00.0, Slot Number:32
	Firmware v7.1.17, rev 116786 Public
	1205.00 GBytes device size
	Internal temperature: 68.41 degC, max 69.89 degC
	Reserve space status: Healthy; Reserves: 100.00%, warn at 10.00%
	Contained VSUs:
	  fioa:	ID:0, UUID:<omitted>

fioa	State: Online, Type: block device
	ID:0, UUID:<omitted>
	1205.00 GBytes device size

PanosPetrou, 2023-11-29:
On fio-status, I have omitted Product Number, SN and the UUID of the volume.

snuf, 2023-11-30:
@PanosPetrou thanks for reporting and providing a fix. I've created a branch [bug#124](https://github.com/RemixVSL/iomemory-vsl/tree/bug%23124) which is going through testing atm. Will merge to main after testing comes back positive.

snuf, 2023-11-30:
Test came back clean, removed branch and closing this.