RemixVSL/iomemory-vsl #126: Can say you are awesome?

cdoublejj, 2024-01-13:
I saw that Kernel 6.6 is supported, i still have my iodrive because it's 3.4tb SSD and that's plenty. 

Thank You!

cdoublejj, 2024-01-13:
ok i've done some reaing is this pretty stable with newer kernels? it seems newer kernels do not like some things? I'm looking to get off xubuntu to popos and was doing research and re-found this github, it's awesome it's still kicking btw,  but, i see my current linux install is kernel 5.4 vs newer kernels in popos. 6.x

snuf, 2024-01-13:
@cdoublejj the driver goes through a test suite. If it states "tested" it means it works without losing data or any major problems/errors as far as the test-suite covers. You should be fine with any 5 or 6 kernel afaik.

cdoublejj, 2024-01-13:
U+2764

closed 2024-01-16 by snuf