RemixVSL/iomemory-vsl #131: [BUG] Build is failing on kernel 6.9.2

AnalogBot, 2024-06-01:
### Bug description
Build is failing on 6.9.2

### How to reproduce

```
git clone https://github.com/snuf/iomemory-vsl.git
cd iomemory-vsl
make module

cd root/usr/src/iomemory-vsl-3.2.16 && \
  		make clean
make[1]: Entering directory '/home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
make \
        -j32 \
	-C /lib/modules/6.9.2-arch1-1/build \
	FIO_DRIVER_NAME=iomemory-vsl \
	FUSION_DRIVER_DIR=/home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
	M=/home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
	EXTRA_CFLAGS+="-I/home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall -Werror" \
	KFIO_LIB=kfio/x86_64_cc141_libkfio.o_shipped \
	clean
make[2]: Entering directory '/usr/lib/modules/6.9.2-arch1-1/build'
make[2]: Leaving directory '/usr/lib/modules/6.9.2-arch1-1/build'
rm -rf include/fio/port/linux/kfio_config.h kfio_config
make[1]: Leaving directory '/home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
cd root/usr/src/iomemory-vsl-3.2.16 && \
  		make gpl
make[1]: Entering directory '/home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
sed -i 's/Proprietary/GPL/g' Kbuild
if [ "141" -gt "74" ];then \
	if [ ! -f "kfio/x86_64_cc141_libkfio.o_shipped" ];then \
		cp kfio/x86_64_cc74_libkfio.o_shipped kfio/x86_64_cc141_libkfio.o_shipped; \
	fi \
fi
./kfio_config.sh -a x86_64 -o include/fio/port/linux/kfio_config.h -k /lib/modules/6.9.2-arch1-1/build -p -d /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config -l 0 -s /lib/modules/6.9.2-arch1-1/source
Detecting Kernel Flags
Config dir         : /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config
Output file        : include/fio/port/linux/kfio_config.h
Kernel output dir  : /lib/modules/6.9.2-arch1-1/build
Kernel source dir  : /lib/modules/6.9.2-arch1-1/build
Starting tests:
  1717205618.730  KFIOC_X_HANDLE_SYSRQ_IS_U8...
  1717205618.769  KFIOC_X_BDOPS_OPEN_GENDISK_AND_BLK_MODE_T...
  1717205618.804  KFIOC_X_BDOPS_RELEASE_1_ARG...
  1717205618.840  KFIOC_X_CAPS_PDE_DATA...
  1717205618.877  KFIOC_X_BIO_SPLIT_TO_LIMITS...
  1717205618.913  KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS...
  1717205618.948  KFIOC_X_TASK_HAS_CPUS_MASK...
  1717205618.984  KFIOC_X_BLK_ALLOC_QUEUE_NODE_EXISTS...
  1717205619.022  KFIOC_X_BLK_ALLOC_QUEUE_EXISTS...
  1717205619.058  KFIOC_X_BLK_ALLOC_DISK_EXISTS...
  1717205619.094  KFIOC_X_HAS_MAKE_REQUEST_FN...
  1717205619.131  KFIOC_X_GENHD_PART0_IS_A_POINTER...
  1717205619.169  KFIOC_X_BIO_HAS_BI_BDEV...
  1717205619.209  KFIOC_X_SUBMIT_BIO_RETURNS_BLK_QC_T...
  1717205619.251  KFIOC_X_VOID_ADD_DISK...
  1717205619.295  KFIOC_X_DISK_HAS_OPEN_MUTEX...
Started tests, waiting for completions...
  1717205620.358  KFIOC_X_HANDLE_SYSRQ_IS_U8=1
  1717205620.379  KFIOC_X_BDOPS_OPEN_GENDISK_AND_BLK_MODE_T=1
  1717205620.400  KFIOC_X_BDOPS_RELEASE_1_ARG=1
  1717205620.421  KFIOC_X_CAPS_PDE_DATA=0
  1717205620.441  KFIOC_X_BIO_SPLIT_TO_LIMITS=1
  1717205620.464  KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS=1
  1717205620.486  KFIOC_X_TASK_HAS_CPUS_MASK=1
  1717205620.507  KFIOC_X_BLK_ALLOC_QUEUE_NODE_EXISTS=0
  1717205620.527  KFIOC_X_BLK_ALLOC_QUEUE_EXISTS=0
  1717205620.549  KFIOC_X_BLK_ALLOC_DISK_EXISTS=0
  1717205620.570  KFIOC_X_HAS_MAKE_REQUEST_FN=0
  1717205620.593  KFIOC_X_GENHD_PART0_IS_A_POINTER=1
  1717205620.613  KFIOC_X_BIO_HAS_BI_BDEV=1
  1717205620.635  KFIOC_X_SUBMIT_BIO_RETURNS_BLK_QC_T=0
  1717205620.657  KFIOC_X_VOID_ADD_DISK=0
  1717205620.680  KFIOC_X_DISK_HAS_OPEN_MUTEX=1
Finished
1717205620.690  Exiting
Preserving configdir due to '-p' option: /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_config

Adding module version 6.9.2-arch1-0a12f15-3.2.16 to source license.c

make \
    -j32 \
-C /lib/modules/6.9.2-arch1-1/build \
FIO_DRIVER_NAME=iomemory-vsl \
FUSION_DRIVER_DIR=/home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
M=/home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
EXTRA_CFLAGS+="-I/home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall -Werror" \
INSTALL_MOD_DIR=extra/fio \
INSTALL_MOD_PATH= \
KFIO_LIB=kfio/x86_64_cc141_libkfio.o_shipped \
modules
make[2]: Entering directory '/usr/lib/modules/6.9.2-arch1-1/build'
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/main.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/license.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/pci.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/sysrq.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/driver_init.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/errno.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/state.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kcache.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfile.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kmem.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio_common.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kcpu.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kscatter.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/ktime.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/sched.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/cdev.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kblock.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kcondvar.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kinfo.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kexports.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/khotplug.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kcsr.o
  COPY    /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/kfio/x86_64_cc141_libkfio.o
  CC [M]  /home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/module_param.o
make[2]: Leaving directory '/usr/lib/modules/6.9.2-arch1-1/build'
make[1]: Leaving directory '/home/myuser/temp/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'

```

### Possible solution
Note: I'm not really familiar with kernel dev, I don't even really know enough C to be anything other then dangerous. But I tried to dig in to see if it was something simple. This is what I found, but it's outside of my capabilities to fix.
 
Looks like in 6.9 these functions now require an additional 'queue_limits' argument. 
blk_alloc_disk
blk_mq_alloc_disk

https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.9 - Looks like most in tree block device drivers were updated to support this change. 
https://github.com/openzfs/zfs/pull/16033/commits/c5ad47920fdbf20dfe56f77368a15568c17b57eb - ZFS also recently updated  to support.

### Environment information
Information about the system the module is used on
1. Linux kernel compiled against (uname -a) - Linux thebria 6.9.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Sun, 26 May 2024 01:30:29 +0000 x86_64 GNU/Linux
2. The C compiler version used (gcc --version) - gcc (GCC) 14.1.1 20240522
3. distribution, and version (cat /etc/os-release) - Arch Linux (rolling)
4. Tag or Branch of iomemory-vsl that is being compiled - Main
5. FIO device used, if applicable
   * fio-status
   * lspci -b -nn

fio-status:
driver isn't loaded

lspci -b -nn
0c:00.0 Mass storage controller [0180]: SanDisk ioDrive2 [1aed:2001] (rev 04)

snuf, 2024-06-02:
@AnalogBot thanks for the detailed report, and doing some sleuthing! I've made a patch for vsl4 that I can test later and then backport to vsl . The single card in my dev box is vsl4 atm, hence doing that one first: https://github.com/RemixVSL/iomemory-vsl4/compare/main...vsl%23131. The MQ code is defunct atm afaik, but I've updated it irrespective.

snuf, 2024-06-03:
@AnalogBot please check https://github.com/RemixVSL/iomemory-vsl/tree/bug%23131, which contains the fix. I'll merge it with master if you can confirm it works as desired? (otherwise we have to wait for me to swap cards at some point)

AnalogBot, 2024-06-04:
https://github.com/RemixVSL/iomemory-vsl/tree/bug%23131 builds and the module loads without issue. 
The device appears to be operating normally with this version loaded. 

Looks like that fixes it. Thank you for your work on this!

snuf, 2024-06-04:
@AnalogBot thanks for the bug report and the feedback!

closed 2024-06-04 by snuf

PanosPetrou, 2024-06-19:
I can confirm it works with kernel 6.9.4 on Fedora 40. Great work guys!