RemixVSL/iomemory-vsl #133: [BUG] directions do nmot mention systemd?

cdoublejj, 2024-07-04:
### Bug description
directions do not mention systemd at all

popos uses sytemd instead of grub. do i need to join the discord?

bplein, 2024-07-04:
Do you mean systemd-boot?

You would have to update your conf file with the kernel command line options just the same as with grub. I can’t tell you how to do that as I have only ever used GRUB in the past decade plus. 

Since popOs! Seems to be built on Debian, I would install the utilities using the .deb (see the Downloads repo under this same org) and I like the DKMS install method and would start there first.

cdoublejj, 2024-07-04:
i did follow dkms and was able ot build some .deb files. idk if i did it right. the iodrive does not show up as a disk in popos. it's been years since installed the  driver successfully on lubntu 18.04 but. now i need to update from lubuntu to popos yada yada.

i'm not sure if it's the lack of the book arguments or failure to install correctly. does installing automatically do the dkms stuff when kernel updates come down from system 76 that the io driver auto installs?

am in too far over my head? https://forum.level1techs.com/t/xubuntu-20-04-lts-how-do-i-compile-and-install-this-ssd-driver/159683/34

bplein, 2024-07-04:
These aren’t like normal drives. I have no idea what popOS is looking for but these aren’t NVMe or SATA.  And you have to use the utilities to attach and detach them. 

from a terminal, run `fio-status -a` and `dmesg | grep fio` and that will get us started.

snuf, 2024-07-05:
@cdoublejj joining the discord always helps for interactivity. 

I've not used `systemd-boot` before, but looking at the documentation that arch provides [here](https://wiki.archlinux.org/title/systemd-boot#Adding_loaders) it seems like it's matter of adding the options required as one would with grub to the specific configuration you use to boot. I've made a [patch](https://github.com/RemixVSL/iomemory-vsl/blob/snuf-patch-1/README.md) for the README that contains this information .

As @bplein pointed out the drive is not a "normal" drive, and providing the information he's asking for helps us to help you. 

I looked at the post on level1techs, it seems there is some confusion. The PPA that is referred to has nothing to do with this project, it has also not been updated for at least a decade. There is also some confusion around what version to use. You seem to be on Pop_OS with kernel 6.9.3, which is not a 5.x kernel. We've switched to using the `main` branch as the leading branch for everything. This is reflected in the current README. Though the tools were not referred to in the README. I've added a link to [vsl_downloads](https://github.com/RemixVSL/vsl_downloads) where the tools reside. Please install the tools, and post a `fio-status -a` and the `dmesg | grep fio` as @bplein requested.

cdoublejj, 2024-07-07:
> These aren’t like normal drives. I have no idea what popOS is looking for but these aren’t NVMe or SATA. And you have to use the utilities to attach and detach them.
> 
> from a terminal, run `fio-status -a` and `dmesg | grep fio` and that will get us started.

[image]

can this attached image be seen?

bplein, 2024-07-08:
You haven’t installed the utils per my note above and dmesg requires root permissions. Have you used Linux much?

cdoublejj, 2024-07-12:
I installed these four deb files made from the github make/compile

[image]

BTW I did choose MAIN!!! I have been reading through the documentation here. i'm going to read through your patched readme and that link. i'm slow. i apologize. i will do my best to do home work and put in some effort here as time allows.

[image]

i ran as sudo and for reference wanted to show the deb files i got when going through the github directions for MAIN.

EDIT: maybe i'm liar, 3 deb files.

[image]

snuf, 2024-08-24:
@cdoublejj documentation has been updated to include systemd, does that resolve the initial documentation request? Also are you up and running?

cdoublejj, 2024-08-27:
i am not up and running, it sounds like i need to let you all close this and join discord sometime. also there is a brand new version of popos 24.04LTS coming out with weyland and cosmic desktop and thay may haven even newer kernel?

snuf, 2024-12-13:
@cdoublejj kay closing this one, don't hesitate to join the discord when you have time or are ready for some help.

closed 2024-12-13 by snuf