RemixVSL/iomemory-vsl #135: Fails to compile on 6.10 kernels. Tested on Fedora 40.

PanosPetrou, 2024-08-16:
### Bug description
Fails to compile on 6.10 kernels. Tested on Fedora 40.

### How to reproduce
git clone https://github.com/snuf/iomemory-vsl.git
cd iomemory-vsl
make dkms

/var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.c: In function ‘kfio_create_disk’:
/var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.c:347:5: σφάλμα: implicit declaration of function ‘blk_queue_max_hw_sectors’; did you mean ‘queue_max_hw_sectors’? [-Wimplicit-function-declaration]
  347 |     blk_queue_max_hw_sectors(rq, max_sectors_per_request);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
      |     queue_max_hw_sectors
/var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.c:348:5: σφάλμα: implicit declaration of function ‘blk_queue_max_segments’; did you mean ‘queue_max_segments’? [-Wimplicit-function-declaration]
  348 |     blk_queue_max_segments(rq, max_sg_elements_per_request);
      |     ^~~~~~~~~~~~~~~~~~~~~~
      |     queue_max_segments
  CC [M]  /var/lib/dkms/iomemory-vsl/3.2.16/build/kcsr.o
/var/lib/dkms/iomemory-vsl/3.2.16/build/kblock.c:349:5: σφάλμα: implicit declaration of function ‘blk_queue_max_segment_size’; did you mean ‘queue_max_segment_size’? [-Wimplicit-function-declaration]
  349 |     blk_queue_max_segment_size(rq, PAGE_SIZE);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |     queue_max_segment_size

snuf, 2024-08-24:
@PanosPetrou thanks for filing!

There is a fix in branch [bug-#135](https://github.com/RemixVSL/iomemory-vsl/tree/bug-%23135). This has been tested on fedora-40 (6.10), archlinux (6.10), ubuntu 24.04 (6.8), ubuntu 22.04 (5.15) and came back clean without errors.

The only issue is that with the 6.10 kernel it will throw a [thunk return warning](https://github.com/torvalds/linux/commit/4461438a8405e800f90e0e40409e5f3d07eed381), when loading and unloading, which does not impact functionality (yet). There are some compile options that mitigate the warning but also surface another warning from objtool. This requires some more figuring out to remove the thunk return and resolve the objtool warning.

If you're comfortable and require the driver to work `now`, the last commit in the [bug-#135](https://github.com/RemixVSL/iomemory-vsl/tree/bug-%23135) should be safe to use.

PanosPetrou, 2024-08-26:
Great, thank you!

snuf, 2024-08-28:
Keeping this open for tracking.

snuf, 2024-10-09:
Fix is in main, and tested.

PanosPetrou, 2024-10-09:
I confirm it compiles successfully in Fedora 40, kernel-6.10.12-200.fc40.x86_64

snuf, 2024-10-10:
Was still an issue, as objtool with older kernels would "complain". Fix is in [main](https://github.com/RemixVSL/iomemory-vsl/compare/c20764e26e9d...000a801c71e6) for that. 

@PanosPetrou thanks for verifying

closed 2024-10-10 by snuf