RemixVSL/iomemory-vsl #138: [BUG] Upgrading Ubuntu 24.04 to 24.10 breaks build

mattmattox, 2024-10-22:
### Bug description
After upgrading to Ubuntu 24.10 `make dpkg` and `make dkms` fails to build

### How to reproduce
What are the steps to reproduce the reported issue.
- Deploy on Ubuntu 24.04
- Upgrade to Ubuntu 24.10
- The device will be missing on reboot
- `make dkms` fails with the error below

### Possible solution

### Environment information
Information about the system the module is used on
1. Linux kernel compiled against (uname -a)
`Linux a0ublokip01 6.11.0-9-generic #9-Ubuntu SMP PREEMPT_DYNAMIC Mon Oct 14 13:19:59 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux`
2. The C compiler version used (gcc --version) 
`gcc (Ubuntu 14.2.0-4ubuntu2) 14.2.0`
3. distribution, and version (cat /etc/os-release) 
```
PRETTY_NAME="Ubuntu 24.10"
NAME="Ubuntu"
VERSION_ID="24.10"
VERSION="24.10 (Oracular Oriole)"
VERSION_CODENAME=oracular
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=oracular
LOGO=ubuntu-logo
```
4. Tag or Branch of iomemory-vsl that is being compiled
main
5. FIO device used, if applicable
   * fio-status
   * lspci -b -nn
```   
Found 1 ioMemory device in this system
Driver version: Driver not loaded

Adapter: Single Controller Adapter
	Dell ioDrive2 1205GB MLC, Product Number:DG1MH, SN:US0DG1MH7605127R0024
	External Power: NOT connected
	PCIe Power limit threshold: Disabled
	Connected ioMemory modules:
	  44:00.0:	Product Number:DG1MH, SN:1229D3710

44:00.0	ioDrive2 Adapter Controller, Product Number:DG1MH, SN:1229D3710
	PCI:44:00.0
	Firmware v7.1.17, rev 116786 Public
	Internal temperature: 67.92 degC, max 69.40 degC
```

### Error log

```
/root/iomemory-vsl/driver_source/kblock.c: In function ‘kfio_create_disk’:
/root/iomemory-vsl/driver_source/include/kblock_meta.h:24:26: error: implicit declaration of function ‘alloc_disk’; did you mean ‘alloc_uid’? [-Wimplicit-function-declaration]
   24 |   #define BLK_ALLOC_DISK alloc_disk
      |                          ^~~~~~~~~~
/root/iomemory-vsl/driver_source/kblock.c:329:18: note: in expansion of macro ‘BLK_ALLOC_DISK’
  329 |         dp->gd = BLK_ALLOC_DISK(FIO_NUM_MINORS);
      |                  ^~~~~~~~~~~~~~
/root/iomemory-vsl/driver_source/kblock.c:329:16: error: assignment to ‘struct gendisk *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  329 |         dp->gd = BLK_ALLOC_DISK(FIO_NUM_MINORS);
      |                ^
/root/iomemory-vsl/driver_source/kblock.c:347:5: error: implicit declaration of function ‘blk_queue_max_hw_sectors’; did you mean ‘queue_max_hw_sectors’? [-Wimplicit-function-declaration]
[... 66 lines cut ...]
      |     ^~~~~~~~
/root/iomemory-vsl/driver_source/kblock.c: In function ‘kfio_alloc_queue’:
/root/iomemory-vsl/driver_source/include/kblock_meta.h:22:29: error: implicit declaration of function ‘blk_alloc_queue’; did you mean ‘blk_mq_alloc_queue’? [-Wimplicit-function-declaration]
   22 |     #define BLK_ALLOC_QUEUE blk_alloc_queue(node);
      |                             ^~~~~~~~~~~~~~~
/root/iomemory-vsl/driver_source/kblock.c:960:10: note: in expansion of macro ‘BLK_ALLOC_QUEUE’
  960 |     rq = BLK_ALLOC_QUEUE;
      |          ^~~~~~~~~~~~~~~
/root/iomemory-vsl/driver_source/kblock.c:960:8: error: assignment to ‘struct request_queue *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  960 |     rq = BLK_ALLOC_QUEUE;
      |        ^
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:244: /root/iomemory-vsl/driver_source/kblock.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [/usr/src/linux-headers-6.11.0-9-generic/Makefile:1931: /root/iomemory-vsl/driver_source] Error 2
make[3]: *** [Makefile:224: __sub-make] Error 2
make[3]: Leaving directory '/usr/src/linux-headers-6.11.0-9-generic'
make[2]: *** [Makefile:134: modules] Error 2
make[2]: Leaving directory '/root/iomemory-vsl/driver_source'
ERROR:
make[1]: *** [debian/rules:102: build-arch-stamp] Error 1
make[1]: Leaving directory '/root/iomemory-vsl'
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
make: *** [Makefile:16: dpkg] Error 2
```

snuf, 2024-10-22:
@mattmattox can you post a `git branch -v` to see what hash you are on?

mattmattox, 2024-10-22:
```
* main 0a12f15 Merge pull request #127 from RemixVSL/fix_missing_proto
```

snuf, 2024-10-22:
@mattmattox please do a `git pull`. 6.11 is not supported on your tag. You're living in March atm, and the last commits were in October.

mattmattox, 2024-10-22:
OK, I got a new error

```
root@a0ublokip01:~/iomemory-vsl# git pull .
From .
 * branch            HEAD       -> FETCH_HEAD
Already up to date.
root@a0ublokip01:~/iomemory-vsl# git status -v
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   debian/control

no changes added to commit (use "git add" and/or "git commit -a")
root@a0ublokip01:~/iomemory-vsl# make dpkg
cd root/usr/src/iomemory-vsl-3.2.16 && \
  		make clean
make[1]: Entering directory '/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
FIOCC is cc
make \
        -j24 \
	-C /lib/modules/6.11.0-9-generic/build \
	FIO_DRIVER_NAME=iomemory-vsl \
	FUSION_DRIVER_DIR=/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
	M=/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16 \
	EXTRA_CFLAGS+="-I/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall -mfunction-return=keep -Werror" \
	KFIO_LIB=kfio/x86_64_cc142_libkfio.o_shipped \
	clean
make[2]: Entering directory '/usr/src/linux-headers-6.11.0-9-generic'
make[2]: Leaving directory '/usr/src/linux-headers-6.11.0-9-generic'
rm -rf include/fio/port/linux/kfio_config.h kfio_config
make[1]: Leaving directory '/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
cd root/usr/src/iomemory-vsl-3.2.16 && \
	make patch_module_version
make[1]: Entering directory '/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
FIOCC is cc

Adding module version 6.11.0-9-79fe888-3.2.16 to source license.c

make[1]: Leaving directory '/root/iomemory-vsl/root/usr/src/iomemory-vsl-3.2.16'
# patch fio_version, fio_short_version in debian/fio_values
cd /root/iomemory-vsl && \
	dpkg-buildpackage -rfakeroot --no-check-builddeps --no-sign
dpkg-buildpackage: info: source package iomemory-vsl
dpkg-buildpackage: info: source version 3.2.16.1732
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Support <support@fusionio.com>
dpkg-buildpackage: info: host architecture amd64
dpkg-buildpackage: warning: using a gain-root-command while being root
 dpkg-source --before-build .
 fakeroot debian/rules clean
make[1]: Entering directory '/root/iomemory-vsl'
dh_testdir
dh_testroot
rm -f build-arch-stamp configure-stamp
# Add here commands to clean up after the build process.
#/usr/bin/make clean
rm -rf \
	"/root/iomemory-vsl/driver_source" \
	"/root/iomemory-vsl/debian/fio" \
	"/root/iomemory-vsl/debian/iomemory-vsl-6.11.0-9-generic"* \
	"/root/iomemory-vsl/debian/iomemory-vsl-6.11.0-9-generic"*
sed -r -i -e 's/^Package: iomemory-vsl(-config)?(-[.0-9]+.*)*$/Package: iomemory-vsl\1/;' debian/control
dh_prep
make[1]: Leaving directory '/root/iomemory-vsl'
 dpkg-source -b .
dpkg-source: warning: no source format specified in debian/source/format, see dpkg-source(1)
[... 487 lines cut ...]
/root/iomemory-vsl/driver_source/kblock.o: warning: objtool: kfio_expose_disk+0x192: 'naked' return found in MITIGATION_RETHUNK build
/root/iomemory-vsl/driver_source/kblock.o: warning: objtool: kfio_disk_stat_write_update+0xd2: 'naked' return found in MITIGATION_RETHUNK build
/root/iomemory-vsl/driver_source/kblock.o: warning: objtool: kfio_disk_stat_read_update+0xd0: 'naked' return found in MITIGATION_RETHUNK build
/root/iomemory-vsl/driver_source/kblock.o: warning: objtool: kfio_get_gd_in_flight+0xc: 'naked' return found in MITIGATION_RETHUNK build
/root/iomemory-vsl/driver_source/kblock.o: warning: objtool: kfio_set_gd_in_flight+0xa: 'naked' return found in MITIGATION_RETHUNK build
/root/iomemory-vsl/driver_source/kblock.o: warning: objtool: kfio_set_write_holdoff+0x12: 'naked' return found in MITIGATION_RETHUNK build
/root/iomemory-vsl/driver_source/kblock.o: warning: objtool: kfio_clear_write_holdoff+0x3b: 'naked' return found in MITIGATION_RETHUNK build
/root/iomemory-vsl/driver_source/kblock.o: warning: objtool: kfio_mark_lock_pending+0xa: 'naked' return found in MITIGATION_RETHUNK build
/root/iomemory-vsl/driver_source/kblock.o: warning: objtool: kfio_unmark_lock_pending+0xa: 'naked' return found in MITIGATION_RETHUNK build
/root/iomemory-vsl/driver_source/kblock.o: warning: objtool: kfio_vectored_atomic+0x6c: 'naked' return found in MITIGATION_RETHUNK build
/root/iomemory-vsl/driver_source/kblock.o: warning: objtool: kfio_count_sectors_inuse+0x29: 'naked' return found in MITIGATION_RETHUNK build
make[5]: *** [scripts/Makefile.build:244: /root/iomemory-vsl/driver_source/kblock.o] Error 214
make[5]: *** Deleting file '/root/iomemory-vsl/driver_source/kblock.o'
make[4]: *** [/usr/src/linux-headers-6.11.0-9-generic/Makefile:1931: /root/iomemory-vsl/driver_source] Error 2
make[3]: *** [Makefile:224: __sub-make] Error 2
make[3]: Leaving directory '/usr/src/linux-headers-6.11.0-9-generic'
make[2]: *** [Makefile:145: modules] Error 2
make[2]: Leaving directory '/root/iomemory-vsl/driver_source'
ERROR:
make[1]: *** [debian/rules:102: build-arch-stamp] Error 1
make[1]: Leaving directory '/root/iomemory-vsl'
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
make: *** [Makefile:16: dpkg] Error 2
root@a0ublokip01:~/iomemory-vsl# 
```

snuf, 2024-10-22:
@mattmattox ah yes the "ubuntu-ism". There is something with Ubuntu and compiling the kernel that's different from other distributions. ATM we check if the kernel is 6.10 and we adapt the CFLAGS for the newer kernel. This works for Fedora and Arch, which it was tested on, but as it seems not for Ubuntu 24.10. This is not a total surprise.

Atm I'm having some issues getting Ubuntu 24.10 into the test environment, as the cloud-image is not behaving as expected. The fix will probably be to have the CFLAGS change based on what the kernel build scripts have configured.

snuf, 2024-10-31:
@mattmattox you can use the [rerethunk](https://github.com/RemixVSL/iomemory-vsl/tree/rerethunk) branch for now. The main issue is that arch and fedora are not happy at module init, though ubuntu 24.10 seems perfectly fine with it. I've removed the CFLAG that caused it in this branch. Once I have a fix for fio_init_driver to not thunk on the other two I'll push that in here and that will end up in main.

mattmattox, 2024-11-01:
Awesome, let me know if there is anything I can do to help

snuf, 2024-12-13:
[rerethunk](https://github.com/RemixVSL/iomemory-vsl/tree/rerethunk) has been merged to main a while back, closing this.

closed 2024-12-13 by snuf